realsenseai/realsense-ros #2582: ROS1 Wrapper support for D405?

SprAJR, 2022-12-19:
Just got this awesome, compact D405.  We depend on ROS1 Noetic for the time being.  I assume the D405 post-dates development from the ROS1-legacy branch?  Any ideas on workarounds?  Thanks!

s######@############:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/s######/.ros/log/46793c14-7fc2-11ed-a543-11ef8158cda8/roslaunch-############-39693.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://############:46719/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/reconnect_timeout: 6.0
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [39718]
process[camera/realsense2_camera-2]: started with pid [39719]
[ INFO] [1671470966.899091952]: Initializing nodelet with 12 worker threads.
[ INFO] [1671470967.018836794]: RealSense ROS v2.3.2
[ INFO] [1671470967.018863138]: Built with LibRealSense v2.50.0
[ INFO] [1671470967.018873357]: Running with LibRealSense v2.50.0
[ INFO] [1671470967.050357110]:  
[ INFO] [1671470967.156025840]: Device with serial number 127122270432 was found.

[ INFO] [1671470967.156073107]: Device with physical ID 2-1-7 was found.
[ INFO] [1671470967.156090174]: Device with name Intel RealSense D405 was found.
[ INFO] [1671470967.156805812]: Device with port number 2-1 was found.
[ INFO] [1671470967.156846429]: Device USB type: 3.2
[FATAL] [1671470967.156909224]: Unsupported device! Product ID: 0x0B5B
[camera/realsense2_camera_manager-1] process has died [pid 39718, exit code 1, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/s######/.ros/log/46793c14-7fc2-11ed-a543-11ef8158cda8/camera-realsense2_camera_manager-1.log].
log file: /home/s######/.ros/log/46793c14-7fc2-11ed-a543-11ef8158cda8/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/s######/.ros/log/46793c14-7fc2-11ed-a543-11ef8158cda8/camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2022-12-19:
Hi @SprAJR  The D405 is supported by the **ros2-development** wrapper but not the ros1-legacy wrapper.  Updates for the ros1-legacy wrapper have ceased and there are not plans to add D405 support to it, unfortunately.

The librealsense **2.51.1** SDK added official support for D405 and the camera had improvements over 2.50.0, where D405 was unsupported but still able to work.  For example, the 'disparity shift' option for changing the camera's minimum depth sensing distance did not work in 2.50.0 but did in 2.51.1.  

As far as I am aware the D405 has not been tested with the combination of ros1-legacy and SDK 2.51.1.  Whilst 2.51.1 is similar enough to 2.50.0 to likely be able to work with the final ros1-legacy wrapper that was designed for 2.50.0 (as 2.51.1 is mostly bug-fixes rather than new features), I do not have the expectation that using 2.51.1 with ros1-legacy would make a difference to D405 support.  You are of course welcome to try a wrapper 2.3.2 / librealsense 2.51.1 combination to confirm this though.

SprAJR, 2022-12-19:
Hello @MartyG-RealSense, thanks for the scoop.  I will give the 2.51.1 sdk a go in a bit and will update the community.

MartyG-RealSense, 2022-12-30:
Hi @SprAJR  Do you have an update about this case that you can provide, please?  Thanks!

SprAJR, 2023-01-04:
Hello @MartyG-RealSense, please excuse the delay due to being away.  I was not able to get wrapper 2.32.2 and librealsense 2.51.1 to work with the D405.

```
...
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [577576]
process[camera/realsense2_camera-3]: started with pid [577577]
[ INFO] [1672845653.027728493]: Initializing nodelet with 12 worker threads.
[ INFO] [1672845653.366961932]: RealSense ROS v2.3.2
[ INFO] [1672845653.367035987]: Built with LibRealSense v2.51.1
[ INFO] [1672845653.367077951]: Running with LibRealSense v2.51.1
[ INFO] [1672845653.423911239]:  
[ INFO] [1672845653.444814288]: Device with serial number 127122270432 was found.

[ INFO] [1672845653.444856867]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video2 was found.
[ INFO] [1672845653.444872219]: Device with name Intel RealSense D405 was found.
[ INFO] [1672845653.445196414]: Device with port number 2-1 was found.
[ INFO] [1672845653.445224960]: Device USB type: 3.2
[FATAL] [1672845653.445329467]: Unsupported device! Product ID: 0x0B5B
...
```

MartyG-RealSense, 2023-01-04:
That's awesome, @SprAJR - thanks so much for the confirmation of the solution at https://github.com/IntelRealSense/realsense-ros/pull/2297

closed 2023-01-04 by SprAJR