realsenseai/realsense-ros #2583: Static TFs appear inverted

alexmillane, 2022-12-21:
When I look at the static transforms published by the `realsense2_camera_node` (after running `ros2 run realsense2_camera realsense2_camera_node`) the translation between the camera optical frames appears to be inverted.

For example in the image below the infra1 image comes from the left camera but appears on the right in the tf tree.

Am I missing something here?

[image]

MartyG-RealSense, 2022-12-21:
Hi @alexmillane  On RealSense 400 Series cameras, the left and right positions of the infrared sensors are from the perspective of standing behind the camera looking forwards.  Because of this, if you look at the camera from the front then the left infrared sensor is on the right side of the camera.

alexmillane, 2022-12-21:
Hi Marty. Thank you very much for the reply.

So when I say left in my original post, I mean from the perspective of behind the camera. But that is good to clarify, thank you.

In any case, if I look at the infra1 image above, it appears to me to be taken from a perspective slightly to the left of the perspective of the infra2 image. But this left-side positioning of the infra1 image is inconsistent with the TF tree, which puts infra1 on the right.

Moreover, the reason I actually started looking at this is that when I project the depth pointcloud between the cameras on the realsense body, I get incorrect results unless I invert the cam-to-cam transforms I get from the tf tree. After inversion, everything works perfectly. This leads me to believe that the tf tree is incorrect.

Do you have another interpretation of the image above?

MartyG-RealSense, 2022-12-21:
The only other relevant discussion I could find on this subject was at https://github.com/introlab/rtabmap_ros/issues/517 where infra1 and infra2 were on the opposite sides to your image, but the RealSense user in that case was receiving an error message about the positions.

[image]

alexmillane, 2022-12-21:
Thank you for the link. That picture looks correct to me. Perhaps a mistake has been introduced since that issue was posted? 

I might have a deeper look in January and make a pull request if I find a solution.

In the meantime, if anyone else believes that the tf tree published by the realsense node is correct, I'm interested in hearing about it :).

MartyG-RealSense, 2022-12-21:
Thanks very much @alexmillane for the PR offer.  I will label this case as **wait to close** so that it is kept open.

SamerKhshiboun, 2023-04-06:
Hi @alexmillane,
I've opened a PR #2684  , you are welcome to review and comment there.