realsenseai/realsense-ros #2584: Pointcloud projected is off scale I think

FPSychotic, 2022-12-21:
I just want warn about an issue I found in d435 and d435 cameras. As it is evident something  is absolutely  off in the depth,  not sure if  in the realsense sdk or ros wrapper. In the image you can see, the lidar/d435's depth images fits and match in scale,size of the things, obviously so better the lidar one,  as do the color image, which will colourise lidar and d435 correctly, but the realsense point cloud from Depth projection (aligned to color,) is totally off, it is not  displaced as it looks in a fast view, it is out of scale, because that looks displaced, quite worse thing that only displaced, displacement could be fixed, scale likely not. 
[18:20]
if you use realsense in a critical application related with distance measurement , I would check that. Tested with 3 d435 and d435i, I don't know in what moment that went wrong, but obviously wasn't like that time ago. Could be the lidar , but in that case would be impossible match and calibrate color with lidar, so it is the depth projection what is wrong 

[image]

Noetic
last realsense apt packages.

MartyG-RealSense, 2022-12-22:
Hi @FPSychotic  Are you using the ROS1 or ROS2 wrapper, please?  And what is your ROS launch instruction?

Does color map onto depth correctly in the **3D** pointcloud mode of the RealSense Viewer tool if you enable the depth stream first and then the RGB stream secondly to automatically map color onto the pointcloud depth coordinates?

FPSychotic, 2022-12-22:
In SDK doing what you said with this setup, I think standard still happens. The table is 205cm, and as you say the measurement tool says 265cm. color and depth always match, but the depth is off, but the lateral distance too, so it is the scale what is off.
[image]
`{
  "colorizer.Color Scheme": "0",
  "colorizer.Frames Queue Size": "16",
  "colorizer.Histogram Equalization Enabled": "0",
  "colorizer.Max Distance": "6",
  "colorizer.Min Distance": "0",
  "colorizer.Stream Filter": "1",
  "colorizer.Stream Format Filter": "1",
  "colorizer.Stream Index Filter": "-1",
  "colorizer.Visual Preset": "0",
  "colorizer.enabled": "1",
  "notifications.fw_update_alert.5.14.0.0.845112071035.next": "1672333145",
  "performance.show_fps": "1",
  "ply.encoding": "1",
  "ply.mesh": "0",
  "ply.normals": "1",
  "viewer_model.is_3d_view": "1",
  "viewer_model.is_measuring": "0",
  "viewer_model.output_open": "0",
  "viewer_model.post_processing.Intel RealSense D435I.RGB Camera.Decimation Filter.Filter Magnitude": "2",
  "viewer_model.post_processing.Intel RealSense D435I.RGB Camera.Decimation Filter.Frames Queue Size": "16",
  "viewer_model.post_processing.Intel RealSense D435I.RGB Camera.Decimation Filter.Stream Filter": "2",
  "viewer_model.post_processing.Intel RealSense D435I.RGB Camera.Decimation Filter.Stream Format Filter": "0",
  "viewer_model.post_processing.Intel RealSense D435I.RGB Camera.Decimation Filter.Stream Index Filter": "-1",
  "viewer_model.post_processing.Intel RealSense D435I.RGB Camera.Decimation Filter.enabled": "0",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Decimation Filter.Filter Magnitude": "2",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Decimation Filter.Frames Queue Size": "16",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Decimation Filter.Stream Filter": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Decimation Filter.Stream Format Filter": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Decimation Filter.Stream Index Filter": "-1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Decimation Filter.enabled": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Depth Huffman Decoder.Frames Queue Size": "16",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Depth Huffman Decoder.Stream Filter": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Depth Huffman Decoder.Stream Format Filter": "28",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Depth Huffman Decoder.Stream Index Filter": "-1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Depth Huffman Decoder.enabled": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Depth to Disparity.enabled": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Disparity to Depth.enabled": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Filter By Sequence id.Frames Queue Size": "16",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Filter By Sequence id.Sequence Id": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Filter By Sequence id.enabled": "0",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.HDR Merge.Frames Queue Size": "16",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.HDR Merge.enabled": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Hole Filling Filter.Frames Queue Size": "16",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Hole Filling Filter.Holes Fill": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Hole Filling Filter.Stream Filter": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Hole Filling Filter.Stream Format Filter": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Hole Filling Filter.Stream Index Filter": "-1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Hole Filling Filter.enabled": "0",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Spatial Filter.Filter Magnitude": "2",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Spatial Filter.Filter Smooth Alpha": "0.5",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Spatial Filter.Filter Smooth Delta": "20",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Spatial Filter.Frames Queue Size": "16",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Spatial Filter.Holes Fill": "0",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Spatial Filter.Stream Filter": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Spatial Filter.Stream Format Filter": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Spatial Filter.Stream Index Filter": "-1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Spatial Filter.enabled": "1",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Temporal Filter.Filter Smooth Alpha": "0.4",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Temporal Filter.Filter Smooth Delta": "20",
  "viewer_model.post_processing.Intel RealSense D435I.Stereo Module.Temporal Filter.Frames Queue Size": "16",
[... cut ...]
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>
      
    </include>
  </group>
</launch>`

with ordered true, pointcloud filter, align true will happen the same.

Firmware version 05.13.00.50
but viewer 2.53.1 is asking 5.14.00 , so it could be the firmaware, I'm going to update and check. but even if that is the issue, you should add a warning because in a depth camera, used in drones, robotics etc, scale is off is a very very bad secundary effect by only have a previous firmware version. No good at all

FPSychotic, 2022-12-22:
I tried with a d435 instead d435i , and it just in first impression is evident the image has less fov and the measurement is right.
[image]

MartyG-RealSense, 2022-12-22:
Thanks very much @FPSychotic for your updates above.

It should not be possible for a D435 to be mis-identified for a D455 as they have different internal circuit boards.  D435 has a D430 depth module and V1 version of the Vision Processor D4.  D455 has a D450 depth module, an IMU and V3 version of the Vision Processor D4.  The firmware driver bases its identification of the model number based on the camera components that it can detect.

If the D435 that you tried in the comment above has correct measurement, could you try resetting the seemingly incorrect D435i to its default factory-new calibration in the RealSense Viewer please to see whether this positively affects the D435i's measurements.  This procedure can be performed using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

MartyG-RealSense, 2022-12-22:
You are very welcome.  That's great to hear that a factory reset solved the issue for you!