realsenseai/realsense-ros #2585: In ROS1, does "align_depth:=true" produce "depth/color/points" based aligned depth frame?

JLeeContoro, 2022-12-22:
Hi,

I am using D455 in ROS noetic. I am streaming multiple realsense devices and aligned each camera's depth frame to color using "align_depth". 

Based on the documentation, 
**- align_depth:** If set to true, will publish additional topics for the "aligned depth to color" image.: /camera/aligned_depth_to_color/image_raw, /camera/aligned_depth_to_color/camera_info.
**The pointcloud, if enabled, will be built based on the aligned_depth_to_color image.**

However, I had a misalignment between color and depth when I visualize "aligned_depth_to_color/image_raw" and "/depth/color/points" in Rviz. Please see the bottom image plot. There is a discrepancy in box object border line between depth and color image. I tested it in ROS melodic, and observed the same misalignment. 

[image]

When pointcloud is enabled with "aligned_depth:=true", should it be based on aligned depth frame to color frame? This misaligned result does not make sense to me. I was wondering if I misunderstood something. Any advice would be appreciated. 

Best,
Jay

MartyG-RealSense, 2022-12-23:
hI @JLeeContoro  There is the possibility that there is error in the depth image because the camera's depth sensing is being confused by the similar looking horizontal rows and vertical stacks.  This phenomenon is called a **repetitive pattern**.  Examples of repetitive patterns are a horizontal row of similar looking fence posts or tree tops, or a vertical stack of window blind slats.

Intel have published a PDF white-paper guide about reducing the negative effects of repetitive patterns at the link below.

https://dev.intelrealsense.com/docs/mitigate-repetitive-pattern-effect-stereo-depth-cameras

An example from the guide of a repetitive pattern that is shown below and has a similar empty hole to your depth image.

[image]

JLeeContoro, 2022-12-27:
Hi @MartyG-RealSense, Thank you for the comment. I don't think it is caused by a repetitive pattern. I scanned the same object using ROS2 foxy. In ROS2 foxy, depth alignment works perfectly. There was no misalignment. This indicates that the currently released ROS1 realsense package would have this problem. This is a bit annoying. Do you have another suggestion?

JLeeContoro, 2022-12-28:
@MartyG-RealSense I have an update. I found that this misalignment was temporarily gone when I switched stream_filter from depth to color after initiation of realsense by using dynamic_reconfigure. But it went back to a misaligned stream after a minute. I attached a recorded video. 

https://user-images.githubusercontent.com/102317291/209740064-ddc0f77b-0438-4c5d-8c4a-6915f6006a74.mp4

At the beginning of the video, I reconfigure stream_filter using rosrun dynamic_reconfigure. This make temporarily aligned images but you can see the images went back to the misaligned state at the end of video. 

I have no idea why this occured. Is this known bug for ROS1 with realsense wrapper? Does anybody had the same issues?

MartyG-RealSense, 2022-12-28:
I do not recall there being a bug that resembles the effect that you experienced.

What happens to the alignment if you change the pointcloud texture by adding either of the commands below to the roslaunch instruction instead of using dynamic_reconfigure:

**pointcloud_texture_stream:=RS2_STREAM_COLOR**

**pointcloud_texture_stream:=RS2_STREAM_ANY**

JLeeContoro, 2022-12-28:
@MartyG-RealSense After a few testing by changing parameters, I figured out that the point cloud was well aligned with the color frame using align_depth:=true. However, color texture mapping was not performed properly. I attached sample images below. You could see a clear bad mapping outcome. (e.g., the front box's border was mapped into the behind row's box.)   

**=== wrong color texture mapping**
[image]

**=== correct color texture mapping**
[image]

Technically, there is no problem with using point cloud data since color texture mapping is not a big deal for my project. However, I just wanted to let you know and figure out how to fix this mapping issue.

MartyG-RealSense, 2022-12-28:
Hi @JLeeContoro  Thanks very much for the update!  I'm pleased to hear that you achieved a solution.

Development of the ROS1 wrapper has ceased and the wrapper renamed to **ros1-legacy**.  The focus on the RealSense ROS development team is now on the ROS2 wrapper.

JLeeContoro, 2022-12-28:
Hi @MartyG-RealSense, I understand ros1-legacy is no longer maintained. But I have one last question regarding ros1-legacy. I am just not using color texture mapping, thus I have not had the right solution for my issue with ros1-legacy (e.g., to acquire proper color texture mapping). Do you have any other suggestions to correct the wrong texture mapping issue? Should I manually re-map a color image to a pointcloud data if I want to get a correct color texture mapping? 

I'd appreciate it if you give any suggestions.

MartyG-RealSense, 2022-12-28:
How does it look if you publish the pointcloud as an 'ordered' cloud instead of the default 'unordered' one by adding **ordered_pc:=true** to your roslaunch instruction?

Do you have an improved image if you drop the align instruction and publish a depth aligned to color pointcloud with this instruction:

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

JLeeContoro, 2022-12-28:
Hi @MartyG-RealSense, thank you for the suggestion. 

Unfortunately, **ordered_pc:=true** does not solve the wrong texture mapping issue. 

For the second question, yes. If I am not using align_depth, depth and color images are well matched. It looks totally fine. Only align_depth:=true cause the problem in color texture mapping.

MartyG-RealSense, 2022-12-28:
As using the pointcloud filter maps color onto the depth coordinates in the same way that align_depth does when it is true, would it be possible for your project to use the pointcloud like this without enabling align_depth?

JLeeContoro, 2022-12-28:
@MartyG-RealSense Thank you for the clarification. As far as I know the extrinsic matrix between depth and color frame, I can make it work without enabling align_depth. So, I can use "/camera/extrinsics/depth_to_color" for my project. I think this is the easiest way. 

On a separate topic, is there a way to get a **depth scale value** in ROS wrapper? In a realsense SDK, there is **get_depth_scale()** in a depth_sensor class. I was wondering if there is an equivalent command or topic having this information. If it does not, would you please advise how to extract depth scale from the device?

MartyG-RealSense, 2022-12-28:
The default depth scale for all 400 Series cameras except D405 is **0.001** and does not change unless deliberately altered by a program instruction or by an input from the RealSense user.

The ROS wrapper does not support **changing** the current depth scale value.  The Intel developer who created the ROS1 wrapper explains the reason for this at https://github.com/IntelRealSense/realsense-ros/issues/277#issuecomment-525676873

JLeeContoro, 2022-12-28:
@MartyG-RealSense I will not change depth scale, so the default value should be enough for me. Thank you!

One more question. If I am not using aligned_depth, can I still get depth info by using the color image pixel coordinates? I am doing some processing from color image coordinates and will need to project pixel to world coordinates. So, I was wondering if it is still one-to-one matching coordinates between depth and color image if I am using the same resolution.

MartyG-RealSense, 2022-12-28:
https://github.com/IntelRealSense/librealsense/issues/10916 may be a helpful reference for obtaining XYZ from 2D XY color in ROS1..

JLeeContoro, 2022-12-29:
@MartyG-RealSense I appreciate all your help on this matter. All looks okay now. Thank you! and happy new year in advance!

closed 2022-12-29 by JLeeContoro

MartyG-RealSense, 2022-12-29:
I'm pleased that I could help.   Thanks very much for your well wishes  :)

TemugeB, 2023-05-23:
Why was this thread closed? The issue still exists in ROS1 realsense_camera2.

I calibrated my camera in ROS2 and I can check the pointcloud in Rviz. Looks fine:

```
ros2 run tf2_ros static_transform_publisher -0.014858278982503664 -0.10319116801244611 0.5629973789587905 -0.4731604876086985 0.49619947687217236 0.5096287293908985 0.5197920644806919 workspace camera_link
```

```
ros2 launch realsense2_camera rs_launch.py rgb_camera.profile:=1280x720x15 align_depth.enable:=true pointcloud.enable:=true
```

This shows the following in ROS2 Foxy rviz, which is what I expect.
[image]

In ROS1, the same setup:
```
rosrun tf2_ros static_transform_publisher -0.014858278982503664 -0.10319116801244611 0.5629973789587905 -0.4731604876086985 0.49619947687217236 0.5096287293908985 0.5197920644806919 workspace camera_link
```

```
roslaunch realsense2_camera rs_camera.launch align_depth:=true depth_width:=1280 depth_height:=720 depth_fps:=15 color_width:=1280 color_height:=720 color_fps:=15 enable_pointcloud:=true
```

This shows this image in ROS1 Noetic rviz:
[image]
Notice that the workspace origin has shifted.

Additional, a tilt is introduced in the pointcloud:
[image]

My guess is that the transformation from pointcloud (in depth camera frame) to optical camera frame was not applied or not properly applied. 

The more important question is if the depth image also has this problem or this only exists in the pointcloud. I will try to investigate this.

TemugeB, 2023-05-23:
I tried a second D415 I had. The problem still exists. The origin is still shifted but no more tilt. The tilt is due to misalignment between the depth frame and the optical frame, which would be correct by applying a transformation between the two frames. Again, this leads me to believe that the depth to optical transformation is not being applied correctly.

[image]

[image]

TemugeB, 2023-05-23:
Hi @MartyG-RealSense 

I tried your suggestion and the result was the same. There is a slight tilt and the center of the coordinate system is shifted.
[image]

This feels like this should be my problem because an issue like this would have been discovered quickly and fixed. So I checked what was different between ROS1 and ROS2. It seems ROS1 uses a different set of frames than ROS2.

ROS1
[image]

ROS2
[image]

Additionally, camera_link is defined at the **depth** frame for ROS1 and **color** frame on ROS2. This means that camera_link to robot_base transformation obtained in ROS2 will not work in ROS1. I thought **camera_aligned_depth_to_color_frame** would align the depth frame to color frame, but it seems that this is the reverse transformation, meaning color frame is aligned to depth frame. I assumed align depth meant that the depth image is aligned to the color pixel but it seems that this is not true? The color pixels are aligned to the depth image in ROS1? Is it also the same in ROS2?

MartyG-RealSense, 2023-05-23:
camera_link should relate to the center of the left IR sensor by default, which is the 0,0,0 origin for depth.   When depth to color alignment is performed then the origin should automatically change to the center of the **color** sensor.

myeatman-bdai, 2023-12-15:
I having

> Hi @MartyG-RealSense
> 
> I tried your suggestion and the result was the same. There is a slight tilt and the center of the coordinate system is shifted. ![ros1_5](https://private-user-images.githubusercontent.com/36071915/240184002-936f2d15-ef45-439a-a304-dcf1ef8bd5ee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI2NTc2ODMsIm5iZiI6MTcwMjY1NzM4MywicGF0aCI6Ii8zNjA3MTkxNS8yNDAxODQwMDItOTM2ZjJkMTUtZWY0NS00MzlhLWEzMDQtZGNmMWVmOGJkNWVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE1VDE2MjMwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ4OGIzMjExOGRiNTAyNDA4M2E3YzU3MDdkMjc2MzhmZTgyNDcwM2JjMDRiMTQ1Mzk3M2Q5MWUyZDMzOGZkNTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.41HA5_6OfFolFhC2RryVrXbDf72rmIZdabUWLX4ElEY)
> 
> This feels like this should be my problem because an issue like this would have been discovered quickly and fixed. So I checked what was different between ROS1 and ROS2. It seems ROS1 uses a different set of frames than ROS2.
> 
> ROS1 ![ROS1](https://private-user-images.githubusercontent.com/36071915/240191436-6b2672bb-1f2f-404d-a643-2ead115e56e2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI2NTc2ODMsIm5iZiI6MTcwMjY1NzM4MywicGF0aCI6Ii8zNjA3MTkxNS8yNDAxOTE0MzYtNmIyNjcyYmItMWYyZi00MDRkLWE2NDMtMmVhZDExNWU1NmUyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE1VDE2MjMwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNmNmE3OTY4M2Q3YmRkNWE2ZTEyOWM0ZjRiOTIwOGM3OTBiZDJhZTJlMzllZTMwZWY1YTM0OGY2MzZhMDIyNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vGsK5d4N1DbpTpCoFwpKq5OlXt9uzh_NWJL7YdmYkrQ)
> 
> ROS2 ![ROS2](https://private-user-images.githubusercontent.com/36071915/240191483-f7224765-d09a-449b-b38c-c12907f211e7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI2NTc2ODMsIm5iZiI6MTcwMjY1NzM4MywicGF0aCI6Ii8zNjA3MTkxNS8yNDAxOTE0ODMtZjcyMjQ3NjUtZDA5YS00NDliLWIzOGMtYzEyOTA3ZjIxMWU3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE1VDE2MjMwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUxOTQxMTk3ZGQzYWVhNzBjYjUwZjNiNDM0Y2ZiMjY5MWUzMDBkZTYyOGQzMjhmMmExNDUyYTQ2N2U3MTZkZGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.o3Yf3KxJ6-35NXiUNrx1WJ817jUkvU_42TlwaaQEYxY)
> 
> Additionally, camera_link is defined at the **depth** frame for ROS1 and **color** frame on ROS2. This means that camera_link to robot_base transformation obtained in ROS2 will not work in ROS1. I thought **camera_aligned_depth_to_color_frame** would align the depth frame to color frame, but it seems that this is the reverse transformation, meaning color frame is aligned to depth frame. I assumed align depth meant that the depth image is aligned to the color pixel but it seems that this is not true? The color pixels are aligned to the depth image in ROS1? Is it also the same in ROS2?

I am having this exact same issue.

TemugeB, 2023-12-15:
In my case, I just moved to using ROS2. Seemed like Intel will not be updating the ROS1 packages going forward.

MartyG-RealSense, 2023-12-17:
@myeatman-bdai  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/12457#issuecomment-1844958286 recently found that their image was tilted.  The tilt did not occur when using another camera in the same position.