realsenseai/realsense-ros #2586: Allow the wrapper to publish the aligned and the raw pointclouds simultaneusly

pepisg, 2022-12-29:
I apologize in advance if this has already been asked

I was wandering if its possible to activate a pointcloud publisher for both the aligned and the raw depth streams simultaneously.

Currently it looks like only one pointcloud can be published (the aligned one if alignment is enabled and the raw otherwise) even if both depth images are available.

Thanks in advance for your help!

MartyG-RealSense, 2022-12-30:
Hi @pepisg  I am not aware of a method of publishing two separate pointclouds (raw and aligned) simultaneously from the same camera in the RealSense ROS wrapper, unfortunately.

pepisg, 2022-12-30:
I'm sorry to hear that.
Is this something that could be easily implemented in the current architecture? would you be able to point me on the right direction on how to implement this?

MartyG-RealSense, 2022-12-30:
It is possible to create and display two separate pointclouds (original and processed) in the librealsense SDK, like in the SDK's **rs-post-processing** C++ example at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing

In regard to ROS, the only approach that I can think of would be to convert depth data to a pointcloud with **depth_image_proc** and subscribe to the published topic **sensor_msgs/PointCloud2** to obtain the XYZ pointcloud.

http://wiki.ros.org/depth_image_proc

A pointcloud can be published by the RealSense ROS1 wrapper if the launch file **rs_rgbd.launch** is used, as described at https://github.com/IntelRealSense/realsense-ros/issues/2454#issuecomment-1221254065

MartyG-RealSense, 2023-01-11:
Hi @pepisg  Was the advice in the above comment helpful to you, please?  Thanks!

pepisg, 2023-01-12:
It was! Its a pity that the driver does not support this out of the box, that would be something I would personally find very useful.

Thanks @MartyG-RealSense !

MartyG-RealSense, 2023-01-12:
You are very welcome, @pepisg   :)

closed 2023-01-24 by MartyG-RealSense