realsenseai/realsense-ros #2587: Windows10 Build Fails: colcon build

mjs513, 2022-12-29:
First happy new year to all.
Now for the issue.  I have ROS2 Humble installed on a Windows 10 x64 pc.  Tested and know its configured correctly using the turtlesim example.  Same thing for librealsense, tested using one of the T265 examples using python3.

Now trying to build the ROS2 wrapper and getting errors.  To build followed the instructions here: https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html#id7
and the directory structure looks like this after the failure:
[image]

I ran the following command with the following errors:
```
c:\dev\ros2_ws>colcon build --symlink-install --merge-install
Starting >>> realsense2_camera_msgs
[Processing: realsense2_camera_msgs]
--- stderr: realsense2_camera_msgs
CMake Error at CMakeLists.txt:16 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< realsense2_camera_msgs [50.4s, exited with code 1]

Summary: 0 packages finished [52.6s]
  1 package failed: realsense2_camera_msgs
  1 package had stderr output: realsense2_camera_msgs
  2 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)

c:\dev\ros2_ws>
```

I am attaching the entire error log - might be easier:
[logger_all.log](https://github.com/IntelRealSense/realsense-ros/files/10322419/logger_all.log)

Any help will be appreciated.  Be forewarned - first doing so be warned.

MartyG-RealSense, 2022-12-30:
Hi @mjs513  Are you using **Ubuntu 22.04 Jammy**, please?  You will need 22.04 if you are going to install the RealSense ROS2 wrapper for Humble.

[image]

mjs513, 2022-12-30:
Good Very Early morning Marty
Thanks for the quick response - as always.   

Yes I am using a windows 10x64 so the instructions really don't work well.  So been reading and experimenting for a couple of hours now and am making some progress - slowly but progress using google and alot of your responses to other issues that I have found.

So far managed to get past my initial issue write by:
1.  Making sure I was running the x64 Native VS command prompt in admin mode.
2.   Making sure I sourced ROS2-humble before starting and
3. Double checking my paths are set up correctly (environment variables)

Right now I am getting an error:
```
CMake Error at CMakeLists.txt:96 (find_package):
  By not providing "Findcv_bridge.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "cv_bridge", but CMake did not find one.
```
which basically is telling me I have to add cv-bridge package to ROS2.  So now in process of figuring out how to do that.

Thoughts

Thanks in advance 
Mike

PS.  what happened to occupancy example in the wrapper?  Was going to play with that if I ever get this running.

MartyG-RealSense, 2022-12-30:
I am not able to assist with T265 related enquiries (including the occupancy example), so will have to limit my support to the wrapper installation process.

Having said that, the RealSense ROS occupancy example can be found at the link below .

https://dev.intelrealsense.com/docs/2d-occupancy-map#3-2d-occupancy-map-d435t265

In regard to the OpenCV error, another RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/2369#issuecomment-1151346430 solved it by changing the directory that they sourced to.

This approach did not work though for another user who commented today in that discussion with the same error who was using a Docker container.

mjs513, 2022-12-30:
Understand about the T265 - not a problem.

as for the Findcv_bridge error msg.  Yep - didn;t work for me either.  Under Humble windows to source ros2 you have to run:
`call C:\humble\install\local_setup.bat`

Anyway I did dowload cv-bridge from the package repository for humble and a bit more progress:
```
c:\dev\ros2_ws>colcon build --symlink-install --merge-install
Starting >>> cv_bridge
Starting >>> realsense2_camera_msgs
Starting >>> image_geometry
[Processing: cv_bridge, image_geometry, realsense2_camera_msgs]
[Processing: cv_bridge, image_geometry, realsense2_camera_msgs]
Finished <<< realsense2_camera_msgs [1min 16s]
Starting >>> realsense2_description
--- stderr: image_geometry
CMake Warning at C:/opencv/OpenCVConfig.cmake:166 (message):
  Found OpenCV Windows Pack but it has no binaries compatible with your
  configuration.

  You should manually point CMake variable OpenCV_DIR to your build of OpenCV
  library.
Call Stack (most recent call first):
  CMakeLists.txt:18 (find_package)

CMake Error at CMakeLists.txt:18 (find_package):
  Found package configuration file:

    C:/opencv/OpenCVConfig.cmake

  but it set OpenCV_FOUND to FALSE so package "OpenCV" is considered to be
  NOT FOUND.

```
Now trying to figure out what that means.

MartyG-RealSense, 2022-12-30:
I found the following suggestion for a solution to this error.

[image]

mjs513, 2022-12-30:
After you posted I did an experiment (and then got sidetracked).

After looking at opencv cmakelist I forced it to be true on fail and then get error messages about VS version errors.  After looking a bit more realized I installed VS2022 versus on VS2019 (recommended on the ROS2 Windows install page) didn't think it would matter, but it did.

After changing from 2022 to 2019 got the following error:
```
CMake Error at C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR python3)
Call Stack (most recent call first):
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/FindBoost.cmake:2344 (find_package_handle_standard_args)
  CMakeLists.txt:28 (find_package)
```
Found that they are got rid of the dependency on Boost in ROS2 however with cv-bridge there is a PR that was incorporated
https://github.com/ros-perception/vision_opencv/issues/401
https://github.com/ros-perception/vision_opencv/issues/289

So for the time being going to try and install boost

MartyG-RealSense, 2022-12-30:
Thanks very much again for sharing your feedback.  I look forward to your next report.  Good luck!

mjs513, 2022-12-30:
Ok Marty - this is an interesting one think I got the ros dependencies sorted but now getting this:
```
[](--- stderr: realsense2_camera
CMake Warning at CMakeLists.txt:108 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2"
  (requested version 2.51.1) with any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

CMake Error at CMakeLists.txt:110 (message):

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases)
```
a bit confused.   Looks like ROS2-realsense wants to rebuild realsense camera even though I have the SDK already installed in the program files.

Does this make sense?

MartyG-RealSense, 2022-12-31:
Can you confirm please if you have installed the RealSense SDK on Windows first before you attempted to build the ROS wrapper, using the **Intel.RealSense.SDK-WIN10** installer program in the 'Assets' file list of the SDK **Releases** page?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

[image]

mjs513, 2022-12-31:
@MartyG-RealSense 
Sorry for the delay.  

I had 2.53.1 installed when trying the build.  I uninstalled 2.53.1 and reinstalled using the 2.51.1 link :
[image]

Unfortunately when I went tried to build it again with 2.53.1 I still get the same message:

```
Could not find a package configuration file provided by "realsense2"
  (requested version 2.51.1) with any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.
```

MartyG-RealSense, 2022-12-31:
The SDK can also be built from source code with CMake on Windows.

https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide

mjs513, 2023-01-01:
@MartyG-RealSense 
Happy and Safe New Year - 2023 already

Anyways was back at it and just about ready to give it up.  The issues now reside in realsense-camera (part of realsense-ros).

Turns out I needed to also add in Eigen3 (which I did by dowloading from Eigen.org ) as well as the geometry2 library (tf2).  well I can get past everything except the geometry2 package - dont think its updated yet for humble.  When I do the compile I get:
```
--- stderr: tf2_msgs
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.
```

according the ros2 website Humble doesn't use the catkin install by colcon.  So now really stuck

MartyG-RealSense, 2023-01-02:
The Humble documentation provides information at the link below about using geometry2 with it.

https://docs.ros.org/en/foxy/Releases/Release-Humble-Hawksbill.html#geometry2

mjs513, 2023-01-02:
@MartyG-RealSense 
Thanks for the geometry2 link - I was using a different version - the version link you posted cleared up the issue with having to use catkinConfig.cmake.

Reran this morning after reading your post so got a new error on missing Bullet library. Did find it eventually at 
https://github.com/bulletphysics/bullet3

Kind of using the issue to document what I did just in case anyone else goes down this path

MartyG-RealSense, 2023-01-02:
Thanks so much @mjs513 for sharing your working-out process for the benefit of others who might attempt the same on Windows in future!

MartyG-RealSense, 2023-01-04:
Thanks again @mjs513 for the continued working-out of your procedure.

mjs513, 2023-01-04:
@MartyG-RealSense 
Had hopes it would finish this time now I have one more problem with realsense_camera that i am at a loss on how to fix.
```
--- stderr: realsense2_camera
CMake Error at ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:267 (message):
  ament_cmake_symlink_install_targets() can't find
  'C:/dev/ros2_ws/build/realsense2_camera/Release/realsense2_camera.lib'
Call Stack (most recent call first):
  ament_cmake_symlink_install_targets_1_Release.cmake:1 (ament_cmake_symlink_install_targets)
  ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:317 (include)
  cmake_install.cmake:36 (include)
```
this is the entire error log for realsense_camera:
```
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.19045.
-- Create Release Build.
-- Found ament_cmake: 1.3.2 (C:/dev/ros2_humble/share/ament_cmake/cmake)
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found builtin_interfaces: 1.2.0 (C:/dev/ros2_humble/share/builtin_interfaces/cmake)
-- Found rosidl_generator_c: 3.1.3 (C:/dev/ros2_humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.3 (C:/dev/ros2_humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.3 (C:/dev/ros2_humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found cv_bridge: 3.2.1 (C:/dev/ros2_ws/install/share/cv_bridge/cmake)
-- Found image_transport: 3.1.4 (C:/dev/ros2_humble/share/image_transport/cmake)
-- Found rmw_implementation_cmake: 6.1.0 (C:/dev/ros2_humble/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 6.2.1 (C:/dev/ros2_humble/share/rmw_fastrtps_cpp/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- TinyXML2 was already found when tinyxml2_vendor was included, adding missing imported target tinyxml2::tinyxml2
-- Found rclcpp_components: 16.0.1 (C:/dev/ros2_humble/share/rclcpp_components/cmake)
-- Found realsense2_camera_msgs: 4.51.1 (C:/dev/ros2_ws/install/share/realsense2_camera_msgs/cmake)
-- Found nav_msgs: 4.2.2 (C:/dev/ros2_ws/install/share/nav_msgs/cmake)
-- Found tf2_ros: 0.25.1 (C:/dev/ros2_ws/install/share/tf2_ros/cmake)
-- Found diagnostic_updater: 3.0.0 (C:/dev/ros2_ws/install/share/diagnostic_updater/cmake)
-- Build for ROS2 Humble
-- Configuring done
-- Generating done
-- Build files have been written to: C:/dev/ros2_ws/build/realsense2_camera
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Building Custom Rule C:/dev/ros2_ws/src/realsense-ros-4.51.1/realsense2_camera/CMakeLists.txt
  realsense2_camera.vcxproj -> C:\dev\ros2_ws\build\realsense2_camera\Release\realsense2_camera.dll
  Building Custom Rule C:/dev/ros2_ws/src/realsense-ros-4.51.1/realsense2_camera/CMakeLists.txt
  realsense2_camera_node.vcxproj -> C:\dev\ros2_ws\build\realsense2_camera\Release\realsense2_camera_node.exe
  Building Custom Rule C:/dev/ros2_ws/src/realsense-ros-4.51.1/realsense2_camera/CMakeLists.txt
-- Execute custom install script
-- Symlinking: C:/dev/ros2_ws/install/lib/realsense2_camera/realsense2_camera_node.exe
-- Symlinking: C:/dev/ros2_ws/install/bin/realsense2_camera.dll
CMake Error at ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:267 (message):
  ament_cmake_symlink_install_targets() can't find
  'C:/dev/ros2_ws/build/realsense2_camera/Release/realsense2_camera.lib'
Call Stack (most recent call first):
  ament_cmake_symlink_install_targets_1_Release.cmake:1 (ament_cmake_symlink_install_targets)
  ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:317 (include)
  cmake_install.cmake:36 (include)
  ```

MartyG-RealSense, 2023-01-04:
A suggested solution for the ament error on Windows is at https://discourse.ros.org/t/ament-tutorial-build-fails-with-symlink/1899

mjs513, 2023-01-04:
Afternoon Marty
thanks for the link but I am not sure that will help.  When I did rebuilt I had deleted all three directories - build, install and log as suggested.  Think the error is that "realsense_camera.lib is not in the C:/dev/ros2_ws/build/realsense2_camera/Release/ directory, I am only seeing a dll and a exe file.

Mike

MartyG-RealSense, 2023-01-05:
The only occurrence of this exact ament error **CMake Error at ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:267 (message):
  ament_cmake_symlink_install_targets() can't find** on Windows during colcon build on Humble that I could find is here:

https://www.reddit.com/r/ROS/comments/wxkfes/colcon_build_failed_in_example_failed_examples/

A commenter in that case said that the error disappeared after deleting the failed package.

mjs513, 2023-01-05:
Morning
This is strange I had posted last night that i had finally got it working?  But here goes again.  After rereading the link I tried just using colcon build to see what would happen and turned out a couple of errors about eigen3 not found popped up again or I just missed it first time around.  Basically had to hard code the entire path
`#include <C:/dev/ros2_ws/src/eigen3/Eigen/Geometry>`
in the following files:
```
tf2_eigen.hpp
jacobian.hpp
tf2_eigen_kdl.hpp
and
base_realsense_node.h
```
Once I did that the build completed successfully. And when I ran: 
`ros2_ws>ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x30 pointcloud.enable:=true`
the node started:
```
[INFO] [launch]: All log files can be found below C:\Users\LattePanda1\.ros\log\2023-01-05-06-13-39-181956-DESKTOP-7VRJE96-10324
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node.EXE-1]: process started with pid [8832]
[realsense2_camera_node.EXE-1] [INFO] [1672917221.480216000] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node.EXE-1] [INFO] [1672917221.480571000] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node.EXE-1] [INFO] [1672917221.480686800] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node.EXE-1] [INFO] [1672917222.067643900] [camera.camera]: Device with serial number 935322071561 was found.
[realsense2_camera_node.EXE-1]
[realsense2_camera_node.EXE-1] [INFO] [1672917222.067842700] [camera.camera]: Device with physical ID \\?\usb#vid_8086&pid_0b07&mi_00#7&247fd53&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global was found.
[realsense2_camera_node.EXE-1] [INFO] [1672917222.067936700] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node.EXE-1] [WARN] [1672917222.068237600] [camera.camera]: Error extracting usb port from device with physical ID: \\?\usb#vid_8086&pid_0b07&mi_00#7&247fd53&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global
[realsense2_camera_node.EXE-1] Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[realsense2_camera_node.EXE-1] [INFO] [1672917222.068402700] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node.EXE-1] [INFO] [1672917222.118114900] [camera.camera]: getParameters...
[realsense2_camera_node.EXE-1] [INFO] [1672917222.120307200] [camera.camera]: JSON file is not provided
[realsense2_camera_node.EXE-1] [INFO] [1672917222.120479700] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node.EXE-1] [INFO] [1672917222.120578500] [camera.camera]: Device Serial No: 935322071561
[realsense2_camera_node.EXE-1] [INFO] [1672917222.120673600] [camera.camera]: Device physical port: \\?\usb#vid_8086&pid_0b07&mi_00#7&247fd53&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global
[realsense2_camera_node.EXE-1] [INFO] [1672917222.120795400] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node.EXE-1] [INFO] [1672917222.120967100] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node.EXE-1] [INFO] [1672917222.121066400] [camera.camera]: Sync Mode: Off
[realsense2_camera_node.EXE-1] [WARN] [1672917228.158716800] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node.EXE-1] [INFO] [1672917233.980944300] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x30
[realsense2_camera_node.EXE-1] [WARN] [1672917234.021380800] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {pointcloud.stream_index_filter} doesn't comply with integer range.
[realsense2_camera_node.EXE-1] [INFO] [1672917234.056573400] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node.EXE-1] [INFO] [1672917234.274445900] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node.EXE-1] [INFO] [1672917234.509077200] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node.EXE-1] [INFO] [1672917234.517106500] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node.EXE-1] [INFO] [1672917234.589803800] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node.EXE-1] [WARN] [1672917234.637540900] [camera.camera]:
[realsense2_camera_node.EXE-1] [INFO] [1672917234.715212700] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node.EXE-1] [INFO] [1672917234.734037700] [camera.camera]: RealSense Node Is Up!
```

But RVIZ2 failed to display the pointcloud probably because of this error:
`[realsense2_camera_node.EXE-1] [WARN] [1672917234.021380800] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {pointcloud.stream_index_filter} doesn't comply with integer range.`

MartyG-RealSense, 2023-01-05:
A RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/2295 who experienced the same error concluded that it was because their PC was running off the battery as it did not occur if the PC was attached to the mains socket.  Are you using your computer on battery power?

mjs513, 2023-01-05:
Interesting.  I am running a SBC plugged into the PC Power Port of the USB Powered Hub.  I just attached the D435 to a USB3 powered hub plugged into the SBC so power should be ok?  But will play more with that.

When I run rviz2 I get the following messages about stereo not supported:
```
c:\dev\ros2_humble>rviz2
[INFO] [1672920972.044580500] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1672920972.046480400] [rviz2]: OpenGl version: 4.4 (GLSL 4.4)
[INFO] [1672920972.282454500] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1672921017.402018700] [rviz2]: Stereo is NOT SUPPORTED
```

Read through the Issue and tried the solution you showed in Issue [#2339](https://github.com/IntelRealSense/realsense-ros/issues/2339)
`ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true`
 and that seemed to get rid of that one error message but when I ran rviz2 it still didn;t display anything.  Status was ok but said showing 0 points of 0 msgs.  When I did a ros2 topic list
```
 c:\dev\ros2_humble>ros2 topic list
/color/camera_info
/color/image_raw
/color/metadata
/depth/camera_info
/depth/color/points
/depth/image_rect_raw
/depth/metadata
/extrinsics/depth_to_color
/extrinsics/depth_to_infra1
/extrinsics/depth_to_infra2
/imu
/infra1/camera_info
/infra1/image_rect_raw
/infra1/metadata
/infra2/camera_info
/infra2/image_rect_raw
/infra2/metadata
/parameter_events
/rosout
/tf_static
```

mjs513, 2023-01-05:
Not sure if this matters but I just did 
```
c:\dev\ros2_humble>ros2 topic echo /camera/depth/color/points
WARNING: topic [/camera/depth/color/points] does not appear to be published yet
Could not determine the type for the passed topic
```
but if I do:
```
c:\dev\ros2_humble>ros2 topic echo /depth/color/points
header:
  stamp:
    sec: 1672923711
    nanosec: 298934272
  frame_id: camera_depth_optical_frame
height: 1
width: 287316
fields:
- name: x
  offset: 0
  datatype: 7
  count: 1
- name: y
  offset: 4
  datatype: 7
  count: 1
- name: z
  offset: 8
  datatype: 7
  count: 1
is_bigendian: false
point_step: 16
row_step: 4597056
data:
- 2
- 59
- 123
- 190
- 236
- 172
- 66
- 190
- 232
- 251
- 169
- 62
- 0
- 0
- 0
- 0
- 11
- 109
- 122
- 190
- 236
- 172
- 66
- 190
- 232
- 251
- 169
- 62
- 0
- 0
- 0
- 0
- 20
- 159
- 121
- 190
- 236
- 172
- 66
- 190
- 232
- 251
- 169
- 62
- 0
- 0
- 0
- 0
- 248
- 144
- 121
- 190
- 8
- 67
- 67
- 190
- 250
- 126
- 170
- 62
- 0
- 0
- 0
- 0
- 100
- 194
- 120
- 190
- 8
- 67
- 67
- 190
- 250
- 126
- 170
- 62
- 0
- 0
- 0
- 0
- 109
- 178
- 120
- 190
- 37
- 217
- 67
- 190
- 13
- 2
- 171
- 62
- 0
- 0
- 0
- 0
- 55
- 161
- 120
- 190
- 65
- 111
- 68
- 190
- 31
- 133
- 171
- 62
- 0
- 0
- 0
- 0
- 197
- 142
- 120
- 190
- 94
- 5
- 69
- 190
- 50
- 8
- 172
- 62
- 0
- 0
- 0
- 0
- '...'
is_dense: true
```

mjs513, 2023-01-05:
Sorry for not getting back sooner. Had some health issues today so first time I am back on the computer.

Anyway, took your suggestion and managed to capture a frame but kept getting out of frame resources.  I did find this: [Out of frame resources #1288](https://github.com/IntelRealSense/realsense-ros/issues/1288) so going to try that next.  

Out of curiosity I did run the python opencv example and it had no problems displaying the camera and depth frames.

mjs513, 2023-01-05:
Well got it streaming finally at 30fps but want to change the fps but getting this error:
`[realsense2_camera_node.EXE-1] [ERROR] [1672961723.177982900] [camera.camera]: Given value, 848x640x10 is invalid. Set ROS param back to: 848x480x30`

I am using this command
`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=848x640x10 pointcloud.enable:=true`

MartyG-RealSense, 2023-01-06:
Thanks very much @mjs513 for the continued updates.  It's great to hear that you are having successes.

As the log message states, the supported resolution is 848x480 rather than 848x640.  The minimum supported FPS on D435 for 848x480 resolution is 6.

mjs513, 2023-01-06:
Morming @MartyG-RealSense .  Yeah realized that after I got the message but when I tried to run it at 848x640x6 it gave me the same invalid message.

MartyG-RealSense, 2023-01-06:
Do you mean **848x480x6**

mjs513, 2023-01-06:
No - meant that 6fps is not supported because when I changed it to 848x640x15 it worked.

MartyG-RealSense, 2023-01-06:
848x640 is not listed as a supported resolution for D435 cameras.  If you go to the bottom of your ROS2 launch log, just above the **RealSense Node is Up!** message, then it should list the resolution and FPS that has been applied by the launch.

[image]

MartyG-RealSense, 2023-01-17:
Hi @mjs513  Do you require further assistance with this case, please?  Thanks!

mjs513, 2023-01-20:
Hi @MartyG-RealSense 
Don't think I need assistance with the Realsense right now - having issues with ROS2 and compiling things like bondcpp which is part of NAV2 etc

closed 2023-01-20 by mjs513

MartyG-RealSense, 2023-01-21:
Thanks very much @mjs513 for the update!

mjs513, 2023-01-21:
Just want to say thanks again for all your help in getting this working for me - doubt I could have done it without your assistance.

Respectfully
mike

MartyG-RealSense, 2023-01-21:
You are very welcome.  It's my pleasure to have been able to help.  :)