realsenseai/realsense-ros #2588: Jetson Nano & PC with D435 : 'rs2::invalid_value_error', hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)

scott364, 2022-12-31:
Hello everyone.  I've been receiving the error below on both my jetson nano and my pc, and while I am able to get the realsense D435  to work without any issues in the realsense viewer, when I use it with realsense-ros, I keep getting the error below.  When I got a near idential error earlier, I went into the realsense viewer and updated the firmware to the latest version, and I am getting the error below:

"
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)    "

Any suggestions on what I should do? 
Thanks,
-Scott

-------------------------------------------------------------------------------------------------------------------------------------------------
full output:
scott@Jetbot:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/scott/.ros/log/aee3cd9e-889d-11ed-a755-4c1d96fd5a98/roslaunch-ScheragaJet-8563.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server ~~~~~~~~~~~~~~~~

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/reconnect_timeout: 6.0
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [8584]
ROS_MASTER_URI=~~~~~~~~~~~~~~~~~~

setting /run_id to aee3cd9e-889d-11ed-a755-4c1d96fd5a98
process[rosout-1]: started with pid [8600]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [8603]
process[camera/realsense2_camera-3]: started with pid [8605]
[ INFO] [1672444638.835116628]: Initializing nodelet with 4 worker threads.
[ INFO] [1672444640.199209519]: RealSense ROS v2.2.13
[ INFO] [1672444640.199551586]: Running with LibRealSense v2.50.0
[ INFO] [1672444640.253455473]:  
[ INFO] [1672444640.568618130]: Device with serial number 939622072153 was found.

[ INFO] [1672444640.568780935]: Device with physical ID 2-1.4-3 was found.
[ INFO] [1672444640.568867493]: Device with name Intel RealSense D435 was found.
[ INFO] [1672444640.570402059]: Device with port number 2-1.4 was found.
[ INFO] [1672444640.579242748]: getParameters...
[ INFO] [1672444640.810637885]: setupDevice...
[ INFO] [1672444640.810727985]: JSON file is not provided
[ INFO] [1672444640.810787565]: ROS Node Namespace: camera
[ INFO] [1672444640.810853968]: Device Name: Intel RealSense D435
[ INFO] [1672444640.810912143]: Device Serial No: 939622072153
[ INFO] [1672444640.810963651]: Device physical port: 2-1.4-3
[ INFO] [1672444640.811012242]: Device FW version: 05.14.00.00
[ INFO] [1672444640.811062240]: Device Product ID: 0x0B07
[ INFO] [1672444640.811110259]: Enable PointCloud: Off
[ INFO] [1672444640.811166090]: Align Depth: Off
[ INFO] [1672444640.811209577]: Sync Mode: Off
[ INFO] [1672444640.811322072]: Device Sensors: 
[ INFO] [1672444640.811444879]: Stereo Module was found.
[ INFO] [1672444640.811525656]: RGB Camera was found.
[ INFO] [1672444640.811600809]: num_filters: 0
[ INFO] [1672444640.811644714]: Setting Dynamic reconfig parameters.
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[camera/realsense2_camera_manager-2] process has died [pid 8603, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/scott/.ros/log/aee3cd9e-889d-11ed-a755-4c1d96fd5a98/camera-realsense2_camera_manager-2.log].
log file: /home/scott/.ros/log/aee3cd9e-889d-11ed-a755-4c1d96fd5a98/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/scott/.ros/log/aee3cd9e-889d-11ed-a755-4c1d96fd5a98/camera-realsense2_camera-3*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2022-12-31:
Hi @scott364  It is important to match the RealSense ROS wrapper version as closely as possible with a specific librealsense SDK version listed in the release notes for that wrapper version.  You are using SDK 2.50.0, which should be matched with wrapper **2.3.2** if you are using ROS Melodic.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2

Your log lists though that you are using wrapper 2.2.13, which is very old and was designed for use with librealsense version 2.33.1 from March 2020.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.13

I recommend installing wrapper version 2.3.2 instead.

scott364, 2022-12-31:
Thank you!  That resolved the issue. I have not yet reverted the firmware version on the Realsense, but as everything seems to be working ok, I won't revert the firmware unless another issue pops up.

MartyG-RealSense, 2022-12-31:
It's great to hear that you succeeded!  Thanks very much for the update  :)

closed 2023-01-06 by MartyG-RealSense