realsenseai/realsense-ros #2589: D405 URDF Description

vonHartz, 2023-01-02:
Hi everyone,

following up on #2317, there still seems to be no URDF description for the D405.
When will this be resolved?

Cheers

MartyG-RealSense, 2023-01-02:
Hi @vonHartz  I will raise the issue with the RealSense ROS developer team.  Thanks very much for your question!

MartyG-RealSense, 2023-01-03:
Hi @vonHartz  The RealSense ROS developer team has advised me that the addition of the URDF for D405 is planned for a future wrapper release.

vonHartz, 2023-01-03:
Thanks for your reply @MartyG-RealSense 
Do you have an estimated release date?

MartyG-RealSense, 2023-01-03:
I do not have a time estimate for implementation at the time of writing this, unfortunately.

MartyG-RealSense, 2023-01-09:
As URDF for D405 is planned for a future release, I have added an **Enhancement** tag to this issue so that it is kept open.

chavasse, 2023-03-07:
+1 for this

weaponeer, 2023-03-07:
As well, ++

SamerKhshiboun, 2023-04-11:
Hi @chavasse, @weaponeer, @vonHartz, 
While we are working to add D405 URDF, 
If you can work with dae / step / stl files, this is a good workaround for you:
You can download the cad file of D405 from here:
https://dev.intelrealsense.com/docs/stereo-depth-camera-d400

Then you can convert this STEP file to dae and stl files 
(I managed to download FreeCAD, and convert (File -> Export) these STEP file to .stl and .dae files)

MartyG-RealSense, 2023-04-23:
Hi @chavasse, @weaponeer, @vonHartz  Was the information in the above comment helpful to you, please?  Thanks!

MartyG-RealSense, 2023-04-25:
Hi @chavasse  No problem, thanks very much for your feedback!

SuperN1ck, 2023-12-01:
Hey,
just a friendly ping. Are there any updates when this is planned to be released? This would be super helpful to have!
Cheers,
-Nick

MartyG-RealSense, 2023-12-01:
Hi @SuperN1ck  Thanks very much for your query about a URDF for the D405.  I have highlighted your question to my colleagues on the RealSense ROs team.

Arun-Prasad-V, 2023-12-26:
Hi @vonHartz,  @chavasse, @weaponeer, @SuperN1ck,
A reference D405 URDF description is now available in `ros2-development` branch. Please refer the PR #2953.

Would really appreciate if you guys can try it out and let us know your feedback.
`ros2 launch realsense2_camera rs_d405_pointcloud_launch.py`

Thanks!

MartyG-RealSense, 2024-01-09:
Hi everyone, have you been able to test the new URDF for the D405 camera model that is now available, please - see https://github.com/IntelRealSense/realsense-ros/issues/2589#issuecomment-1869631467

MartyG-RealSense, 2024-01-15:
Case closed due to D405 URDF having been implemented and no further comments received.

closed 2024-01-15 by MartyG-RealSense