realsenseai/realsense-ros #2591: Check patter problem

robotdevel, 2023-01-05:
Hello sir.

Now i am testing D435 rgbd camera based on your ros program.

when i use d435 camera...

I had a problem like which I upload the picture.

can you give me solution about it?

there is empty place but always check pattern is appeared infront of d435 camera

best regards

MartyG-RealSense, 2023-01-05:
Hi @robotdevel  Would it be possible to provide an RGB image of the scene that the camera is observing, please?  If there are floating areas of depth detail on the image that do not come from objects / surfaces in the real world, this could be a phenomenon known as "ghost noise".   

A cause of ghost-noise false depth data can be if the real-world scene has **repetitive patterns**, a series of similar-looking repeating horizontal or vertical shapes (for example, floor / ceiling tiles, a row of tree-tops or a length of fence posts).  If the problem is caused by repetitive patterns, Intel has a guide at the link below for reducing this negative effect.

https://dev.intelrealsense.com/docs/mitigate-repetitive-pattern-effect-stereo-depth-cameras

robotdevel, 2023-01-05:
I attached the image
[image]

it usually happens on stereo camera. am i wrong?

D435 rgbd camera is stereo camera?

MartyG-RealSense, 2023-01-05:
Yes, the D435 and all 400 Series camera models are stereo cameras.

I see on the RGB image that you have fluorescent strip lights on the ceiling.  These can create interference on the image, as they contain heated gas that flickers at frequencies that are difficult to see with the human eye.  Changing a setting called **Power Line Frequency** to a value that more closely matches the operating frequency of the lights can help to reduce the disruption.  Available settings are 30, 50 and 60.  The 50 Hz frequency is commonly used in lights in European regions, whilst 60 Hz is used in North American regions.

robotdevel, 2023-01-05:
we have four D435 cameras but one camera dont have check board patter but other three camera has the problem.

and also the light is LED.

on LED light, above problem can be appeared?

now we test about 60 hz, but the problem is still existing, can you suggest other solution?

MartyG-RealSense, 2023-01-05:
I do not believe that the fluorescent light problems occurs for LED lights.

Looking closely at the ceiling, it also has a tile pattern that corresponds to the checker arrangement on the image.  This increases the chances that it is a 'repetitive pattern' type of disruption.

[image]

robotdevel, 2023-01-05:
Thank you so much about your detail comments.

actually our roslaunch file has "<arg name="enable_color"        default="false"/>"

this parameter for running program.

that problem is still happend when we dont use rgb image?

robotdevel, 2023-01-05:
<arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="640"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="false"/>

  <arg name="fisheye_fps"         default="15"/>
  <arg name="depth_fps"           default="15"/>
  <arg name="infra_fps"           default="15"/>
  <arg name="color_fps"           default="15"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>
  <arg name="enable_pose"         default="false"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_ANY"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="false"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default="pointcloud,disparity,spatial,temporal,decimation"/>
  <arg name="clip_distance"             default="3"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="true"/>
  <arg name="unite_imu_method"          default="copy"/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="false"/>
  <arg name="allow_no_texture_points"   default="false"/>

MartyG-RealSense, 2023-01-05:
The repetitive pattern problem is not related to the RGB image.  It comes from how the camera's depth sensing algorithm translates the scene that it is observing into depth image coordinates.

robotdevel, 2023-01-05:
Ok now I understand well.

So we do not have a solution to avoid that problem in realsense-ros parameter?

i am still curious that why only one camera is fine when I use new camera on this setup.

used camera have a problem but new camera(just opened it from box) this dont have a problem.

MartyG-RealSense, 2023-01-05:
You could eliminate the possibility that the other three used cameras have developed a mis-calibration by resetting them to their default factory-new calibration configuration in the RealSense Viewer tool using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

MartyG-RealSense, 2023-01-05:
You are very welcome.  Yes, I will be here to reply to you.  :)

robotdevel, 2023-01-05:
I solved a problem removing 

```
<arg name="json_file_path" default="camera/MidResHighAccuracyPreset.json"/>

```

can you guess why?

MartyG-RealSense, 2023-01-05:
The **High Accuracy** preset is known for removing excess depth detail from the image and leaving a lot of holes / gaps in the image because it is very strict about which depth values it considers as accurate enough to allow to be shown on the image.  **Medium Density** can be a better choice, as it has a good balance beween accuracy and the amount of depth detail shown on the image.

robotdevel, 2023-01-05:
oh I am sorry i said unclearly.

```
<arg name="json_file_path" default="camera/MidResHighAccuracyPreset.json"/>

```
Using above option, i had a checker board pattern on camera but

```
<arg name="json_file_path" default=""/>

```

I delete that option from ros launch file.

so that option makes the checker board right?

MartyG-RealSense, 2023-01-06:
It would indeed seem that using the High Accuracy preset is the cause of the checker pattern in your particular project by making some of the ceiling tiles be rendered on the depth image but not others.

robotdevel, 2023-01-09:
Thank you so much !

it was very helpful discussion

closed 2023-01-09 by robotdevel

MartyG-RealSense, 2023-01-09:
You are very welcome, @robotdevel - thanks very much for the update!

robotdevel, 2023-02-21:
To solve this issue before that i found, do you have any suggestions?

because I want to avoid this problem before appears.

because I have so many realsense cameras on my robot, I am using your product to develop my service robot.

MartyG-RealSense, 2023-02-21:
If you are using a custom-made ROS launch file then you could try increasing the **laser_power** setting, up to a maximum value of '360'.  Increasing the laser power can make surfaces more easily readable by the camera if those surfaces are not reflective.

<rosparam>
/camera/stereo_module/laser_power: 360
</rosparam>