realsenseai/realsense-ros #2592: External trigger docs and dynamic reconfigure error

dongdongbh, 2023-01-06:
## Multi-Camera configurations
I'm using D435i, I want to use custom external trigger to trigger depth camera. According to white paper of [Multi-Camera configurations](https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration). I should use pins 5 (SYNC) and pins 9 (Ground). In software, set up `inter_cam_sync_mode` as 0=default, 1=Master, and 2=Slave. Then the doc says that "The frequency of the signal generator needs to exactly match the sensor frame rate. ", Does this means the external trigger need to be exactly match the sensor frame rate set in realsense launch file? If it is. Then that isn't my propose. I want to sync with my external clock. 

## External Synchronization
I noticed that  External Synchronization (genlock)  mode, but its white paper link: [https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras](https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#2-principles-of-operation) broken. I guess `inter_cam_sync_mode=4` is what I want. 

## The trigger time
The white paper of Multi-Camera configurations says " the HW SYNC input requires a 100 microsecond positive pulse". but what exact time the camera triggered? Is the time positive pulse start or end, or is related to camera expose time, how can I calculate it?
I also saw there is laser ON-OFF trigger mode with inter_cam_sync_mode 259 and 260. Is it means every trigger will take two frame wirh laser ON-OFF? what is the intervel between these two frames?

## Dynamic reconfigure not work
I encounter the same problem as #984. I use 
```
rosrun dynamic_reconfigure dynparam set /camera/stereo_module inter_cam_sync_mode 0
error updating parameters: can't set parameter 'inter_cam_sync_mode' of <class 'int'>: invalid literal for int() with base 10: '1.0'

dynamic_reconfigure dynparam get /camera/stereo_module
{'enable_auto_exposure': True, 'error_polling_enabled': True, 'output_trigger_enabled': False, 'emitter_on_off': False, 'global_time_enabled': True, 'emitter_always_on': False, 'hdr_enabled': False, 'exposure': 8500, 'gain': 16, 'visual_preset': 0, 'emitter_enabled': 0, 'frames_queue_size': 16, 'inter_cam_sync_mode': 0, 'sequence_name': 0, 'sequence_size': 2, 'sequence_id': 0, 'laser_power': 150.0, 'groups': {'id': 0, 'parent': 0, 'name': 'Default', 'type': '', 'state': True, 'groups': {}, 'parameters': {}, 'exposure': 8500, 'gain': 16, 'visual_preset': 0, 'emitter_enabled': 0, 'frames_queue_size': 16, 'inter_cam_sync_mode': 0, 'sequence_name': 0, 'sequence_size': 2, 'sequence_id': 0, 'laser_power': 150.0, 'enable_auto_exposure': True, 'error_polling_enabled': True, 'output_trigger_enabled': False, 'emitter_on_off': False, 'global_time_enabled': True, 'emitter_always_on': False, 'hdr_enabled': False}}
```       

and I do `rosrun rqt_reconfigure rqt_reconfigure` also not work.

## Test environment
* Thinkpad with `ubuntu 20.04 server`
* RealSense ROS `v2.3.2`
* LibRealSense `v2.53.1`
* ROS1

MartyG-RealSense, 2023-01-06:
Hi @dongdongbh  You are correct, if you are using a trigger generated by an external device instead of by a master RealSense camera then the trigger frequency needs to be matched very closely to the FPS of the slave camera.  This close match can be difficult to achieve.

Trigger sync is easier to achieve with the External Synchronization method of hardware sync.  However, Intel have recently removed the External Synchronization white-paper document from availability.  This is because the paper was an experimental proof of concept and was not officially validated by Intel or considered a mature solution.  The original **Multi-camera configurations** method (modes 1 and 2) is considered by Intel to be mature and validated.

**********

If you would prefer to use mode 4 (genlock) with your external trigger then I have attached an offline PDF version of the genlock paper below.  As mentioned though, this method was experimental and does not have technical support provided for it beyond the information in the paper.

[External Synchronization.pdf](https://github.com/IntelRealSense/realsense-ros/files/10359354/External.Synchronization.pdf)

Modes 259 and 260 are discussed at https://github.com/IntelRealSense/librealsense/issues/10516#issuecomment-1127421620 and the comments beneath it.  259 and 260 produce two frames per trigger with laser ON-OFF (259) and OFF-ON (260).  The difference between 259 and 260 is that with 259, the Laser ON frame is sent first and then the Laser OFF frame secondly.  With 260 it is the opposite - the Laser OFF frame is sent first and then the Laser ON frame secondly.  Information about the interval between frames with these modes is not available.

************

In genlock mode, the camera waits indefinitely for a trigger and does not capture during this wait period.  Once it receives the trigger, exposure starts.

[image]

*********

Are you able to set the inter_cam_sync_mode value if you do so through the graphical interface for dynamic_reconfigure that can be opened with the command **rosrun rqt_reconfigure rqt_reconfigure**

********

librealsense 2.53.1 is not a suitable SDK version to use with ROS1 wrapper 2.3.2.  It is important to match a particular ROS wrapper version as closely as possible with the librealsense version listed in that particular wrapper's release notes.  The best match is version **2.50.1**, and **2.51.1** should also work.

dongdongbh, 2023-01-06:
According the discussion iusse you refered, seem in mode 259 and 260, the interval of two frames may be same. I will test it later.

I have another idea that may produce similar result as I set inter_cam_sync_mode as 259.

If I set inter_cam_sync_mode=4, and enable emitter_on_off option. For example, I use my custom trigger as 30hz in mode 4, there will be 15 image with emitter and 15 without it, they alternate streaming. In this case, I have exact  same timestamp as my 30hz external trigger timestamp which I already know by my external clock. And I can identify the image with emitter on or off by the frame counte (since they alternate). Am I right? Does emitter_on_off still works on inter_cam_sync_mode 4?

I addition, I want to use a microcontroller to generate the trigger signal. The signal has 3.3v positive pulse. Does realsense can tolerate 3.3v?

MartyG-RealSense, 2023-01-06:
emitter_on_off should still work when inter_cam_sync_mode 4 is active.  The emitter component of the camera is independent from the depth sensors so whilst emitter_on_off does indeed alternate the emitter on-off on a per-frame basis (emitter on, then off, then on, etc), it would not be in sync with hardware-sync triggering of depth capture.

The link https://github.com/IntelRealSense/librealsense/pull/3066 with information about the emitter on-off mode has an animated image with a fast-strobing light.

The trigger pulse should be 1.8v.  Some RealSense users performing hardware sync have used a component known as a voltage shifter or level shifter to convert their voltage to 1.8v.  A couple of users have used one similar to the **SparkFun 12009** component mentioned in the guide below.

https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide/all

dongdongbh, 2023-01-07:
“it would not be in sync with hardware-sync triggering of depth capture.” what is this means? If it is not sync with hardware-sync triggering, how does the emitter on and off triggered (or worked)? In my understanding, it should be triggered by external trigger signal. E.g. the first positive pulse trigger the first frame and the emitter on, the second pulse trigger next frame and the emitter off, then the third pulse on, etc.

MartyG-RealSense, 2023-01-07:
Inside the camera, the IR emitter is a separate hardware component from the sensors.  It is part of the projector and operates independently from them.  It projects a semi-random pattern of IR dots onto surfaces in the real-world scene to aid the left and right sensors' ability to read those surfaces for depth information, as the dots act as an analyzable surface texture.

In regard to 'Laser ON all the time': the projector component on the D435 / D435i models pulses in time with exposure, whilst the D415's projector component is on all the time.  It is possible to set the projector to always-on to emulate the behaviour of the D415''s projector on the D435 / D435i.  It would be impractical for the projector to be allowed to be always on in a mode where the projector is toggled on-off though.

dongdongbh, 2023-01-07:
Yes, genlock mode 259 and 260 are designed to work in harmony with hardware sync. 

What I really confused about is, in **genlock mode 4 with emitter_on_off enabled**, the emitter on-off on a per-frame basis, which means both the emitter-on-off and the depth camera are triggered the external trigger hardware pulse. The first positive pulse trigger the first camera frame with the emitter on, the second pulse trigger next camera frame with the emitter off, then the third pulse with emitter on, etc. Is this statement right?

MartyG-RealSense, 2023-01-07:
My understanding is that when the trigger is received, two frames are sent.  The mode value (259 or 260) depends on whether the first frame is Laser On or Laser Off.  So one trigger produces a pair of frames.

dongdongbh, 2023-01-08:
I understand how Genlock modes 259 and 260 works. But I'm asking **inter_cam_sync_mode=4 with emitter_on_off enabled**, not the Genlock modes 259 and 260!!!

MartyG-RealSense, 2023-01-08:
My meaning in the above explanation is that Emitter On Off is a function of the RealSense SDK that was not designed for use with hardware sync.  If your intention is to have Emitter On Off enabled **and** have Inter Cam Sync Mode set to 259 or 260 then you do not need to do both.  Only use one or the other.

If you want the laser to be constantly alternated on-off on a per-frame basis (one frame off, one frame on, one frame off) then enable the Emitter On Off function.  You cannot control the timing of this function with a hardware sync trigger though.  Once enabled, it will continuously alternate the laser until it is disabled.

If you want to precisely control the laser on-off with a trigger then use Inter Cam Sync Mode set to 259 or 260, and keep the Emitter On Off function **disabled** (it is disabled by default).

dongdongbh, 2023-01-08:
Yes, I want to precisely control the laser on-off with a trigger, and I also want to know the trigger time of every frame, and you said "Information about the interval between frames with these modes is not available", but I want to control the trigger of all frames (even each frame time of emitter on and off). So I guess that Inter Cam Sync Mode set to 259 or 260 can not exactly do that. I'm thinking about maybe using inter_cam_sync_mode=4 with emitter_on_off enabled can do that. You said emitter on-off is automatical and uncontrollable. So I think the only way to do that is using mode 259 or 260, and log the timestamp between the emitter on-off pair and align the second one of the pair with my external hardware clock. Thanks for your explanation.

MartyG-RealSense, 2023-01-08:
You are very welcome, @dongdongbh

closed 2023-01-09 by dongdongbh