realsenseai/realsense-ros #2593: Reset realsense in [ROS2]

AlexTiniakov, 2023-01-11:
Hello everyone. In ROS 1 there was a service for reset device: realsense2_camera/reset. But in ROS 2 there is only one service from documentation: device_info. And in reality I see some other services:
realsense2_camera/describe_parameters
realsense2_camera/get_parameter_types
realsense2_camera/get_parameters
realsense2_camera/list_parameters
realsense2_camera/set_parameters
realsense2_camera/set_parameters_atomically

The question is how to reset camera with ROS 2?

MartyG-RealSense, 2023-01-11:
Hi @AlexTiniakov  Are you able to open the **rqt_reconfigure** graphical interface for changing settings dynamically during runtime with the command `ros2 run rqt_reconfigure rqt_reconfigure`

[image]

If you can open the rqt_reconfigure interface, are you able to find a parameter called **initial_reset** under the `/camera` category and set it to True to see whether it results in a reset?

AlexTiniakov, 2023-01-11:
Thanks for the answer. But I haven't problem with initial_reset. I need sometimes reset camera where it became silent during workflow. Can I do it somehow in ROS 2? As it was in ROS 1 with service realsense2_camera/reset

MartyG-RealSense, 2023-01-11:
I'm not aware of a ROS2 reset mechanism other than initial_reset.  If a reset service is not listed when using the command **ros2 service list** then there is likely not a ROS2 equivalent of the ROS1 service.

AlexTiniakov, 2023-01-11:
I tried ros2 run rqt_reconfigure rqt_reconfigure:
[image]
I don't see camera at all. Maybe I need change something in launch file or parameters?

MartyG-RealSense, 2023-01-18:
Ji @AlexTiniakov  Do you require further assistance with this case, please?  Thanks!

AlexTiniakov, 2023-01-19:
Hello. I suppose that camera will work better on ROS 2 and automatically resets if needed. I don't find any mechanism to reset camera when I need to.

MartyG-RealSense, 2023-01-24:
Hi @AlexTiniakov   Can you confirm whether you require further information about your reset problem, please?  Thanks!

MartyG-RealSense, 2023-01-30:
Case closed due to no further comments received.

closed 2023-01-30 by MartyG-RealSense

weeshal, 2023-02-08:
Hi, just read through this thread and am wondering why the reset service was removed? It was used quite extensively for me, and I would like to request some more information on why this was removed in ros2 (if there was some fundamental blocker in ros2) or if this can be made into a feature request. Thanks.

MartyG-RealSense, 2023-02-08:
Hi @weeshal  It looks as though this service was removed some time ago in the previous generation of ROS2 wrapper that is now obsoleted and renamed **ros2-legacy**, so the reason for the removal is not available.

The ros2_legacy wrapper had a service for starting / stopping all streaming sensors.  Could this service be used by you as an alternative to a reset service if it also works in the current **ros2-development** wrapper (as it is not listed in the services section of the ros2-development wrapper's documentation).

For example:

`ros2 service call /camera/enable std_srvs/srv/SetBool "{data: False}"`
 
If this service is not suited to your needs or does not work in ros2_development then I will be happy to raise the subject of a ROS2 reset service with the ros2-development wrapper's developer team.

MartyG-RealSense, 2023-02-26:
After talking to my colleagues on the RealSense ROS developer team about a ROS2 reset service, they thought that it was a good idea and plan to add it in the near future.

PrasRsRos, 2024-10-15:
ROS2 reset service was implemented and merged: 
https://github.com/IntelRealSense/realsense-ros/tree/ros2-development?tab=readme-ov-file#hw_reset