realsenseai/realsense-ros #2594: Depth Image Registration

mconway2579, 2023-01-11:
The problem also documented [here](https://www.reddit.com/r/ROS/comments/108jknc/improving_point_cloud_alignment/)
I am trying to register the depth frame of the real sense to the RGB frame of a thermal camera using ROS [depth_image_proc](http://wiki.ros.org/depth_image_proc) I know the thermal camera is calibrated properly and the tf is correct.

I believe the topic `/realsense/camera/depth/camera_info` is publishing incorrect camera info
`rostopic echo /realsense/camera/depth/camera_info | grep K`
`K: [432.8171081542969, 0.0, 421.9891357421875, 0.0, 432.8171081542969, 239.3887939453125, 0.0, 0.0, 1.0]`
this gives a `focal lengths of (432, 421)` and` principal points of (432, 239) `in the realsense calibration 

when running the realsense health check I get a value of `Health-Check: -0.04 (good)` and when I look at the calibration data I see `focal lengths of (979.963, 970.963)` and `principal points of (955.447, 538.616)`

`/realsense/camera/depth/camera_info` publishing the wrong info would explain the errors I'm seeing, any way to get the real sense wrapper to publish the correct camera info?

MartyG-RealSense, 2023-01-12:
Hi @mconway2579  The principal point and focal length calibration information will be different for each resolution, so you should ensure that you are looking at the calibration data for the particular RealSense depth resolution that you are using in the ROS wrapper.

For example, below is a comparison of my D435 camera's calibration information for 640x480 and 848x480, generated by **rs-enumerate-devices -c**

[image]

[image]

I recommend checking the values that you obtain from **realsense/camera/depth/camera_info** against the values from `rs-enumerate-devices -c` for the specific depth resolution that you are using in your ROS launch to see whether the values match up.

mconway2579, 2023-01-12:
Thank you Marty
When I run `rostopic echo /realsense/camera/depth/camera_info` I can see the published info has `height: 480, width: 848, focal lengths (432, 432) and prinicpal point(421,239)`
When I run `rs-enumerate-devices -c` I can find 
`Intrinsic of "Depth" / 848x480 / {Z16}
  Width:      	848
  Height:     	480
  PPX:        	421.989135742188
  PPY:        	239.388793945312
  Fx:         	432.817108154297
  Fy:         	432.817108154297
  Distortion: 	Brown Conrady
  Coeffs:     	0  	0  	0  	0  	0  
  FOV (deg):  	88.82 x 58.02
`
These are right
I will raise another issue when I know what question I'd like to ask

closed 2023-01-12 by mconway2579

MartyG-RealSense, 2023-01-12:
You are very welcome, @mconway2579 - please do feel free to raise further questions.  :)