realsenseai/realsense-ros #2595: setting align_depth causes misaligned coloured point cloud

christian-rauch, 2023-01-12:
When I set the parameter `align_depth.enable` to `True`, the coloured point cloud will have misaligned colour, i.e. the colour will not be correctly registered to the points.

Consider the following ROS2 launch file:
```yaml
launch:
- node_container:
    pkg: rclcpp_components
    exec: component_container
    name: camera_container
    namespace: realsense
    composable_node:
    -   pkg: realsense2_camera
        plugin: realsense2_camera::RealSenseNodeFactory
        name: camera
        param:
        -   name: rgb_camera.profile
            value: 640x480x30
        -   name: "align_depth.enable"
            value: True
        -   name: pointcloud.enable
            value: True
        -   name: pointcloud.stream_filter
            value: 2 # RS2_STREAM_COLOR
```
The launch file will start the node, set VGA resolution @ 30 fps (`rgb_camera.profile`), align the colour with the depth (`align_depth.enable`) and publish a coloured point cloud (`pointcloud`).

With
```yaml
        -   name: "align_depth.enable"
            value: True
```
the point cloud is shown in RViz2 with misaligned colour: [rs_not_aligned.zip](https://github.com/IntelRealSense/realsense-ros/files/10402071/rs_not_aligned.zip). The image shows the point cloud of a RealSense box. It is clearly visible that the white background is registered to the majority of the box points in the foreground.

When set to `False` or removed, the colour is correctly aligned with the points: [rs_aligned.zip](https://github.com/IntelRealSense/realsense-ros/files/10402079/rs_aligned.zip)

It looks like the colour is registered twice before it is assigned to the points.

MartyG-RealSense, 2023-01-12:
Hi @christian-rauch  If the pointcloud filter is enabled then alignment should automatically be used without having to set align_depth to true.

[image]

christian-rauch, 2023-01-12:
> Hi @christian-rauch If the pointcloud filter is enabled then alignment should automatically be used without having to set align_depth to true.

I am sure that in the default configuration, the node will align depth and colour. If I don't use `align_depth`, then the alignment in the point cloud is correct. But I am additionally setting `align_depth` because I also need the aligned depth images on topic `/aligned_depth_to_color/image_raw`. Enabling both somehow registers the images twice as if the point cloud registration would use the already registered image from `/aligned_depth_to_color/image_raw` instead of the original depth image.

MartyG-RealSense, 2023-01-12:
What happens if you set align_depth to true and change the pointcloud texture configuration by setting **pointcloud.stream_filter** to 0

MartyG-RealSense, 2023-01-12:
I have highlighted this issue to my Intel RealSense colleagues on the ROS team.  Thanks very much for your patience.

cardboardcode, 2023-01-14:
+1. Encountered the same issue of misaligned colored pointcloud after setting `aligned_depth` to true as well. Will update this comment to record and provide the current state of misaligned colored pointcloud shortly.

## Environment
1. Ubuntu 22.04
2. realsense-lib ``
3. ROS2 Humble

## Steps To Reproduce
To be elaborated shortly.

## Verify
To be elaborated shortly.

MartyG-RealSense, 2023-01-14:
Thanks very much @cardboardcode for your report.

cardboardcode, 2023-01-14:
@MartyG-RealSense 
The report has been fully elaborated. Hope it helps.

MartyG-RealSense, 2023-01-19:
After discussion between my Intel RealSense colleagues, an official internal report has been created about this issue so that it can be investigated further by the RealSense ROS development team at a future date.  If other RealSense ROS users experience this issue then they are encouraged to report it on this discussion.

This issue should be kept open whilst the internal report is active.

SamerKhshiboun, 2023-03-27:
Hi @christian-rauch 

Can you please tell me which camera model are you using ? is it D455 as @cardboardcode reported above ?

Thanks,
Samer

christian-rauch, 2023-03-29:
> Can you please tell me which camera model are you using ? is it D455 as @cardboardcode reported above ?

Yes, it is the D455.

SamerKhshiboun, 2023-04-13:
Hi @MartyG-RealSense , I started to look on this, and managed to reproduce this.
looks like when there is a bug when we are trying to texture the aligned frames... will continue analyzing this and will update asap.

cardboardcode, 2023-06-16:
> Hi @christian-rauch & @cardboardcode, Could you please try the PR #2775 and let us know if that works?
> 
> Thanks.

Will update here once I have done so. Thanks @Arun-Prasad-V 😄 👍

MartyG-RealSense, 2026-10-31:
Case closed due PR fix https://github.com/realsenseai/realsense-ros/pull/2775 having been merged in June 2023.

closed 2026-10-31 by MartyG-RealSense