realsenseai/realsense-ros #2596: cannot subscribe to point cloud in RViz when use_intra_process_comms

christian-rauch, 2023-01-12:
When I set `use_intra_process_comms` to true, it is not possible to subscribe to the point cloud in RViz2 via the `/depth/color/points` topic. But it is possible to subscribe to the images.

I assume this is because the point cloud message `_msg_pointcloud` is not published as unique shared pointer like the image messages.

MartyG-RealSense, 2023-01-12:
Hi @christian-rauch  The RealSense ROS wrapper provides instructions at the link below for intra_process_comms.

https://github.com/IntelRealSense/realsense-ros#efficient-intra-process-communication

Known limitations of the process include the `/static_tf` topic being disabled and compressed images using `image_transport` being disabled.

In regard to the static_tf issue, a RealSense ROS user submitted a PR to work around this limitation at https://github.com/IntelRealSense/realsense-ros/pull/2381

christian-rauch, 2023-01-13:
I don't see the point cloud messages mentioned in this documentation. With `use_intra_process_comms` set, I can subscribe to the point clouds on the command line but they do not show up in RViz.

I noticed that RViz prints `[rviz]: Message Filter dropping message: frame 'camera_color_optical_frame' [...] for reason 'discarding message because the queue is full'` when this happens. Do you think this is related to the tf and compressed image issues mentioned in the documentation?

MartyG-RealSense, 2023-01-13:
The last time that I handled a `discarding message because the queue is full` error, a partial solution was achieved by reducing the FPS from 30 to 15, though the message still occurred but less frequently.