realsenseai/realsense-ros #2598: D405 how to turn auto exposure on while using ros2 launch realsense2_camera rs_launch.py

EkamBehl, 2023-01-16:
HI,
I've got a d405 camera .I  subscribed to /camera/color/image_rect_raw and couldn't do detection(was trying to detect chessboard corners  .I opened the image feed on realsense-viewer and it is all white due to lighting condition..The code works on the screenshot of the image after i turn auto exposure on in realsense-viewer.
Is the auto exposure on while running the launch file.If not,how do i turn it on?
i am running ros2 humble .
Thanks for any help

MartyG-RealSense, 2023-01-16:
Hi @EkamBehl  Auto-exposure should be enabled by default in both the realsense-viewer tool and the ROS wrapper.

EkamBehl, 2023-01-17:
Hi @MartyG-RealSense ,
For some reason the auto exposure is not on by default.
Is there a way to turn it on?
I've attached a few images one from rviz on the topic camera/colour/image_rect_raw
And one on real sense with auto exposure on:
[image]
[image]

Adum888, 2023-01-17:
as a preliminary workaround you can use rqt. The check-mark for auto-exposure will be active. Deactivate it and reactivate it. Looking at the infrar image (in Rviz) should confirm the change instantly.

MartyG-RealSense, 2023-01-17:
Please try enabling the auto-exposure after launch during runtime with the ROS2 command below.

`ros2 param set /camera/camera depth_module.enable_auto_exposure true`

EkamBehl, 2023-01-18:
@MartyG-RealSense thanks it worked!!

MartyG-RealSense, 2023-01-18:
That's excellent to hear, @EkamBehl - thanks very much for the update!

MartyG-RealSense, 2023-01-22:
Case closed due to solution achieved and no further comments received.

Adum888, 2023-01-23:
How is this a sufficient solution?
a parameter not working when set during launch
Has to be set after launch (or depending on the initial configuration firstly turned false and then true again)
This is a mediocre workaround at best and not documented at all.

SamerKhshiboun, 2023-01-31:
See my comments in #2507 for more information.
The camera control of auto_exposure was turned off due other parameters in the launch file (e.g. depth_module.exposure.1)

SamerKhshiboun, 2023-03-22:
#2612

MartyG-RealSense, 2023-03-29:
Case closed due to no further comments received.

closed 2023-03-29 by MartyG-RealSense

daichang01, 2024-03-10:
According to my attempts, commenting out the following can achieve automatic exposure
 {'name': 'depth_module.exposure.1',      'default': '7500', 'description': 'Depth module first exposure value. Used for hdr_merge filter'},
                           {'name': 'depth_module.gain.1',          'default': '16', 'description': 'Depth module first gain value. Used for hdr_merge filter'},
                           {'name': 'depth_module.exposure.2',      'default': '1', 'description': 'Depth module second exposure value. Used for hdr_merge filter'},
                           {'name': 'depth_module.gain.2',          'default': '16', 'description': 'Depth module second gain value. Used for hdr_merge filter'},