realsenseai/realsense-ros #2602: Parameter odom_frame_id is not used in ros2-development

elanius, 2023-01-19:
Why is param `odom_frame_id` not used anymore? There is only a default value which is "odom_frame".
I usually use just "odom", and now I cannot overwrite it. 

In the readme file, this parameter is mentioned.
Another misinformation is related to `base_frame_id`, which now appends parameter value to the `camera_name.`

MartyG-RealSense, 2023-01-19:
Hi @elanius  What ROS launch instruction are you using with the ros2-development wrapper, please?  Also, which RealSense camera model are you using?

MartyG-RealSense, 2023-01-24:
Hi @elanius  Do you require further assistance with this case, please?  Thanks!

elanius, 2023-01-27:
Hi @MartyG-RealSense. Now, I use only published odom messages and generate TF by myself. But at least it should be fixed in README.md to be clear for developers that the name for odom frame is fixed. The same is true for `base_frame_id`, which behaves differently as it is written.

MartyG-RealSense, 2023-01-27:
I will label this case as **Documentation** so that it can be looked at by RealSense team members responsible for documentation at a future date.  The case should be kept open, but you do not need to take further action.  Thanks!

MartyG-RealSense, 2026-10-31:
Case closed due to inactivity for over 3 years.

closed 2026-10-31 by MartyG-RealSense