realsenseai/realsense-ros #2606: which parameters are available to modify through ROS2?

patrickwasp, 2023-01-26:
In The RealSense Viewer app you can save the current configuration which lists the parameters shown below.

Are all of these parameters available through the ROS2 package? Specifically, I'm interested in

- `controls-autoexposure-auto`
- `controls-autoexposure-manual`- 
- `controls-color-autoexposure-auto` 
- `controls-color-autoexposure-manual`
-  `controls-laserstate`
- `controls-laserpower`

If it is possible to adjust these in ROS, what are the corresponding launch file parameter names?

```
    "parameters": {
        "aux-param-autoexposure-setpoint": "1536",
        "aux-param-colorcorrection1": "0.298828",
        "aux-param-colorcorrection10": "-0",
        "aux-param-colorcorrection11": "-0",
        "aux-param-colorcorrection12": "-0",
        "aux-param-colorcorrection2": "0.293945",
        "aux-param-colorcorrection3": "0.293945",
        "aux-param-colorcorrection4": "0.114258",
        "aux-param-colorcorrection5": "-0",
        "aux-param-colorcorrection6": "-0",
        "aux-param-colorcorrection7": "-0",
        "aux-param-colorcorrection8": "-0",
        "aux-param-colorcorrection9": "-0",
        "aux-param-depthclampmax": "65536",
        "aux-param-depthclampmin": "0",
        "aux-param-disparityshift": "0",
        "controls-autoexposure-auto": "False",
        "controls-autoexposure-manual": "22222",
        "controls-color-autoexposure-auto": "False",
        "controls-color-autoexposure-manual": "22",
        "controls-color-backlight-compensation": "0",
        "controls-color-brightness": "0",
        "controls-color-contrast": "50",
        "controls-color-gain": "128",
        "controls-color-gamma": "300",
        "controls-color-hue": "0",
        "controls-color-power-line-frequency": "3",
        "controls-color-saturation": "64",
        "controls-color-sharpness": "50",
        "controls-color-white-balance-auto": "True",
        "controls-color-white-balance-manual": "4490",
        "controls-depth-gain": "16",
        "controls-laserpower": "150",
        "controls-laserstate": "on",
        "ignoreSAD": "0",
        "param-amplitude-factor": "0",
        "param-autoexposure-setpoint": "1536",
        "param-censusenablereg-udiameter": "9",
        "param-censusenablereg-vdiameter": "9",
        "param-censususize": "9",
        "param-censusvsize": "9",
        "param-depthclampmax": "65536",
        "param-depthclampmin": "0",
        "param-depthunits": "1000",
        "param-disableraucolor": "0",
        "param-disablesadcolor": "0",
        "param-disablesadnormalize": "0",
        "param-disablesloleftcolor": "0",
        "param-disableslorightcolor": "0",
        "param-disparitymode": "0",
        "param-disparityshift": "0",
        "param-lambdaad": "800",
        "param-lambdacensus": "26",
        "param-leftrightthreshold": "24",
        "param-maxscorethreshb": "2047",
        "param-medianthreshold": "500",
        "param-minscorethresha": "1",
        "param-neighborthresh": "7",
        "param-raumine": "1",
        "param-rauminn": "1",
        "param-rauminnssum": "3",
        "param-raumins": "1",
        "param-rauminw": "1",
        "param-rauminwesum": "3",
        "param-regioncolorthresholdb": "0.0499022",
        "param-regioncolorthresholdg": "0.0499022",
        "param-regioncolorthresholdr": "0.0499022",
        "param-regionshrinku": "3",
        "param-regionshrinkv": "1",
        "param-robbinsmonrodecrement": "10",
        "param-robbinsmonroincrement": "10",
        "param-rsmdiffthreshold": "4",
        "param-rsmrauslodiffthreshold": "1",
        "param-rsmremovethreshold": "0.375",
        "param-scanlineedgetaub": "72",
        "param-scanlineedgetaug": "72",
        "param-scanlineedgetaur": "72",
        "param-scanlinep1": "60",
        "param-scanlinep1onediscon": "105",
        "param-scanlinep1twodiscon": "70",
        "param-scanlinep2": "342",
        "param-scanlinep2onediscon": "190",
        "param-scanlinep2twodiscon": "130",
        "param-secondpeakdelta": "325",
        "param-texturecountthresh": "0",
        "param-texturedifferencethresh": "0",
        "param-usersm": "1",
        "param-zunits": "1000"
    },
```

patrickwasp, 2023-01-26:
This [comment](https://github.com/IntelRealSense/librealsense/issues/6667#issuecomment-647523933) helped me load the parameters I'm interested in directly from the saved json files. It would be nicer to have all the parameters available directly in ROS so we could specify them in a yaml launch file instead of pointing to a json file. 

For reference this is how you could include your realsense json file into your package and load it using a yaml launch file

In your bringup package or wherever you have your launch and config file (I put my json file in a `config` directory inside the package where I launch realsense from) edit your `setup.py` file's `data_files` list to include the json file.

`setup.py`
```python
from setuptools import setup
import os
from glob import glob

package_name = 'my_bringup'

setup(
    name=package_name,
    version='0.0.0',
    packages=[package_name],
    data_files=[
        ('share/ament_index/resource_index/packages',
            ['resource/' + package_name]),
        ('share/' + package_name, ['package.xml']),
        (os.path.join('share', package_name),
         glob('launch/*launch.[pxy][yma]*')),
        (os.path.join('config'),
         glob('config/*')),
    ],
...
```

`my_realsense_launch.yaml`
```yaml
launch: 

- include: 
    file: "$(find-pkg-share realsense2_camera)/launch/rs_launch.py"
    arg:
    - 
        name: "json_file_path"
        value: "$(find-pkg-prefix my_bringup)/config/realsense.json"
    - 
        name: "rgb_camera.profile"
        value: "1280x720x6"
```

closed 2023-01-26 by patrickwasp

MartyG-RealSense, 2023-01-26:
Thanks so much @patrickwasp for sharing your solution with the RealSense ROS community!