realsenseai/realsense-ros #2607: How to spawn camera model using the hand-eye calibration result

ozangungortuhh, 2023-01-26:
Hello,

I applied the eye-in-hand calibration method between my UR5e and Intel Realsense d435i camera and the result is the static frame publishing the relationship between the wrist3_link of my robot and the camera_color_optical_frame. When I run this static publisher I can visualize the TF representation of the camera_color_optical_frame as well as the robot frames. However, I would like to use this result in order to accurately spawn the camera model alongside with the robot. I understand I most probably need to merge the URDF models of the camera and the robot descriptions but what's the exact place where we define the relationship between the color frame and the wrist3 frames?

Thank you in advance for any kind of help.

MartyG-RealSense, 2023-01-26:
Hi @ozangungortuhh  Are you using ROS1 (Kinetic, Melodic, Noetic).  If you are, do you achieve the desired result if you use the UDRF-utilizing roslaunch instruction below?

`roslaunch realsense2_camera rs_d435_camera_with_model.launch`

https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/launch/rs_d435_camera_with_model.launch#L106-L110

ozangungortuhh, 2023-01-26:
Hello, sorry for not mentioning it earlier, I am using ROS Noetic.

Yes, I can see the camera model if I launch the above file. However, what I want to do is to launch both robot and camera models, and to fix the camera_color_optical_frame to a certain distance from the wrist_3 of the robot, which will be the distance I calculated from the hand eye calibration.

[image]

MartyG-RealSense, 2023-01-26:
When seeking to create a link between a joint on a robot and the camera, **Gazebo** could be used to achieve that attachment.  The link below has a guide for doing so with the RealSense ROS1 wrapper.

https://roboticsknowledgebase.com/wiki/tools/gazebo-simulation/

ozangungortuhh, 2023-01-27:
Thank you. I have already merged the camera model together with robot and the gripper. The problem is, I don't know how to define the distance between the camera and the robot by using the TF wrist_3_link -> camera_color_optical_frame. 
I can give the wrist_3_link as a parent in my URDF but it does not accept child_link as an argument. And when I do the following, it uses the robot_camera_bottom_screw and robot_camera_link, which are different to the camera_color_optical_frame positions.

```
  <xacro:sensor_d435 parent="wrist_3_link" name="robot_camera">
        <origin xyz="-0.0328966 -0.0597738 0.0545327" rpy="0 -1.57 0"/>
  </xacro:sensor_d435>
```

```
 child(1):  base
    child(2):  base_link_inertia
        child(1):  shoulder_link
            child(1):  upper_arm_link
                child(1):  forearm_link
                    child(1):  wrist_1_link
                        child(1):  wrist_2_link
                            child(1):  wrist_3_link
                                child(1):  robot_camera_bottom_screw_frame
                                    child(1):  robot_camera_link
```

[image]

MartyG-RealSense, 2023-02-10:
Case closed due to no further comments received.

closed 2023-02-10 by MartyG-RealSense

ozangungortuhh, 2023-02-10:
> Hi @ozangungortuhh Do you require further assistance with this case, please? Thanks!

Apologies for my late response, although I could not fully solve my problem, I understand that it's not a camera or driver issue, more of a ROS application. Thanks for the help @MartyG-RealSense

MartyG-RealSense, 2023-02-10:
You are very welcome.  It may be worth asking the question on the official **ROS Answers** support forum.

https://answers.ros.org/questions/