realsenseai/realsense-ros #2609: Does the L515 make the hardware synchronization among IMU, camera and LiDAR?

jiejie567, 2023-01-29:
I think that L515 doesn't make the hardware synchronization. So how to do that?

MartyG-RealSense, 2023-01-29:
Hi @jiejie567  Hardware synchronization on the L515 camera model has a different purpose to the sync system on the 400 Series cameras.  

Whilst hardware synchronization on the 400 Series aims to have cameras sync their timestamps to the timing of a master camera or an externally generated trigger signal, the L515 sync system negates interference between multiple L515 cameras with overlapping fields of view by time-staggering when separate L515 cameras project light in order to reduce optical interference.  This process is described in the white-paper guide at the link below.

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

jiejie567, 2023-01-29:
@MartyG-RealSense Thank you for reply. I think I don't convey myself clearly. I mean in one L515, whether the hardware synchronization among the camera module, IMU module, and LiDAR module is done?

MartyG-RealSense, 2023-01-29:
Synchronization between sensors on the same camera is a different subject to hardware synchronization, yes.  On 400 Series cameras equipped with an IMU, each IMU data packet is timestamped using the depth sensor hardware clock to allow temporal synchronization between gyro, accel and depth frames.  As far as I am aware the same is true for the L515.

jiejie567, 2023-01-29:
But in issue #1370 , it seems show that they are not synchronized in L515.

MartyG-RealSense, 2023-01-29:
On the 400 Series models, a feature called **Global Time** that aligns device time and system time is enabled by default.  On the L515 model, Global Time is disabled by default.  In the discussion that you quoted, a RealSense team member recommends at https://github.com/IntelRealSense/realsense-ros/issues/1370#issuecomment-777276346 to try enabling Global Time.

jiejie567, 2023-01-29:
hi @MartyG-RealSense , if a IMU message and a image message have the same timestamp, will they correspond to the same real time？

MartyG-RealSense, 2023-01-29:
If IMU and image have the same timestamp then that timestamp should represent the best match between those different stream types that the SDK is able to provide.

jiejie567, 2023-01-30:
[image]
Hi, @MartyG-RealSense , what's more, I wonder that why the "Sync" button in the realsense-viewer is always gray？

MartyG-RealSense, 2023-01-30:
The grayed-out Sync button is not related to hardware sync.  It controls if librealsense will try to bundle frames at software level.  In other words, it is a **software sync** button.  It usually does not work though and so can be ignored.

jiejie567, 2023-01-31:
Hi @MartyG-RealSense , I want to confirm whether the clock sources of IMU and other modules are unified.

MartyG-RealSense, 2023-01-31:
An L515 will behave like the D435i camera model in terms of interactions between the IMU and other sensors.  And as mentioned above at https://github.com/IntelRealSense/realsense-ros/issues/2609#issuecomment-1407589580 the IMU of the D435i (a RealSense 400 Series camera) will have each data packet timestamped using the depth sensor hardware clock to allow temporal synchronization between gyro, accel and depth frames.

At https://github.com/IntelRealSense/librealsense/issues/8249 a RealSense team member also confirms that the coordinate systems of the IMU and depth sensors are aligned and that the L515 uses the same coordinate system as D435i.