realsenseai/realsense-ros #2613: Rviz : camera_depth_frame message filter dropping message because the queue is full

woooj02, 2023-02-03:
Hi there,
I am using ros2-humble and trying to get this running for slam_toolbox. I have converted the depth camera image raw to laserscan using the package from
 https://github.com/ros-perception/depthimage_to_laserscan/tree/foxy-devel.
I can see the topic published by the conversion is displaying the laserscan message and giving valid values. Even though I cannot visualize the topic in rviz using laserscan.  
But whenever I run the launch file  in rviz terminal it displays a line showing 
[rviz]: Message Filter dropping message: frame 'camera_depth_frame' at time 5.000 for a reason 'discarding message because the queue is full'.
How do I fix this problem? I have to use it for the slam toolbox to map the environment and cannot use lidar for this scenario.

MartyG-RealSense, 2023-02-03:
Hi @woooj02  There was a reference to a previous case of this message at the link below.

https://answers.ros.org/question/393773/slam-toolbox-message-filter-dropping-message-for-reason-discarding-message-because-the-queue-is-full/

The RealSense user in that case found that reducing their FPS from 30 to 15 helped to reduce the occurrences of the message, though not eliminate them completely.  They were using a RealSense D455 camera, ROS2 Galactic and a Jetson Xavier NX computing board with depthimage_to_laserscan.

Are you using an Nvidia Jetson board as your computing device, please?

woooj02, 2023-02-04:
Hi,
Thanks for the response.
I should've mentioned it early but I am doing it using the gazebo camera (libgazebo_ros_camera.so). In it, I did try slowing down the update rate but that did not help either.

MartyG-RealSense, 2023-02-04:
If you are not using a RealSense camera or a RealSense launch file then I cannot assist you unfortunately, as this is a support forum for Intel RealSense.  The **ROS Answers** official ROS support forum may be the best place to post your question.

https://answers.ros.org/questions/

MartyG-RealSense, 2023-02-10:
Hi @woooj02  Given the response in the comment above, do you require further assistance with this case please?  Thanks!

MartyG-RealSense, 2023-02-16:
Case closed due to no further comments received.

closed 2023-02-16 by MartyG-RealSense

NorrieDS, 2023-12-15:
Hello, @MartyG-RealSense, I hope you are doing well.

I am running a gazebo simulation with the RealSense D435i plugin attached to my robot model. The architecture and TF are shown in the rqt_graph and tree diagram below. I am using Humble with Cyclone DDS as recommended.

For some reason, rviz2 doesn't show local and global costmaps and says that the **odom** message filter drops message because **the queue is full**.

Do you have any clue what may be causing the issue?
Thank you very much for taking a look at this!

[image]

[frames_2023-12-14_17.53.53.pdf](https://github.com/IntelRealSense/realsense-ros/files/13680235/frames_2023-12-14_17.53.53.pdf)

MartyG-RealSense, 2023-12-15:
Hi @NorrieDS  In a past case where this message was reported, it was found that reducing FPS in the RealSense ROS wrapper from 30 FPs to 15 FPS helped.

https://answers.ros.org/question/393773/slam-toolbox-message-filter-dropping-message-for-reason-discarding-message-because-the-queue-is-full/

NorrieDS, 2023-12-15:
Thanks a lot @MartyG-RealSense for your reply!
Actually, the issue just disappeared when I launched all the nodes from a single launch file, in the same order.
I have no idea what had caused and then solved it.

MartyG-RealSense, 2023-12-15:
You are very welcome.  It's great to hear that you achieved a solution!

MartyG-RealSense, 2025-04-17:
Hi @shrutichakraborty  If you are on a USB 2.1 connection (which does indeed provide reduced performance) then 6 FPS is the only FPS supported on D435i for 1280x720 resolution.  It is also the minimum FPS of the D435i camera overall, so you cannot set an FPS lower than 6 even with a different resolution, unfortunately.