realsenseai/realsense-ros #2614: Output sync signal of pin5 and pin9

linmingyuan, 2023-02-07:
Hi, I have one D455. I try to output the sync signal of pin5 as the external trigger for other cameras. What I expect to get is a pulse wave with rising and falling edges indicating the start and end of the exposure when imaging. However, I get nothing when I connect pin5 and pin9 to an oscilloscope, while getting a 50Hz sine wave when connecting a single pin. It seems to be the same as what [1520870736](https://support.intelrealsense.com/hc/en-us/community/posts/4407190249875/comments/4407191198611) described.

**The launch file I use:**
```
<launch>
  <arg name="serial_no"           default="151422255182"/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="rs"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="360"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="false"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="360"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="360"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="-1"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="wait_for_device_timeout"   default="-1.0"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>

  <arg name="stereo_module/exposure/1"  default="7500"/>
  <arg name="stereo_module/gain/1"      default="16"/>
  <arg name="stereo_module/exposure/2"  default="1"/>
  <arg name="stereo_module/gain/2"      default="16"/>
  
  

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="output"                   value="$(arg output)"/>
      <arg name="respawn"                  value="$(arg respawn)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye"           value="$(arg enable_fisheye)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
      <arg name="depth_height"             value="$(arg depth_height)"/>
      <arg name="enable_depth"             value="$(arg enable_depth)"/>

      <arg name="confidence_width"         value="$(arg confidence_width)"/>
      <arg name="confidence_height"        value="$(arg confidence_height)"/>
      <arg name="enable_confidence"        value="$(arg enable_confidence)"/>
      <arg name="confidence_fps"           value="$(arg confidence_fps)"/>

      <arg name="color_width"              value="$(arg color_width)"/>
      <arg name="color_height"             value="$(arg color_height)"/>
      <arg name="enable_color"             value="$(arg enable_color)"/>

      <arg name="infra_width"              value="$(arg infra_width)"/>
      <arg name="infra_height"             value="$(arg infra_height)"/>
      <arg name="enable_infra"             value="$(arg enable_infra)"/>
      <arg name="enable_infra1"            value="$(arg enable_infra1)"/>
      <arg name="enable_infra2"            value="$(arg enable_infra2)"/>
      <arg name="infra_rgb"                value="$(arg infra_rgb)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="reconnect_timeout"        value="$(arg reconnect_timeout)"/>
      <arg name="wait_for_device_timeout"  value="$(arg wait_for_device_timeout)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="stereo_module/exposure/1" value="$(arg stereo_module/exposure/1)"/>
      <arg name="stereo_module/gain/1"     value="$(arg stereo_module/gain/1)"/>
      <arg name="stereo_module/exposure/2" value="$(arg stereo_module/exposure/2)"/>
      <arg name="stereo_module/gain/2"     value="$(arg stereo_module/gain/2)"/>

      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>
    </include>
  </group>
  <rosparam>
    /rs/stereo_module/intel_cam_sync_mode: 1
  </rosparam>  
</launch>
```

**Two pins I connect  to an oscilloscope:**
[image]

**The positive pole of the oscilloscope is connected to Pin5, and the negative pole of the oscilloscope is connected to Pin9:**
[image]

**The positive pole of the oscilloscope is connected to Pin5/Pin9, and the negative pole of the oscilloscope is not connected:**
[image]
[image]

I want to know if the above phenomenon is normal.

linmingyuan, 2023-02-08:
Thanks for your reply. I wonder if it is correct to only add the below codes in rs_camera.launch for setting inter_cam_sync_mode to '1'. 
```
  <rosparam>
    /rs/stereo_module/intel_cam_sync_mode: 1
  </rosparam> 
```
It did not work in Fig. 2 above.

MartyG-RealSense, 2023-02-08:
The above instruction looks mostly correct.  Can you confirm though please that you have set the mode with **inter_cam_sync_mode** and not 'intel_cam' as shown above?

linmingyuan, 2023-02-08:
**The log is shown below:**
```
... logging to /home/dvs-group/.ros/log/3076d41c-a7a4-11ed-9285-2973cc319159/roslaunch-dvs-group-3054.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dvs-group:43879/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /rs/realsense2_camera/accel_fps: -1
 * /rs/realsense2_camera/accel_frame_id: rs_accel_frame
 * /rs/realsense2_camera/accel_optical_frame_id: rs_accel_optical_...
 * /rs/realsense2_camera/align_depth: True
 * /rs/realsense2_camera/aligned_depth_to_color_frame_id: rs_aligned_depth_...
 * /rs/realsense2_camera/aligned_depth_to_fisheye1_frame_id: rs_aligned_depth_...
 * /rs/realsense2_camera/aligned_depth_to_fisheye2_frame_id: rs_aligned_depth_...
 * /rs/realsense2_camera/aligned_depth_to_fisheye_frame_id: rs_aligned_depth_...
 * /rs/realsense2_camera/aligned_depth_to_infra1_frame_id: rs_aligned_depth_...
 * /rs/realsense2_camera/aligned_depth_to_infra2_frame_id: rs_aligned_depth_...
 * /rs/realsense2_camera/allow_no_texture_points: False
 * /rs/realsense2_camera/base_frame_id: rs_link
 * /rs/realsense2_camera/calib_odom_file:
 * /rs/realsense2_camera/clip_distance: -2.0
 * /rs/realsense2_camera/color_fps: 90
 * /rs/realsense2_camera/color_frame_id: rs_color_frame
 * /rs/realsense2_camera/color_height: 360
 * /rs/realsense2_camera/color_optical_frame_id: rs_color_optical_...
 * /rs/realsense2_camera/color_width: 640
 * /rs/realsense2_camera/confidence_fps: -1
 * /rs/realsense2_camera/confidence_height: -1
 * /rs/realsense2_camera/confidence_width: -1
 * /rs/realsense2_camera/depth_fps: 90
 * /rs/realsense2_camera/depth_frame_id: rs_depth_frame
 * /rs/realsense2_camera/depth_height: 360
 * /rs/realsense2_camera/depth_optical_frame_id: rs_depth_optical_...
 * /rs/realsense2_camera/depth_width: 640
 * /rs/realsense2_camera/device_type:
 * /rs/realsense2_camera/enable_accel: False
 * /rs/realsense2_camera/enable_color: True
 * /rs/realsense2_camera/enable_confidence: False
 * /rs/realsense2_camera/enable_depth: True
 * /rs/realsense2_camera/enable_fisheye1: False
 * /rs/realsense2_camera/enable_fisheye2: False
 * /rs/realsense2_camera/enable_fisheye: False
 * /rs/realsense2_camera/enable_gyro: False
 * /rs/realsense2_camera/enable_infra1: False
 * /rs/realsense2_camera/enable_infra2: False
 * /rs/realsense2_camera/enable_infra: False
 * /rs/realsense2_camera/enable_pointcloud: False
 * /rs/realsense2_camera/enable_pose: False
 * /rs/realsense2_camera/enable_sync: False
 * /rs/realsense2_camera/filters:
 * /rs/realsense2_camera/fisheye1_frame_id: rs_fisheye1_frame
 * /rs/realsense2_camera/fisheye1_optical_frame_id: rs_fisheye1_optic...
 * /rs/realsense2_camera/fisheye2_frame_id: rs_fisheye2_frame
 * /rs/realsense2_camera/fisheye2_optical_frame_id: rs_fisheye2_optic...
 * /rs/realsense2_camera/fisheye_fps: -1
 * /rs/realsense2_camera/fisheye_frame_id: rs_fisheye_frame
 * /rs/realsense2_camera/fisheye_height: -1
 * /rs/realsense2_camera/fisheye_optical_frame_id: rs_fisheye_optica...
 * /rs/realsense2_camera/fisheye_width: -1
 * /rs/realsense2_camera/gyro_fps: -1
 * /rs/realsense2_camera/gyro_frame_id: rs_gyro_frame
 * /rs/realsense2_camera/gyro_optical_frame_id: rs_gyro_optical_f...
[... 78 lines cut ...]
[ INFO] [1675855922.518502408]: Sync Mode: On
[ INFO] [1675855922.518552728]: Device Sensors:
[ INFO] [1675855922.608891050]: Stereo Module was found.
[ INFO] [1675855922.621312468]: RGB Camera was found.
[ INFO] [1675855922.621495411]: Motion Module was found.
[ INFO] [1675855922.621902581]: num_filters: 1
[ INFO] [1675855922.621930784]: Setting Dynamic reconfig parameters.
[ INFO] [1675855925.761697944]: Done Setting Dynamic reconfig parameters.
[ INFO] [1675855925.763286673]: depth stream is enabled - width: 640, height: 360, fps: 90, Format: Z16
[ INFO] [1675855925.763858074]: color stream is enabled - width: 640, height: 360, fps: 90, Format: RGB8
[ INFO] [1675855925.765055844]: setupPublishers...
[ INFO] [1675855925.769910202]: Expected frequency for depth = 90.00000
[ INFO] [1675855925.819956772]: Expected frequency for color = 90.00000
[ INFO] [1675855925.850611836]: Expected frequency for aligned_depth_to_color = 90.00000
[ INFO] [1675855925.876639486]: setupStreams...
 08/02 19:32:05,957 WARNING [140587885909760] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
[ INFO] [1675855926.243151864]: SELECTED BASE:Depth, 0
[ INFO] [1675855926.265620536]: RealSense Node Is Up!
```
where we can find:
```
* /rs/stereo_module/intel_cam_sync_mode: 1
```

I find that the phenomenon in Fig. 2 above is similar to what [snakehaihai](https://github.com/snakehaihai)tps://github.com/IntelRealSense/realsense-ros/issues/2323#issuecomment-1112278047) described.

MartyG-RealSense, 2023-02-08:
Are you able to use the RealSense Viewer tool on your computer?

If you are, is a detectable trigger picked up if you go to the **Stereo Module > Controls** section of the Viewer's options side-panel, enable the **Output Trigger Enabled** option to output a trigger on each frame and then enable the depth stream?

[image]

linmingyuan, 2023-02-09:
Following your comment, I can detect positive pulses as shown below:
[image]

I find that enabling ''output_trigger'' and setting''inter_cam_sync_mode'' is not equal in RealSense Viewer, where ''inter_cam_sync_mode'' is 0 and cannot be changed. Does this mean that setting ''intel_cam_sync_mode'' in the launch file is useless for outputting triggers? So how to enable ''output_trigger'' in the launch file?
[image]

Besides, how to realize ''aligned_depth_to_color'' function in RealSense Viewer?

MartyG-RealSense, 2023-02-09:
The Inter Cam Sync Mode value must be set in the Viewer **before** the depth stream is enabled, as the setting becomes locked once the stream is enabled.  You can set it with a slider or by left-clicking on the pencil icon beside the option and manually typing in '1' with the keyboard.

In the ROS wrapper you could try setting the parameter **output_trigger_enabled** to true instead of setting inter_cam_sync_mode to '1'.

As you are using the ROS1 wrapper (version 2.3.2), you could also try setting inter_cam_sync_mode during runtime after launch has completed with the following command:

`rosrun dynamic_reconfigure dynparam get /camera/stereo_module inter_cam_sync_mode 1`

The equivalent rosrun command for output_trigger_enabled should be:

`rosrun dynamic_reconfigure dynparam get /camera/stereo_module output_trigger_enabled true`

*********

A command that can be controlled with dynamic_reconfigure should also have a launch file **rosparam** equivalent with the same parameter name.  So I would expect launch file rosparam equivalents of the above two commands to look something like this:

```
<rosparam>
/camera/stereo_module/output_trigger_enabled: true
</rosparam>
```

```
<rosparam>
/camera/stereo_module/inter_cam_sync_mode: 1
</rosparam>
```

******

The RealSense Viewer does not have an alignment function in its 2D mode, but you can map RGB texture onto 3D depth points by going into **3D** pointcloud mode and first enabling the depth stream and then secondly enabling the RGB stream.  This type of alignment is not the same as the align_depth alignment in the ROS wrapper though.

The RealSense SDK does have depth-color alignment example programs that work on the same principle as align_depth.

**C++**
https://github.com/IntelRealSense/librealsense/tree/master/examples/align

**Python**
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

linmingyuan, 2023-02-09:
I try the codes below and it works! Thanks!!!
```
<rosparam>
/camera/stereo_module/output_trigger_enabled: true
</rosparam>
```

closed 2023-02-09 by linmingyuan

MartyG-RealSense, 2023-02-09:
You are very welcome.  It's great to hear that you were successful.  Thanks very much for the update!