realsenseai/realsense-ros #2616: Feature request: Publish exposure time

schmidtp1, 2023-02-08:
Would it be possible to publish the exposure time as part of the metadata? Thank you!

MartyG-RealSense, 2023-02-08:
Hi @schmidtp1  The actual exposure value for depth can be obtained from metadata when auto-exposure is enabled.  However, the exposure value for RGB cannot be obtained whilst auto-exposure is enabled, as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2549#issuecomment-431863920

Is RGB exposure the metadata type that you are seeking, please?

schmidtp1, 2023-02-08:
Thank you for the quick reply and the helpful reference. That is a bit disappointing that we can't get RGB exposure.

Yes, it's the value I'm looking for but the request is about exposing it in ROS (e.g. in form of a topic).

MartyG-RealSense, 2023-02-09:
It may be possible to retrieve the **manual** RGB camera exposure value using the `rosparam get <parameter name>` command, though this is not exposing it in the form of a topic. 

http://wiki.ros.org/rosparam#rosparam_get

`rosparam get /` lists all dynamically configurable parameters.

schmidtp1, 2023-02-10:
Thank you for following up but we need auto-exposure.
I assume we can get the parameter for depth?
We need it for every frame, i.e. 30Hz. Wouldn't it be better to have a topic at that frequency?

MartyG-RealSense, 2023-02-10:
I have highlighted your query about the possibility of RGB exposure metadata to the RealSense ROS team.  Thanks very much for your patience!

Depth exposure is retrievable during auto-exposure from the metadata parameter **ACTUAL_EXPOSURE**.

MartyG-RealSense, 2023-02-10:
The advice that I have received is that that ROS publishes all metadata that librealsense exposes, so if librealsense cannot expose RGB exposure metadata when auto-exposure is enabled then ROS cannot publish it either, unfotunately.

schmidtp1, 2023-02-10:
That makes sense. Just to make sure I understand correctly, does "publishes all metadata" mean it is published as a topic, or can be retrieved via parameter (as described above)?

MartyG-RealSense, 2023-02-10:
Regarding publishing to a topic, my colleagues on the RealSense ROS developer team explained that ROS has a generic mechanism that iterates through all of the supported metadata values and writes them on the json string for publishing on the metadata topic.  That means it publishes all metadata that the librealsense SDK exposes.

Furthermore, librealsense also has a generic mechanism which gets the metadata attributes from the firmware and its **validity flag**.  With the 400 Series' RGB sensor, once you activate auto-exposure then the manual exposure validity flag turns false at the firmware side.  This behaviour can be seen in the RealSense Viewer tool.  Once you turn on the auto-exposure, the manual exposure value disappears from the metadata list.  

As the origin of this block is from the firmware, the RealSense ROS team cannot affect it.  RGB exposure therefore cannot be retrieved with a parameter, unfortunately.

schmidtp1, 2023-02-11:
That's great that it is already published (within the known limitations). Thank you for pointing me where to look! Closing

closed 2023-02-11 by schmidtp1