realsenseai/realsense-ros #2620: Error when trying to use Realsense-ROS

pfcouto, 2023-02-10:
I am following this guide https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i and I am having some issues. I am using Ubuntu 20.04 through distrobox (ON my machine I have fedora 37 and then I have a distro box with ubuntu 20.04 (this might be the cause of the error)).

Done all the steps until this one (https://github.com/IntelRealSense/realsense-ros) that outputs the error (second picture). As visible it is not entering the install folder. So in order to make it accessible I went and added the /install on the local_setup.bash (third image)
[image]

[image]

[image]
 
And then when I try to run the command `ros2 launch realsense2_camera opensource_tracking.launch` it outputs the following error. As visible it is accessing the install folder looking for the file and not the src folder to which I cloned your the github repo. Can you help out??? Thanks! By the way, I search the files looking for the opensource_tracking.launch file but didn't find it. Can you tell me its location? Thanks!

[image]

pfcouto, 2023-02-11:
OK. This file only is present in ROS1 branch.

I want to to a slam using only a D435i using ROS2. Can you guide me? Thanks!

MartyG-RealSense, 2023-02-11:
Hi @pfcouto  Of the limited number of available information resources about using SLAM with RealSense and ROS2, they have typically involved using a combination of slam_toolbox and depthinage_to_laserscan, as described at https://github.com/IntelRealSense/realsense-ros/issues/2387

pfcouto, 2023-02-11:
OK. I will look into that. IN the mean time I was trying to use ROS1. As per the guide in https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i I ran the command `roslaunch realsense2_camera opensource_tracking.launch` and I can't see any output in RViz. I haven't personalize RViz like in the guide, However I would like to fix that error and provide the odometry. How can I do that?? If I run the command `roslaunch realsense2_camera rs_d435_camera_with_model.launch` I have the output shown in the second image. It is not the SLAM but it is the 3D (Image+depth) of the camera. What I need to do now is to create a SLAM using my D435i. Which steps should I do now to make it? Thanks!

[image]

[image]

MartyG-RealSense, 2023-02-12:
Hi @pfcouto  The IMU gyro and accel topics are disabled by default in the ROS1 wrapper, so they should be enabled in the roslaunch instruction.  They were enabled by default at the time when the D435i SLAM guide was originally written, which is why the guide does not mention this.  

`roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation`

pfcouto, 2023-02-13:
Hi. Tried the command and also personalized RViz like in https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i. I am still getting some error as shown in the pictures. One of them being the the same as before, about the ometry. And I still can't see the SLAM, do I have to enable more options? What should I do? Thanks!

[image]

[image]

[image]

pfcouto, 2023-02-13:
Managed to solve the first error evolving the octomap_rviz_plugin by running the command `sudo apt-get install ros-noetic-octomap-rviz-plugins`. I get the error about the odometry but not always, I believe that it is caused when I move the camera to fast. However, I still can't see the slam. When I activate the option `ColorOccupancyGrid` I can see a "SLAM" made with the depths (I believe), and when I enable map a 2D image also appears, but I can't find a way to show the SLAM. As shown in the picture 3 I get several warnings about time difference between RGB and Depth, can that cause  the problem? Also I sometimes get the error in picture 3 when I enable the option MapCloud.  Can you help? Thanks!

[image]

[image]

[image]

MartyG-RealSense, 2023-02-13:
In an **rtabmap_ros** support forum case at https://github.com/introlab/rtabmap_ros/issues/785#issuecomment-1196850676 that features this  "the time difference between rgb and depth frames is high" warning with a RealSense D435i, they suggest including **align_depth:=true** in the roslaunch instruction to align depth to color if depth and color are not using the same resolution.

pfcouto, 2023-02-13:
Ok. I will add it. However, why can't I see the SLAM in RViz?

pfcouto, 2023-02-14:
I think I managed to make it work. Thanks! However I do have more questions. I found this code https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py that makes a pointCloud. The thing is, I would like to inference on the RGB image, make detections (using YOLO or Mask), represent it as a 3D bounding box on pointcloud and make a SLAM with that pointcloud. Meaning that I would have a SLAM with object detections represented as 3D Bounding boxes. Do you know if anyone as done this before, or do you know a way to guide me? I believe the main issues are how to create and represent a 3D bounding box and how to create a SLAM from the point clouds with the bounding boxes. Thanks!

MartyG-RealSense, 2023-02-14:
The project at the link below may meet some of your needs of RealSense with YOLO, point clouds and SLAM.  It also makes use of Darknet.

https://github.com/Mazhichaoruya/Object-Detection-and-location-RealsenseD435

If you need **3D** bounding boxes then you could try using **darknet_ros_3d**

https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d

In regard to hardware acceleration, if you were using C++ language then you could use the RealSense SDK's **GLSL** feature to offload processing work for alignment, pointclouds, etc from the CPU onto the graphics GPU by changing RS2:: script instructions to RS2::GL::.  This will not work with the Python language though as pyrealsense2 scripts do not use RS2:: instructions.

https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

If the program was being run on an Nvidia Jetson computing board then a RealSense pointcloud could take advantage of **CUDA** graphics acceleration.  CUDA only works with Nvidia graphics GPUs, like the one installed on Nvidia Jetson boards.

MartyG-RealSense, 2023-02-19:
Hi @pfcouto  Do you require further assistance with this case, please?  Thanks!

pfcouto, 2023-02-19:
I guess not. I will be closing this issue. Thanks!

closed 2023-02-19 by pfcouto