realsenseai/realsense-ros #2621: Unable to visualize DepthCloud from bag file.

echoGee, 2023-02-11:
I bag filed couple of topics after running the d435 with ros noetic. I'm able to see the DepthCloud in rviz when I run this command
`roslaunch realsense2_camera rs_d435_camera_with_model.launch depth_width:=1280 depth_height:=720 infra_width:=1280 infra_height:=720 infra_fps:=6 depth_fps:=6`
But after bagging few of the topics 
```
/camera/color/image_raw
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/clicked_point
/clock
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static
```
and playing them back, I'm not able to see the depthCloud in rviz 
bag file is here: https://drive.google.com/file/d/1k68oZgBgtb8Up1-ioSMLsGEuoMIkw8PW/view?usp=sharing

MartyG-RealSense, 2023-02-12:
Does it play back if you use **rs_camera.launch** as the launch file instead of 'rs_d435_camera_with_model' to record your bag file, please?

roslaunch realsense2_camera rs_camera.launch depth_width:=1280 depth_height:=720 infra_width:=1280 infra_height:=720 infra_fps:=6 depth_fps:=6

If it still does not work, Intel's D435i SLAM guide has advice about replaying a RealSense bag file and visualizing it in RViz.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

*********

```
roscore >/dev/null 2>&1 &
rosparam set use_sim_time true
rosbag play my_bagfile_1.bag --clock
roslaunch realsense2_camera opensource_tracking.launch offline:=true
```

Change the roslaunch instruction to the one that you are using, making sure to put **offline:=true** on the end of it.

echoGee, 2023-02-13:
Thanks @MartyG-RealSense 
I see that following two topics are also required 
```
/camera/depth/metadata 
/camera/depth/camera_info
```

closed 2023-02-13 by echoGee

MartyG-RealSense, 2023-02-14:
You are very welcome, @echoGee - it's excellent to hear that you achieved a solution.  Thanks very much for the update!