realsenseai/realsense-ros #2622: colcon build fails for realsense_camera package

fahmyadan, 2023-02-12:
``` bash 
OS: Ubuntu 22.04 Jammy 
ROS Distro: Hawksbill Humble
``` 
I had an issue when building the realsense packages, specifically realsense_camera. I think it happened because cmake could not find the include directory for some reason. The error is shown below; 
```bash 
CMake Error at /lib/cmake/realsense2/realsense2Config.cmake:11 (message):
  File or directory //include referenced by variable realsense2_INCLUDE_DIR
  does not exist !
Call Stack (most recent call first):
  /lib/cmake/realsense2/realsense2Config.cmake:33 (set_and_check)
  CMakeLists.txt:44 (find_package)
```

I was able to get the packages to build successfully by changing `set_and_check(..."${PACKAGE_PREFIX_DIR}/include")` to `set_and_check(..."${PACKAGE_PREFIX_DIR}/usr")`  in line 33 of  `/lib/cmake/realsense2/realsense2Config.cmake`. 

I was able to launch the nodes and echo the topics from the camera so it seems to work ok for now. I'm not sure why this works but I would be grateful if someone could provide a brief explanation in case this is not good practice. 

Thanks in advance.

MartyG-RealSense, 2023-02-12:
Hi @fahmyadan  Which version of the RealSense ROS wrapper are you using, please?  The only version that supports Humble at the time of writing this is the **4.51.1** ROS2 wrapper.

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.51.1

fahmyadan, 2023-02-13:
I am using 4.51.1 and it appears to be working fine with some minor changes. Thanks for the help!

closed 2023-02-13 by fahmyadan

MartyG-RealSense, 2023-02-13:
I'm pleased to hear that you were successful, @fahmyadan - thanks very much for the update!