realsenseai/realsense-ros #2626: Showing depth as 0.0 always

Gaurav37, 2023-02-14:
Hi,

Im using realsense on Nvidia Tegra Xavier.
For some reason it displays the distance of object as 0.0 always. I'm unable to find error if any in my script. PFB launch file and my script.
```
<?xml version="1.0" ?>
<launch>
    <arg default="false" name="reset"/>
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="camera" value="Main" />
        <arg name="tf_prefix" value="Main" />
        <arg name="serial_no" value="146222253369" />
        <arg name="filters" value="pointcloud" />
        <arg name="depth_width" value="480" />
        <arg name="depth_height" value="270" />
        <arg name="depth_fps" value="5"/>
        <!--arg name="clip_distance" value="6.0" /> -->
        <arg name="color_width" value="480" />
        <arg name="color_height" value="270" />
        <arg name="color_fps" value="5" />
	    <arg name="enable_infra" value="false" />
        <arg name="enable_color" value="true" />
        <!-- <arg name="enable_gyro" value="true" /> -->
        <!-- <arg name="enable_accel" value="true" /> -->
        <!-- <arg name="unite_imu_method" value="linear_interpolation" /> -->
        <arg name="pointcloud_texture_stream" value="RS2_STREAM_ANY" />
        <arg name="initial_reset" value="$(arg reset)" />
    </include>
</launch>
```
The script is:
```
#!/usr/bin/env python3
import sys
import numpy as np
from sensor_msgs.msg import Image
import rospy
import sys
import cv2
import struct
import ros_numpy

def imgmsg_to_cv2_depth(img_msg):
    image_opencv = ros_numpy.numpify(img_msg)
    # print("Encoding is",img_msg.encoding)
    # if img_msg.encoding != "bgr8":
    #     rospy.logerr("This Coral detect node has been hardcoded to the 'bgr8' encoding.  Come change the code if you're actually trying to implement a new camera")
    # dtype = np.dtype("uint16") # Hardcode to 8 bits... uint16
    # dtype = dtype.newbyteorder('>' if img_msg.is_bigendian else '<')
    # image_opencv = np.ndarray(shape=(img_msg.height, img_msg.width), # and three channels of data. Since OpenCV works with bgr natively, we don't need to reorder the channels.
    #                 dtype=dtype, buffer=img_msg.data)
    # # If the byt order is different between the message and the system.
    # if img_msg.is_bigendian == (sys.byteorder == 'little'):
    #     image_opencv = image_opencv.byteswap().newbyteorder()
    return image_opencv

def imgmsg_to_cv2(img_msg):
    # print("Encoding is",img_msg.encoding)
    # if img_msg.encoding != "bgr8":
    #     rospy.logerr("This Coral detect node has been hardcoded to the 'bgr8' encoding.  Come change the code if you're actually trying to implement a new camera")
    image_opencv = ros_numpy.numpify(img_msg)
    # dtype = np.dtype("uint8") # Hardcode to 8 bits...
    # dtype = dtype.newbyteorder('>' if img_msg.is_bigendian else '<')
    # image_opencv = np.ndarray(shape=(img_msg.height, img_msg.width, 3), # and three channels of data. Since OpenCV works with bgr natively, we don't need to reorder the channels.
    #                 dtype=dtype, buffer=img_msg.data)
    # # If the byt order is different between the message and the system.
    # if img_msg.is_bigendian == (sys.byteorder == 'little'):
    #     image_opencv = image_opencv.byteswap().newbyteorder()
    return image_opencv

def cv2_to_imgmsg(cv_image):
    img_msg = Image()
    img_msg.height = cv_image.shape[0]
    img_msg.width = cv_image.shape[1]
    img_msg.encoding = "rgb8"
    img_msg.is_bigendian = 0
    img_msg.data = cv_image.tostring()
    img_msg.step = len(img_msg.data) // img_msg.height # That double line is actually integer division, not a comment
    # img_msg= ros_numpy.msgify(Image, cv_image.data())
    return img_msg

net = cv2.dnn.readNetFromCaffe("/home/ascendbir/catkin_human/src/human_detection/scripts/MobileNetSSD_deploy.prototxt", "/home/ascendbir/catkin_human/src/human_detection/scripts/MobileNetSSD_deploy.caffemodel")
inScaleFactor = 0.007843
meanVal       = 127.53
classNames = ("background", "aeroplane", "bicycle", "bird", "boat",
              "bottle", "bus", "car", "cat", "chair",
              "cow", "diningtable", "dog", "horse",
              "motorbike", "person", "pottedplant",
              "sheep", "sofa", "train", "tvmonitor")
cv_depth_image= np.array(1)
# cv_depth_image_main= np.array(1)

class image_converter:

  def __init__(self):
    print("Im init")
    # self.image_pub = rospy.Publisher("image_topic_2",Image,queue_size=10)
    self.image_sub = rospy.Subscriber("/Main/color/image_raw",Image,self.callback)
    self.image_sub_depth= rospy.Subscriber("/Main/depth/image_rect_raw", Image,self.callback_depth)
    

  def callback_depth(self, data):
    global cv_depth_image
    print("Im here")
    cv_depth_image= imgmsg_to_cv2_depth(data)
    # cv2.imshow("Depth Image",cv_depth_image)# Display depth image just to check that we are correct
    # cv2.waitKey(1)

  def callback(self,data):
    print("callback")
    cv_image = imgmsg_to_cv2(data)
    height, width = cv_image.shape[:2]
    expected = 600
    aspect = width / height
    resized_image = cv2.resize(cv_image, (round(expected * aspect), expected))
    crop_start = round(expected * (aspect - 1) / 2)
    crop_img = resized_image[0:expected, crop_start:crop_start+expected]
    # img_msg = Image()
    # img_msg.height = cv_image.shape[0]
    # img_msg.width = cv_image.shape[1]
    # img_msg.encoding = "rgb8"
    # img_msg.is_bigendian = 0
    # img_msg.data = cv_image.tostring()
    # img_msg.step = len(img_msg.data) // img_msg.height # That double line is actually integer division, not a comment
    blob = cv2.dnn.blobFromImage(crop_img, inScaleFactor, (expected, expected), meanVal, False)
    net.setInput(blob, "data")
    detections = net.forward("detection_out")

    label = detections[0,0,0,1]
    conf  = detections[0,0,0,2]
    xmin  = detections[0,0,0,3]
    ymin  = detections[0,0,0,4]
    xmax  = detections[0,0,0,5]
    ymax  = detections[0,0,0,6]

    className = classNames[int(label)]
    scale = height / expected
    xmin_depth = int((xmin * expected + crop_start) * scale)
    ymin_depth = int((ymin * expected) * scale)
    xmax_depth = int((xmax * expected + crop_start) * scale)
    ymax_depth = int((ymax * expected) * scale)

    # print(xmin_depth, xmax_depth, ymin_depth,ymax_depth)

    depth = cv_depth_image[xmin_depth:xmax_depth,ymin_depth:ymax_depth].astype(float)
    # print(depth)
    depth= depth*0.001
    dist,_,_,_ = cv2.mean(depth)
    if className=="person":
      print("dist of",className," is:", dist)
    cv2.rectangle(crop_img, (int(xmin * expected), int(ymin * expected)),(int(xmax * expected), int(ymax * expected)), (255, 255, 255), 2)
    cv2.putText(crop_img, className+str(dist)+"m", (int(xmin * expected), int(ymin * expected) - 5),cv2.FONT_HERSHEY_COMPLEX, 0.5, (255,255,255))
    cv2.imshow("Color Image",crop_img)
    cv2.waitKey(1)
    # # try:
    #   self.image_pub.publish(cv2_to_imgmsg(crop_img))
    # #   print("success")
    # except:
    #   print("error")

def main(args):
  print("Started image")
  ic = image_converter()
  print("Started image")
  rospy.init_node('image_converter2', anonymous=True)
  try:
    rospy.spin()
  except KeyboardInterrupt:
    print("Shutting down")
  cv2.destroyAllWindows()

if __name__ == '__main__':
    main(sys.argv)
```

MartyG-RealSense, 2023-02-14:
Hi @Gaurav37  As a first step in resolving your issue, it may be helpful to compare your script to the RealSense ROS wrapper's Python example script **show_center_depth.py**, which displays the distance to the coordinate at the center of the image.

https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/scripts/show_center_depth.py

Is this example script able to correctly display the depth if you test it?

Gaurav37, 2023-02-15:
Actually I cannot use this script as I am using ROS melodic and script uses cv_bridge which is only installable in python2 but I need to use cv2.dnn module which uses python3, so I cannot call python2's cv_bridge in same script.
I was anyways able to run my code. The only issue I found affecting (I could be wrong but I think), was that "expected" value in code should be around resolution passed in launch file, otherwise it embeds lots of depth pixel values as zero (in case we stretch depth image alot).

closed 2023-02-15 by Gaurav37