realsenseai/realsense-ros #2628: Resource temporarily unavailable on Jetson Xavier Board

Mardank, 2023-02-15:
Hey together,
I tried to install the realsense-ros wrapper on a Jetson Xavier board. Below you can find some general information about my setup.

- Jetson Xavier
- Ubuntu 18.04
- JetPack 6.1.3
- Realsense camera D435
- Version of LibRealSense v2.50.0
- Version of RealSense ROS v2.3.2
- ROS1 Melodic

I have been following the instructions in the comment of the following [Issue](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-706643114) for installing LibRealSense  and the instructions for ROS1 on how to install RealSenseROS [from source](https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#step-2-install-intel-realsense-ros-from-sources).

In the realsense-viewer the camera works as expected, but in the terminal I get the following Warning:
`15/02 15:14:43,719 WARNING [547764231968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

Additionally after running `roslaunch realsense2_camera rs_camera.launch` it states that the resource is temporarily unavailable. The realsense-viewer is closed and the following output is generated.

Is there anything I can do to counteract against the issue?

Regards, Markus

```
dronelab2@ubuntu:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/dronelab2/.ros/log/5eb372a8-ad3b-11ed-a7db-48b02d5fd446/roslaunch-ubuntu-10780.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:41525/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
[... 103 lines cut ...]
[ INFO] [1676470609.711519462]: Expected frequency for color = 30.00000
[ INFO] [1676470609.807935049]: setupStreams...
 15/02 15:16:50,064 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1676470610.115190997]: SELECTED BASE:Depth, 0
[ INFO] [1676470610.172058693]: RealSense Node Is Up!
[ WARN] [1676470610.184185734]: 
 15/02 15:16:50,185 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:16:50,236 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:16:50,287 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:16:50,530 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:16:50,581 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:16:50,633 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:16:50,684 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:16:50,835 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:16:50,885 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:16:51,037 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:16:51,087 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

MartyG-RealSense, 2023-02-15:
Hi @Mardank  Please first try adding **initial_reset:=true** to your roslaunch instruction to reset the camera at launch in order to see whether this resolves your **control_transfer returned** warnings.

I note in your specification listing at the top of this issue that your JetPack version is listed as 6.1.3.  The newest JetPack version is only 5.1 at the time of writing this.

Mardank, 2023-02-15:
Hi @MartyG-RealSense, I tried to run is with the option ` initial_reset:=true`. Unfortunately I encounter the same Warnings.

Thanks for noticing, I corrected it to the correct version of JetPack.

```
 dronelab2@ubuntu:~$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/dronelab2/.ros/log/1783eba0-ad41-11ed-9f87-48b02d5fd446/roslaunch-ubuntu-11317.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:36067/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 88 lines cut ...]
[ INFO] [1676473073.473375906]: Done Setting Dynamic reconfig parameters.
[ INFO] [1676473073.475086434]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1676473073.476523202]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1676473073.476654242]: setupPublishers...
[ INFO] [1676473073.486309604]: Expected frequency for depth = 30.00000
[ INFO] [1676473073.620920662]: Expected frequency for color = 30.00000
[ INFO] [1676473073.719729476]: setupStreams...
 15/02 15:57:53,977 WARNING [547305017728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1676473074.028812847]: SELECTED BASE:Depth, 0
[ INFO] [1676473074.087513399]: RealSense Node Is Up!
[ WARN] [1676473074.097389561]: 
 15/02 15:57:54,098 WARNING [547305017728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:57:54,149 WARNING [547305017728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:57:54,200 WARNING [547305017728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:57:54,444 WARNING [547305017728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:57:54,495 WARNING [547305017728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 15:57:54,546 WARNING [547305017728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

MartyG-RealSense, 2023-02-15:
The RealSense SDK officially supports JetPack 4.6 when used with Ubuntu 18.04, so there does not seem to be a problem there.  

Your ROS wrapper version 2.50.0 is also correctly matched with ROS wrapper 2.3.2.

Pointcloud and align_depth are set to false, so they will not be causing problems.

Sometimes the **control_transfer returned** warning will repeat continuously and the topics will still be published normally in RViz without apparent negative effects.  Are you able to visualize the depth and color streams normally in RViz despite the warnings, please?

MartyG-RealSense, 2023-02-21:
Hi @Mardank  Do you require further assistance with this case, please?  Thanks!

Mardank, 2023-02-22:
Hey @MartyG-RealSense, yes, most likely I still need further assistance. Unfortunately, I haven't been able to do more testing until now. Sorry for the delay!

MartyG-RealSense, 2023-02-22:
It's no trouble at all.  Please let me know when you are ready to resume the discussion and I will be happy to help. Good luck!

Mardank, 2023-02-28:
Hey, I'm actually able to visualize the stream of the camera in `RViz` and also in `rqt_image_viewer`. So can I assume that it will work or will there be any restrictions to the usage of the ros_realsense ROS wrapper?

[image]

MartyG-RealSense, 2023-02-28:
The RealSense ROS wrapper is basically publishing ROS topics containing camera data that can be viewed in RViz and rqt_image_viewer.  

The wrapper may not support every possible feature that people can think of, but it can publish depth, infrared, color and IMU topics.

Mardank, 2023-02-28:
Okay, thanks for the help!
One more question: As our aim is to build a small car which is navigating autonomously, is there a SLAM project you would recommend to use with the this setup?

MartyG-RealSense, 2023-02-28:
Intel provide a D435i SLAM project for the ROS1 wrapper at the link below.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

This project would not be suitable for the D435 camera model though as it does not have an IMU component.  So that will limit your options in regard to choice of SLAM solutions.  A popular one that can work with D435 and can be used with or without IMU is **ORB_SLAM3**

https://github.com/UZ-SLAMLab/ORB_SLAM3

An example of a discussion about use of D435 with ORB-SLAM3 is at https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/192

Mardank, 2023-03-07:
No, I don't think so. But if I do, I will open a new issue or write here again. Thank you very much for the help!

closed 2023-03-07 by Mardank

MartyG-RealSense, 2023-03-07:
You are very welcome, I am pleased that I could help.  Thanks very much for the update!

zhihou7, 2024-04-20:
Hi, I suffer from a similar issue on Ubuntu 20.04. Could you help me solve it? 

The error message after I add "initial_reset:=true" is: (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0

Btw, in my case, Rivz screen is not started, and the command got stuck in the error message.

Best,

MartyG-RealSense, 2024-04-20:
Hi @zhihou7  Does the launch behave normally in RViz if you do not add initial_reset to the launch instruction?  If there is not a problem with the launch then it is not necessary to reset it.

zhihou7, 2024-04-22:
@MartyG-RealSense, the launch fails in RViz with roslaunch realsense2_camera rs_camera.launch with the similar error.

zhihou7, 2024-04-22:
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xxx:37691/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
[... 64 lines cut ...]
[ INFO] [1713769101.261977942]: RGB Camera was found.
[ INFO] [1713769101.262146130]: Motion Module was found.
[ INFO] [1713769101.262177538]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1713769101.262204203]: num_filters: 0
[ INFO] [1713769101.262219611]: Setting Dynamic reconfig parameters.
[ INFO] [1713769104.205329860]: Done Setting Dynamic reconfig parameters.
[ INFO] [1713769104.205689285]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1713769104.206561420]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1713769104.206585701]: setupPublishers...
[ INFO] [1713769104.207759505]: Expected frequency for depth = 30.00000
[ INFO] [1713769104.221967472]: Expected frequency for color = 30.00000
[ INFO] [1713769104.229695166]: setupStreams...
 22/04 14:58:24,396 WARNING [139959255676672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1713769104.447187652]: SELECTED BASE:Depth, 0
[ INFO] [1713769104.450315077]: RealSense Node Is Up!
[ WARN] [1713769104.516629106]:
 22/04 14:58:24,517 WARNING [139959255676672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/04 14:58:24,567 WARNING [139959255676672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/04 14:58:24,618 WARNING [139959255676672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 22/04 14:58:25,130 WARNING [139959255676672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/04 14:58:25,281 WARNING [139959255676672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/04 14:58:25,331 WARNING [139959255676672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2024-04-22:
@zhihou7  Which methods did you use to install the librealsense SDK and RealSense ROS1 wrapper, please?

One approach is to install librealsense and the ROS1 wrapper separately, with the SDK installed first and the wrapper secondly afterwards.

Another approach is to use the command below to install librealsense and the ROS1 wrapper from packages together at the same time.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

With the apt-get method it is important that librealsense has not already been installed before the command is run, as you could then end up with two conflicting installations of librealsense on the same computer.