realsenseai/realsense-ros #2629: Incorrect parameter name for "ordered_pc" in rs_launch.py on tag 4.51.1

abake48, 2023-02-15:
I know this isn't an issue in the `ros2-development` branch but when using the release tag `4.51.1`, the real sense launch file, `rs_launch.py` [inaccurately defines the parameter for `ordered_pc`.](https://github.com/IntelRealSense/realsense-ros/blob/2a65533ee7431bdc05fe5744798efc7f5713f866/realsense2_camera/launch/rs_launch.py#L50) Changing the parameter name to `pointcloud.ordered_pc` fixed my issues. This is actually noted in the ReadMe but wasn't reflected in the release. 

Since this is the latest stable version, I felt submitting this issue was needed. I apologize if this issue has already been reported.

MartyG-RealSense, 2023-02-15:
Hi @abake48  The parameter is **ordered_pc** in the rs_launch.py file for the **ros2_legacy** wrapper, which is the previous generation of ROS2 wrapper that is now obsolete.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-legacy/realsense2_camera/launch/rs_launch.py#L77

The **ros2-development** branch, which currently has 4.51.1 as the latest version at the time of writing this, correctly has **pointcloud.ordered_pc** as the parameter.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_launch.py#L59

The rs_camera.launch file on the branch that you linked to was last updated 10 months ago, whilst the latest version of the launch file on the ros2-development branch was updated only 2 weeks ago at the time of writing this.

abake48, 2023-02-15:
Hi @MartyG-RealSense, 
The link I provided pointed to the commit instead of the tag but the version of the launch file on that tag doesn't match the latest version. 

https://github.com/IntelRealSense/realsense-ros/blob/4.51.1/realsense2_camera/launch/rs_launch.py#L50

I would like to freeze the version of this package that I am using which is why I'd prefer not to use the **ros2-development**

MartyG-RealSense, 2023-02-15:
Would it be possible to visit the ros2-development branch's GitHub page (which currently has the 4.51.1 version) and create a custom **fork** version of the wrapper that is controlled by yourself?  Then this separate forked version should give you control over whether it updates.

[image]

MartyG-RealSense, 2023-02-21:
Hi @abake48  Do you require further assistance with this case, please?  Thanks!

schornakj, 2023-02-22:
In the version of `rs_launch.py` which is part of the tagged `4.51.1` release (https://github.com/IntelRealSense/realsense-ros/blob/4.51.1/realsense2_camera/launch/rs_launch.py#L50), the previous incorrect `ordered_pc` parameter name is used.

@abake48 and I are using the released version of `realsense-ros` from the ROS 2 buildfarm. While the name of this parameter has been [fixed](https://github.com/IntelRealSense/realsense-ros/commit/ef25e043b3221317353a26322ae4f6bf17a4137b) in the ros2-development branch, this fix was added [after](https://github.com/IntelRealSense/realsense-ros/tree/ef25e043b3221317353a26322ae4f6bf17a4137b/realsense2_camera) `4.51.1` was released, so it isn't present in the latest binary release of these packages.

In other words, the underlying problem here is that the most recent binary release of these packages is very old and contains functional bugs that have already been fixed in the dev branch for a long time. @MartyG-RealSense would it be possible for the project maintainers to put together an updated release to resolve this?

MartyG-RealSense, 2023-02-22:
I will highlight this issue to the RealSense ROS developer team.  Thanks very much for the reports!

MartyG-RealSense, 2023-02-23:
Hi @schornakj  The RealSense ROS developers advised that the rs_launch.py file is not part of the binary and is a separate Python file, so you can edit the launch file and correct the parameter name without the need to rebuild the binary.  You could even write your own custom launch file if you wish to do so.

MartyG-RealSense, 2023-03-01:
Hi @abake48 and @schornakj   Bearing in mind the information in the comment above, do either of you require further assistance with this case please?  Thanks!

closed 2023-03-07 by MartyG-RealSense