realsenseai/realsense-ros #2630: Trouble setting up the project

tomasmendes01, 2023-02-18:
Surely I'm doing something wrong here. 
Can anyone tell me which distribution of Ubuntu should I use as well as the ROS distribution in order to correctly run this project?

tomasmendes01, 2023-02-18:
Surely I'm doing something wrong here.

Can anyone tell me what Ubuntu and ROS distributions should I use in order to correctly run this project?

MartyG-RealSense, 2023-02-18:
Hi @tomasmendes01  The Ubuntu version that you should use will depend on the version of ROS that you are using.  The links below recommend which Ubuntu version to use with which ROS version.

**ROS1**
https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#step-1-install-the-ros-distribution

**ROS2**
https://github.com/IntelRealSense/realsense-ros#step-1-install-the-ros2-distribution

In regard to the RealSense ROS wrapper, a particular wrapper version should be matched as closely as possible with the specific librealsense SDK version listed in that wrapper's release notes.

https://github.com/IntelRealSense/realsense-ros/releases

For example, ROS1 wrapper 2.3.2 should be matched with librealsense 2.50.0 or 2.51.1, whilst ROS2 wrapper 4.51.1 should be matched with librealsense 2.51.1.

tomasmendes01, 2023-02-18:
I'm using 
- Ubuntu Kinetic (22.04)
- ROS2 (humble)
- librealsense (latest version)

I can run `ros2 launch realsense2_camera rs_launch.py` as well as other python scripts and `realsense-viewer`, but I can't run nor find the `opensource_tracking.launch`

MartyG-RealSense, 2023-02-19:
22.04 is called Jammy.  :)

opensource_tracking.launch is only on the ROS1 wrapper, unfortunately.  Of the limited number of available information resources about using SLAM with RealSense and ROS2, they have typically involved using a combination of **slam_toolbox** and **depthimage_to_laserscan**, as described at https://github.com/IntelRealSense/realsense-ros/issues/2387

tomasmendes01, 2023-02-20:
Yes Jammy, my mistake! 

Thank you for your support, I ended up correctly setting up the project :)

closed 2023-02-20 by tomasmendes01