realsenseai/realsense-ros #2631: Method of getting combined IMU and images other than rs-convert and rs_bag2image?

HouChentripleg, 2023-02-20:
I want to collect color camera, acceleration and gyro data using the rs_camera.launch, in which I set **unite_imu_method:=linear_interpolation** for combining acc and gyro together to a topic "/camera/imu" following the guide, later I use "rosbag record" to generate a bag file with topic "**/camera/imu**" and "/camera/color/image_raw". However, there is no topic "**/file_version**", I can not obtain the images with the official tool rs-convert or the rs_bag2image. I wonder is there any other method that I can get both the combined imu data and the images?

HouChentripleg, 2023-02-20:
I'd like to add one more thing, I used the package "**image_view**" which extract .jpg from .bag, but the amount of extracted data is not identical with the bag file. For example, I might get a bag with topic "/camera/color/image_raw" having 545 msgs, but the output of image_view only gets 156 .jpg images, which is very strange.

HouChentripleg, 2023-02-20:
> Hi @HouChentripleg The RealSense SDK (librealsense) uses a bag format that is based on the ROS rosbag standard but has some differences - as described by a RealSense team member in [IntelRealSense/librealsense#3020 (comment)](https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766) - and so a bag file created with rosbag record may not be readable by **librealsense** based tools such as rs-convert and rs_bag2image (which are designed to extraxt from bags recorded in the RealSense SDK and not ROS).
> 
> .If you are able to use the **realsense-viewer** tool then you could enable the color and IMU streams and then click the **Record** button to record the color, gyro and accel data to a bag that can then be used with rs-convert to extract the color and IMU frames.
> 
> In regard to inconsistencies in the number of frames recorded compared to the number of extracted frames, it is not unusual for this to occur.

Thanks for reply:)  As far as I know, the acc and gyro are published in two different topics, how can I combined the two msgs into one synced imu stream in realsense-viewer? Will acc and gyro be out of alignment in timestamp if gyro's sampling frequency is not the integer times of acc's frequency?

MartyG-RealSense, 2023-02-20:
Yes, accel and gyro are separate streams in realsense-viewer and cannot be combined into a single message.  However, they should be well matched as long as you use the recommended speed combinations (the default of 63 accel, 200 gyro or alternatively, 250 accel and 400 gyro).

HouChentripleg, 2023-02-20:
> 

I'm sorry my camera is D455, it seems that the acc's frequency can only be selected either 100Hz or 200Hz.

MartyG-RealSense, 2023-02-20:
Thanks @HouChentripleg  You are correct.  The D455 did originally support 63 / 200 and 250 / 400 but newer models from mid 2022 onwards use 100 / 200.  So just use whatever the default speeds in the Motion Module are without changing them.

HouChentripleg, 2023-02-20:
> Thanks @HouChentripleg You are correct. The D455 did originally support 63 / 200 and 250 / 400 but newer units of D455 manufactured from mid 2022 onwards use 100 / 200 due to a change in the IMU component. So just use whatever the default speeds in the Viewer's Motion Module are without changing them.

I tried 100Hz acc and 200Hz gyro just now, but it is quite odd that the bag file recorded by realsense-viewer includes 1865 acc msgs and 3681 gyro msgs, which is not 2x relationship according to the theory.
Another odd thing is that the output csv file using rs-convert includes only 1863 acc msgs and 3676 gyro msgs. Should I use the linear interpolation to deal with these data in order to keep 100Hz imu output? Or any other suggestions?

HouChentripleg, 2023-02-20:
btw. the actual record time is 18.44s if you might ask

MartyG-RealSense, 2023-02-20:
It is not impossible to extract frames from a rosbag file created with rosbag record but may require more work than the librealsense bag tools.  For example, color frames can be extracted as jpeg images from a rosbag with **extract_images**.

http://wiki.ros.org/rosbag/Tutorials/Exporting%20image%20and%20video%20data#Exporting_jpegs_from_bag_file 

For IMU data you may need to search for an IMU export tool for ROS bag data, such as the one at the link below that can export IMU data to a .txt file.

https://github.com/dvdmc/bag_extractor

If you do use ROS to record the bag, the IMU data may be more stable if you set unite_imu_method to **copy** instead of linear_interpolation.

HouChentripleg, 2023-02-21:
I solved my problem finally! 
Considering the whole tedious work, I decided to write a script to directly extract imu and image topic from the bag file generated by ROS. That's more efficient than using a lot of other packages.

Another thing worth to be mentioned is that the image_view I have tried is quite odd if using launch file, later I used **rosrun image_view extract_images _sec_per_frame:=0.0005 image:=/camera/color/image_raw** in which the **_sec_per_frame** can control the output number of images, but it's still looks very strange because the real frequency is 30Hz. Anyway, it works by adjusting this parameter. Btw. when using image_view to extract imgs from bag, we should turn off the camera, otherwise it will record both the bag imgs and the current imgs, that makes our output out of sequence like my situation.

Thanks for your help!

MartyG-RealSense, 2023-02-21:
You are very welcome, @HouChentripleg - it's excellent to hear that you achieved a solution.  Thanks so much for the update and for sharing the details of your solution with the RealSense ROS community!

closed 2023-02-26 by MartyG-RealSense