realsenseai/realsense-ros #2633: There may be some opencv version problems

987428377, 2023-02-22:
device;：D435i
Ubuntu18.04 ros-melodic
opencv : 3.2.0
When I use a stereo camera：

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="true"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
  <arg name="infra_rgb"           default="false"/>

The terminal displays the following error and exits：

`[ WARN] [1677087254.760677489]: Failed to set option: white_balance: xioctl(VIDIOC_G_CTRL) failed Last Error: No such device
[ INFO] [1677087254.767096539]: Done Setting Dynamic reconfig parameters.
[ INFO] [1677087254.767327143]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1677087254.767445851]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1677087254.767757091]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
OpenCV Error: Assertion failed (s >= 0) in setSize, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp, line 307
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/`core`/src/matrix.cpp:307: error: (-215) s >= 0 in function setSize

[camera/realsense2_camera_manager-1] process has died [pid 8427, exit code -6, cmd  .......................................`

Can you tell me how to solve the problem?

Thank you very much ！

987428377, 2023-02-22:
I have made some attempts. When<arg name="align_depth" default="true"/>, problems like the above will occur and cannot be used with<arg name="enable_depth" default="true"/>！

MartyG-RealSense, 2023-02-23:
Hi, can you provide me with the following information please?

1.  Which Linux kernel version are you using?

2.  Which versions of the librealsense SDK and RealSense ROS wrapper are you using?

987428377, 2023-02-25:
Linux kernel version : Linux 5.4.0-139-generic x86_64
 librealsense SDK version : 2.53.1
 RealSense ROS wrapper :2.3.1

Thank you for your reply！
I am trying to add the configuration of roslaunch one by one

MartyG-RealSense, 2023-02-25:
It is important to match the ROS wrapper version as closely as possible with the librealsense SDK version listed in that particular wrapper's release notes.  Wrapper 2.3.1 should be used with the old librealsense version **2.48.0**.

The nearest available match to librealsense 2.53.1 would be wrapper **2.3.2**, which works with SDK 2.50.0 and 2.51.1.   I do not know if it will work with 2.53.1.  The ROS1 wrapper has ceased development and is now called **ros1-legacy** so there will be no further ROS1 wrapper versions for newer librealsense versions that are released.  The focus of the RealSense ROS developer team is now on the ROS2 wrapper **ros2-development**.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2

987428377, 2023-03-05:
> Do you require further assistance with this case, please? Thanks!

Sorry, it doesn't work.
  The launch file is as follows：
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="-1"/>
  <arg name="depth_height"        default="-1"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="true"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
  <arg name="infra_rgb"           default="true"/>

  <arg name="color_width"         default="-1"/>
  <arg name="color_height"        default="-1"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="-1"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="-1"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="wait_for_device_timeout"   default="-1.0"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
[... 302 lines cut ...]
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/color/metadata
/camera/gyro/imu_info
/camera/gyro/metadata
/camera/gyro/sample
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/pointcloud/parameter_descriptions
/camera/pointcloud/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

Thank you for your reply！

MartyG-RealSense, 2023-03-05:
If you are using SDK 2.53.1 then the firmware version should be **5.14.0.0**.  Is this firmware version installed in your camera, please?

987428377, 2023-03-06:
Thank you very much for your reply. Do you have any documentation so that I can check the correct match between different versions

MartyG-RealSense, 2023-03-06:
The recommended firmware for each SDK version is listed in the 'Recommended Firmware' section of the **Release Notes** for each SDK version.

https://github.com/IntelRealSense/librealsense/releases

The release notes for the RealSense ROS wrapper list a recommended SDK version to match with a particular wrapper version in the 'Supported RealSense SDK' section of the notes.

https://github.com/IntelRealSense/realsense-ros/releases

987428377, 2023-03-08:
Thank you for your patience. I don't think I will bother you with this question in the future！

closed 2023-03-08 by 987428377

MartyG-RealSense, 2023-03-08:
It's no trouble at all, I assure you.  Please do feel free to create further issues in future if you need to  :)