realsenseai/realsense-ros #2634: roslaunch realsense2_camera opensource_tracking.launch

EduardoQD, 2023-02-23:
Hello,

I'm trying to run this command in ROS2, with ros2 run realsense2_camera opensource_tracking.launch. But I didn't find this script in the file even in the folders to ROS1.

I would like to run the VSLAM in ROS2 if it's possible.

MartyG-RealSense, 2023-02-23:
Hi @EduardoQD  opensource_tracking.launch is only on the ROS1 wrapper, unfortunately.  Of the limited number of available information resources about using SLAM with RealSense and ROS2, they have typically involved using a combination of slam_toolbox and depthimage_to_laserscan, as described at https://github.com/IntelRealSense/realsense-ros/issues/2387

In the ROS1 wrapper, opensource_tracking.launch is on the **ros1-legacy** branch of the wrapper at the folder location linked to below.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy/realsense2_camera/launch

EduardoQD, 2023-02-23:
So, are you advise me to install ros1 instead ros2 if i want to use the VSLAM?

MartyG-RealSense, 2023-02-23:
The ROS1 wrapper will be an easier method of implementing SLAM with RealSense than the ROS2 wrapper.  Intel have a ROS1 SLAM guide that makes use of opensource_tracking.launch:

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

MartyG-RealSense, 2023-03-05:
Hi @EduardoQD  Do you require further assistance with this case, please?  Thanks!

EduardoQD, 2023-03-17:
Yes, I need help because I want to run the VSLAM in my robot

EduardoQD, 2023-03-17:
I would like to use my d435i with slam_toolbox, firstly to create a map and after that to navigate with my robot. I'm using ROS2 foxy.
Is it possible that idea?

MartyG-RealSense, 2023-03-17:
For a ROS2 SLAM solution, RealSense users of ROS2 have typically combined slam_toolbox with **depthimage_to_laserscan**.  

https://github.com/ros-perception/depthimage_to_laserscan

https://github.com/IntelRealSense/realsense-ros/issues/2387 is an example of doing this with Foxy.

EduardoQD, 2023-03-20:
But in this example, how can I run this script to convert image to laser message?

MartyG-RealSense, 2023-03-20:
If you are using Foxy, does the commands below to install the **foxy_devel** branch of depthimage_to_laserscan work?

```
source /opt/robot_devkit/robot_devkit_setup.bash
mkdir -p ~/ros2_ws/src && cd ~/ros2_ws/src
git clone https://github.com/ros-perception/depthimage_to_laserscan -b foxy-devel
cd .. && colcon build
source ~/ros2_ws/install/local_setup.bash
```

These instructions are based on ones at the guide below.

https://intel.github.io/robot_devkit_doc/pages/rs_slam.html#slam-with-realsense

EduardoQD, 2023-03-22:
I solved it with sudo apt get install ros-foxy-depthimage-to-laserscan.

But now I have this problem, 
Because it seems that the commando that you send me it worked good

Now I have this graph on ros2

[image]

But I have this problem with the map topic:

[image]
eduardo@eduardo-CM11EBV716W:~/ros2_ws$ ros2 launch depthimage_to_laserscan depthimage_to_laserscan-launch.py  

[INFO] [launch]: All log files can be found below /home/eduardo/.ros/log/2023-03-21-12-23-46-115488-eduardo-CM11EBV716W-18344 

[INFO] [launch]: Default logging verbosity is set to INFO 

/opt/ros/foxy/share/depthimage_to_laserscan/launch/depthimage_to_laserscan-launch.py:27: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead 

  Node( 

/opt/ros/foxy/share/depthimage_to_laserscan/launch/depthimage_to_laserscan-launch.py:27: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead 

  Node( 

[INFO] [depthimage_to_laserscan_node-1]: process started with pid [18346]

MartyG-RealSense, 2023-03-22:
A warning that something is 'deprecated' often means that it is supported and still works but may not be supported in the future so an alternative is recommended.  It should still work with Foxy though despite the warning messages.

It looks as though the warnings are suggesting to edit the depthimage_to_laserscan-launch.py launch file to change 'node_executable' to 'executable' and 'node_name' to 'name'.

https://github.com/ros-perception/depthimage_to_laserscan/blob/foxy-devel/launch/depthimage_to_laserscan-launch.py#L29-L30

EduardoQD, 2023-03-22:
Hello @MartyG-RealSense ,

I've solved this problem, but now I have a problem with the tree, because I have some warnings in rviz2.

[image]

The tree is the following:
[image]

How can I change it? 
Because the odom needs the data from the camera.

MartyG-RealSense, 2023-03-22:
If you have no odom, you may need to enable the IMU topics accel and gyro in your launch, as they are disabled by default in the RealSense ROS wrapper.  You can do so in the ROS2 4.51.1 wrapper by adding the commands below onto the end of the launch instruction.

`enable_gyro:=true enable_accel:=true unite_imu_method:=2`

If the older ROS2 wrapper **ros2-legacy** is being used then use tis instruction instead:

`enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation`

EduardoQD, 2023-03-22:
I've launched with these options:
ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=2 

I've gotten these logs from the camera node:
  
eduardo@eduardo-CM11EBV716W:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=2 

```
[INFO] [launch]: All log files can be found below /home/eduardo/.ros/log/2023-03-22-10-28-28-084283-eduardo-CM11EBV716W-10548 

[INFO] [launch]: Default logging verbosity is set to INFO 

[INFO] [realsense2_camera_node-1]: process started with pid [10550] 

[realsense2_camera_node-1] [INFO] [1679477308.290963574] [camera.camera]: RealSense ROS v4.51.1 

[realsense2_camera_node-1] [INFO] [1679477308.291147131] [camera.camera]: Built with LibRealSense v2.53.1 

[realsense2_camera_node-1] [INFO] [1679477308.291170063] [camera.camera]: Running with LibRealSense v2.53.1 

[realsense2_camera_node-1] [INFO] [1679477308.303011818] [camera.camera]: Device with serial number 838412070540 was found. 

[realsense2_camera_node-1]  

[realsense2_camera_node-1] [INFO] [1679477308.303083913] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 was found. 

[realsense2_camera_node-1] [INFO] [1679477308.303098752] [camera.camera]: Device with name Intel RealSense D435 was found. 

[realsense2_camera_node-1] [INFO] [1679477308.303369759] [camera.camera]: Device with port number 2-1 was found. 

[realsense2_camera_node-1] [INFO] [1679477308.303386486] [camera.camera]: Device USB type: 3.2 

[realsense2_camera_node-1] [INFO] [1679477308.304888914] [camera.camera]: getParameters... 

[realsense2_camera_node-1] [INFO] [1679477308.307018235] [camera.camera]: JSON file is not provided 

[realsense2_camera_node-1] [INFO] [1679477308.307065458] [camera.camera]: Device Name: Intel RealSense D435 

[realsense2_camera_node-1] [INFO] [1679477308.307083735] [camera.camera]: Device Serial No: 838412070540 

[realsense2_camera_node-1] [INFO] [1679477308.307097642] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 

[realsense2_camera_node-1] [INFO] [1679477308.307112222] [camera.camera]: Device FW version: 05.14.00.00 

[realsense2_camera_node-1] [INFO] [1679477308.307125355] [camera.camera]: Device Product ID: 0x0B07 

[realsense2_camera_node-1] [INFO] [1679477308.307137826] [camera.camera]: Sync Mode: Off 

[realsense2_camera_node-1] [INFO] [1679477308.320069158] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30 

[realsense2_camera_node-1] [WARN] [1679477308.321444726] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range. 

[realsense2_camera_node-1]  22/03 10:28:28,321 WARNING [140516462909184] (backend-v4l2.cpp:1780) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16! 

[realsense2_camera_node-1] [INFO] [1679477308.327396406] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x30 

[realsense2_camera_node-1] [INFO] [1679477308.333618325] [camera.camera]: Stopping Sensor: Stereo Module 

[realsense2_camera_node-1] [INFO] [1679477308.337900121] [camera.camera]: Starting Sensor: Stereo Module 

[realsense2_camera_node-1] [INFO] [1679477308.346591964] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30 

[realsense2_camera_node-1] [INFO] [1679477308.346764059] [camera.camera]: Stopping Sensor: RGB Camera 

[realsense2_camera_node-1] [INFO] [1679477308.357487949] [camera.camera]: Starting Sensor: RGB Camera 

[realsense2_camera_node-1] [INFO] [1679477308.359239306] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30 

[realsense2_camera_node-1] [INFO] [1679477308.359642728] [camera.camera]: RealSense Node Is Up! 

[realsense2_camera_node-1] [WARN] [1679477308.476263814] [camera.camera]: 
```

 
I still get this graph from the tf2:

[image]

EduardoQD, 2023-03-22:
Thanks for the information, but i think that the IMU  hasn't started because if I see the topics that I have: 

```
/camera/color/camera_info 

/camera/color/image_raw 

/camera/color/image_raw/compressed 

/camera/color/image_raw/compressedDepth 

/camera/color/metadata 

/camera/depth/camera_info 

/camera/depth/color/points 

/camera/depth/image_rect_raw 

/camera/depth/image_rect_raw/compressed 

/camera/depth/image_rect_raw/compressedDepth 

/camera/depth/metadata 

/camera/extrinsics/depth_to_color 

/camera/imu 

/clock 

/map 

/map_metadata 

/parameter_events 

/rosout 

/scan 

/slam_toolbox/feedback 

/slam_toolbox/graph_visualization 

/slam_toolbox/scan_visualization 

/slam_toolbox/update 

/tf 

/tf_static 
```

I miss these ones:
```
/camera/accel/imu_info
/camera/accel/metadata
/camera/accel/sample
/camera/extrinsics/depth_to_accel
/camera/extrinsics/depth_to_gyro
/camera/gyro/imu_info
/camera/gyro/metadata
/camera/gyro/sample
```

MartyG-RealSense, 2023-03-22:
Looking at your launch log again, it does seem to be ignoring the IMU and just enabling depth and color.  It should say something like this in the 4.51.1 wrapper when IMU is enabled.

```
Starting Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1670472578.175258668] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 100
[realsense2_camera_node-1] [INFO] [1670472578.175413741] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
```

Are you able to access the RealSense Viewer tool and check in that whether the Motion Module is able to work?

MartyG-RealSense, 2023-03-22:
If IMU is supported then expanding open the **Stereo Module** category in the options side-panel should show a **Motion Module** section.

Can you also confirm that your camera model is a D435i please, as a D415 or D435 / D435f does not have an IMU.

EduardoQD, 2023-03-22:
I have this version:
[image]
I think that its the d435
So I cant' use it forr vslam rigth ?

MartyG-RealSense, 2023-03-22:
Correct, the D435 is not equipped with a Motion Module.  A D435i or D455 will be required.

MartyG-RealSense, 2023-03-22:
Though it is possible to use an external IMU with ROS.

EduardoQD, 2023-03-22:
So I have to get one of these.
or I see in some repositories that use the VSLAM with tow camera, one d435 and the other T265

EduardoQD, 2023-03-22:
How can I use an external IMU with ROS2?

MartyG-RealSense, 2023-03-22:
You could use a T265 with the RealSense ROS wrapper.  The T265 is still available but it is a retired camera model and so is only available whilst remaining stocks last.  The ROS2 wrapper front page also has a note about removing T265.  So an external IMU may be a safer choice.

https://store.intelrealsense.com/buy-intel-realsense-tracking-camera-t265.html

My expectation is that an external IMU's data would be published on an **/imu/data** topic and the topic would be subscribed to in order to access its data.  External IMUs are not often discussed on this RealSense ROs forum though.

MartyG-RealSense, 2023-03-22:
If you need Visual SLAM then the T265 may be worth considering as it is designed exactly for that application.

https://dev.intelrealsense.com/docs/intel-realsensetm-visual-slam-and-the-t265-tracking-camera

As mentioned above, a disadvantage may be that you have to keep using the same SDK and ROS wrapper version permanently if support for T265 is removed in a future SDK.

EduardoQD, 2023-03-23:
Thank you for everything, but I think that the best way is to change other algorithm for instace ORBSLAM
What do you reckon?

MartyG-RealSense, 2023-03-23:
The link below of an ORB-SLAM2 session with a RealSense D415 (no IMU) should help you to decide whether ORB-SLAM on D435 can meet your SLAM reuirements.

https://www.youtube.com/watch?v=yl6MgnYNDE8

EduardoQD, 2023-03-29:
Hello @MartyG-RealSense ,
I'm trying to install ORBSLAM but I'm having  a lot of problems in the instalation. i don't know why because I'm following the steps of the repository.
https://github.com/Mauhing/ORB_SLAM3/blob/master/README.md

MartyG-RealSense, 2023-03-29:
It appears that the version of ORB-SLAM3 in that link is a copy that was 'forked' from the original repository, which has been updated more recently than the fork (2 years ago instead of 3 years).  So it may be worth attempting to install the original version using its instructions.

https://github.com/UZ-SLAMLab/ORB_SLAM3

MartyG-RealSense, 2023-03-30:
The repository's branch name is Master, indicating that it is the current version.

EduardoQD, 2023-04-14:
I' ve just installed the orbslam3 with out ros and now I have to launch with my camera.
Do you know any tutoial about that ?

MartyG-RealSense, 2023-04-15:
Hi @EduardoQD  ORB-SLAM3 has some RealSense examples for the D435i camera model in its **Examples** folder.

https://github.com/UZ-SLAMLab/ORB_SLAM3#4-running-orb-slam3-with-your-camera

EduardoQD, 2023-04-17:
Hi, @MartyG-RealSense  I'll check it and I'll tell you the news.

MartyG-RealSense, 2023-04-17:
Thanks very much.  Good luck!