realsenseai/realsense-ros #2636: 405 urdf and mesh

chavasse, 2023-02-28:
Currently, the library supports the D405 in ROS 2, however, there isn't a correct urdf model or mesh.

Are there plans to add support for this in the near future with all of the correct transforms?

Thanks,

MartyG-RealSense, 2023-02-28:
Hi @chavasse  A D405 URDF is planned by the RealSense ROS developer team for implementation in a future release of the ROS wrapper.  I do not have a time estimate that I can provide, unfortunately.

MartyG-RealSense, 2023-03-07:
Hi @chavasse  Bearing in mind the comment above, do you require further assistance with this case please?  Thanks!

chavasse, 2023-03-07:
Ideally, we would have an accurate TF model for the camera mounting to the camera frame included with the package as none of the others that we are forced to use are correct.

This issue is actually a duplicate of this issue, apologies, I did search before but didn't find it previously:
https://github.com/IntelRealSense/realsense-ros/issues/2589

MartyG-RealSense, 2023-03-07:
No need for apologies  :)   If you leave a comment on https://github.com/IntelRealSense/realsense-ros/issues/2589 then you will be notified of any updates about a D405 URDF that are posted there.

closed 2023-03-21 by MartyG-RealSense