realsenseai/realsense-ros #2637: Static transforms are inverted

MasonLilly, 2023-02-28:
We recently updated our system from using the ROS 1 realsense node (commit `44a703ed`) to the ROS 2 version (commit `839d8378`). Since the upgrade, we've found that the static transforms published by the camera node have their translations inverted compared to the old version.

From a backup still running ROS 1:
```bash
root# rosrun tf tf_echo camera_front_link camera_front_color_frame
At time 0.000
- Translation: [-0.000, -0.059, 0.000]
- Rotation: in Quaternion [-0.000, -0.002, -0.001, 1.000]
            in RPY (radian) [-0.000, -0.005, -0.001]
            in RPY (degree) [-0.026, -0.259, -0.073]
root# rosrun tf tf_echo camera_front_color_frame camera_front_color_optical_frame

At time 0.000
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [-0.500, 0.500, -0.500, 0.500]
            in RPY (radian) [-1.571, -0.000, -1.571]
            in RPY (degree) [-90.000, -0.000, -90.000]
root# rosrun tf tf_echo camera_front_depth_frame camera_front_depth_optical_frame

At time 0.000
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [-0.500, 0.500, -0.500, 0.500]
            in RPY (radian) [-1.571, -0.000, -1.571]
            in RPY (degree) [-90.000, -0.000, -90.000]
```

From the new system using ROS 2:
```bash
~$ ros2 run tf2_ros tf2_echo camera_front_link camera_front_color_frame
At time 0.0
- Translation: [0.000, 0.059, 0.000]
- Rotation: in Quaternion [0.000, 0.002, -0.000, 1.000]
~$ ros2 run tf2_ros tf2_echo camera_front_color_frame camera_front_color_optical_frame
At time 0.0
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [-0.500, 0.500, -0.500, 0.500]
~$ ros2 run tf2_ros tf2_echo camera_front_depth_frame camera_front_depth_optical_frame
At time 0.0
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [-0.500, 0.500, -0.500, 0.500]
```

This is reflected in a change in where the optical frames of the camera lenses land in our overall TF tree relative to the root of the robot (that is, it isn't a trick of perspective). We've tested this with multiple cameras running different firmware versions and found the behavior to be consistent. So far all tests have been on arm64 host machines. This may be related to or the same as #2583 and #2500.

Because a lot of our system derives navigation info from how these transforms fit into the overall tree, this has turned into a substantial blocker for us. We're currently planning to mitigate by turning off the static TFs entirely and publishing the correct (nominal) ones ourselves as they're defined in the URDF file, but we'd definitely rather use the calibrated extrinsics. Please let me know if there's any other information I can provide or steps we can try to further isolate the problem.

MasonLilly, 2023-02-28:
Worth adding: We're using a D455

MartyG-RealSense, 2023-02-28:
Hi @MasonLilly  As mentioned in https://github.com/IntelRealSense/realsense-ros/issues/2500 the RealSense ROS developer team are already aware of this issue and have created an internal report to investigate it.  

There is not additional information that I can provide about the issue unfortunately, but I will add your case to the internal report in case the issue's priority can be raised.

MartyG-RealSense, 2023-03-07:
As this issue is related to https://github.com/IntelRealSense/realsense-ros/issues/2500 i have added an **Enhancement** label to it to keep it open.  There is no further action that can be provided at this time, though Intel's internal report about the static transform inversion remains ongoing.

SamerKhshiboun, 2023-04-06:
Hi @MasonLilly,
I've opened a PR #2684 , you are welcome to review and comment there.

MartyG-RealSense, 2023-04-12:
A fix has been merged into the RealSense ROS wrapper - see https://github.com/IntelRealSense/realsense-ros/pull/2684 - and so this case can now be closed.

closed 2023-04-12 by MartyG-RealSense