realsenseai/realsense-ros #2638: Realsense ROS on L515 No Default Profile

husseinalijaafar, 2023-03-01:
Hi

I am trying to run 
```
ros2 launch realsense2_camera rs_launch.py
```

On ROS2 Foxy installed from binaries with an Intel Realsense L515. I have updated the firmwrae, and I am still finding some issues, specifically no default profile found. 

```
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-03-01-12-01-05-416829-ubuntu-2019
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [2026]
[realsense2_camera_node-1] [INFO] [1677690067.635629354] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1677690067.636172948] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1677690067.636316334] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1677690068.060025027] [camera.camera]: Device with serial number f0232820 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1677690068.060329818] [camera.camera]: Device with physical ID /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1677690068.060434870] [camera.camera]: Device with name Intel RealSense L515 was found.
[realsense2_camera_node-1] [INFO] [1677690068.062776277] [camera.camera]: Device with port number 1-1.1.3 was found.
[realsense2_camera_node-1] [INFO] [1677690068.063090443] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1677690068.063226850] [camera.camera]: Device f0232820 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1677690068.074960860] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1677690068.076649349] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1677690068.076944401] [camera.camera]: Device Name: Intel RealSense L515
[realsense2_camera_node-1] [INFO] [1677690068.077055599] [camera.camera]: Device Serial No: f0232820
[realsense2_camera_node-1] [INFO] [1677690068.077136954] [camera.camera]: Device physical port: /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1677690068.077219818] [camera.camera]: Device FW version: 01.05.08.01
[realsense2_camera_node-1] [INFO] [1677690068.077299974] [camera.camera]: Device Product ID: 0x0B64
[realsense2_camera_node-1] [INFO] [1677690068.077374818] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1677690068.109864297] [camera.camera]: Set ROS param depth_module.profile to default: 320x240x30
[realsense2_camera_node-1] [ERROR] [1677690068.128956745] [camera.camera]: Error starting device: No default profile found
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[realsense2_camera_node-1] [INFO] [1677690090.130131514] [rclcpp]: signal_handler(signal_value=2)
[INFO] [realsense2_camera_node-1]: process has finished cleanly [pid 2026]
```

I have tried:
- adding initial reset
- swapping the usb c cable around

Note, that I am trying to run this on a RPi 3 (usb 2.0). 

Thanks

MartyG-RealSense, 2023-03-01:
Hi @husseinalijaafar  If a default profile is not detected for some reason, we can try defining a custom profile to replace it.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x30 rgb_camera.profile:=640x480x30`

husseinalijaafar, 2023-03-01:
Hi Marty,

Good hearing from you again.

It forces it back to 320x240..It reminds me of  a usb 2.0 issue?

```
ubuntu@ubuntu:~$ ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x30 rgb_camera.profile:=640x480x30
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-03-01-13-36-14-433964-ubuntu-1961
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [1963]
[realsense2_camera_node-1] [INFO] [1677695779.970882089] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1677695779.977007126] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1677695779.977174053] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1677695780.407405401] [camera.camera]: Device with serial number f0232820 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1677695780.407748369] [camera.camera]: Device with physical ID /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1677695780.407869931] [camera.camera]: Device with name Intel RealSense L515 was found.
[realsense2_camera_node-1] [INFO] [1677695780.409982166] [camera.camera]: Device with port number 1-1.2 was found.
[realsense2_camera_node-1] [INFO] [1677695780.410248207] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1677695780.410388832] [camera.camera]: Device f0232820 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1677695780.448448014] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1677695780.450891446] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1677695780.451156133] [camera.camera]: Device Name: Intel RealSense L515
[realsense2_camera_node-1] [INFO] [1677695780.451332018] [camera.camera]: Device Serial No: f0232820
[realsense2_camera_node-1] [INFO] [1677695780.451460143] [camera.camera]: Device physical port: /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1677695780.451768475] [camera.camera]: Device FW version: 01.05.08.01
[realsense2_camera_node-1] [INFO] [1677695780.451904100] [camera.camera]: Device Product ID: 0x0B64
[realsense2_camera_node-1] [INFO] [1677695780.452038996] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [ERROR] [1677695780.489693908] [camera.camera]: Given value, 640x480x30 is invalid. Set ROS param back to: 320x240x30
[realsense2_camera_node-1] [ERROR] [1677695780.519368788] [camera.camera]: Error starting device: No default profile found
```

MartyG-RealSense, 2023-03-01:
It may be using USB 2 because of the USB-C cable orientation. Please reverse the cable again and then repeat the custom profile launch.

husseinalijaafar, 2023-03-01:
Hi Marty

Unfortunately that has no effect; the port is actually USB 2 though. It's the RPi 3, with no USB 3 ports.

MartyG-RealSense, 2023-03-01:
Are you able to use the RealSense Viewer tool?  If you look at the selectable resolutions for depth and RGB in the Viewer then that will tell you which resolutions and FPS speeds you are able to set in the custom launch in ROS.

husseinalijaafar, 2023-03-01:
Unfortunately I can't build the viewer tool on the RPi 3, it doesn't have enough memory to compile the sdk.
Is there a binary for the viewer tool?

MartyG-RealSense, 2023-03-01:
There is not a separate binary for the Viewer on Linux.  You may be able to create **virtual memory** on your Pi by increasing the **swapfile** size though.

https://pimylifeup.com/raspberry-pi-swap-file/

Please try the launch instruction below to see if it will work on USB2.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=320x240x30 rgb_camera.profile:=1920x1080x6`

Or depth on its own.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=320x240x30`

MartyG-RealSense, 2023-03-11:
Is the camera still undetectable if you turn the USB-C cable back in the other direction (I know that you tried reversing the micro-sized end of the cable when you were using the camera with Pi).

husseinalijaafar, 2023-03-11:
Yes I swapped it both ways, and still no result.

MartyG-RealSense, 2023-03-11:
On Jetson Nano specifically, Intel strongly recommend enabling the **barrel jack** power connector using the instructions at the link below if your model of Nano board has a barrel jack.

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

husseinalijaafar, 2023-03-13:
Hii Marty,

After trying that, I still get the same issue... No Realsense devices were found! Even when I swap the L515 for a D435i, it is still the same problem.

However, with the D435i I get a new line about ds5 groups

```
realsense2_camera_node-1] [INFO] [1678720715.397363516] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1678720715.397618112] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1678720715.397680823] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [WARN] [1678720715.413303871] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1]  13/03 11:18:35,413 WARNING [548222822816] (ds5-factory.cpp:1152) DS5 group_devices is empty.
[realsense2_camera_node-1] [WARN] [1678720721.519600801] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1]  13/03 11:18:41,519 WARNING [548222822816] (ds5-factory.cpp:1152) DS5 group_devices is empty.
[realsense2_camera_node-1] [WARN] [1678720727.623403358] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1]  13/03 11:18:47,623 WARNING [548222822816] (ds5-factory.cpp:1152) DS5 group_devices is empty.

```

Could it be a permission issue with udev and the usb ports? I tried copying the `99-realsense-libusb.rules` to `rules.d` but still same issue. I installed with `sudo apt install ros-foxy-realsense2-camera`

MartyG-RealSense, 2023-03-13:
This log does indicate that there is a general camera detection problem, possibly related to the USB port or the udev rules (which handle device access), and that the problem is not with a particular camera (since neither L515 or D435i are detected).

Are you using the cameras with a computer using a Linux operating system such as Ubuntu or on a Windows computer with 'WSL2' Linux?

Are you using Ubuntu 20.04, as this is required for ROS2 Foxy.

[image]

If you are using Ubuntu 20.04, does it make a difference if you remove all RealSense related packages from the computer with the command below and then repeat **sudo apt install ros-foxy-realsense2-camera** to install the librealsense SDK and ROS wrapper again?

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

husseinalijaafar, 2023-03-13:
I'm using a custom image of Xubuntu 20.04 from the Jetson [forums](https://forums.developer.nvidia.com/t/xubuntu-20-04-focal-fossa-l4t-r32-3-1-custom-image-for-the-jetson-nano/121768). 

 I tried removing all realsense packages and running `sudo apt install ros-foxy-realsense2-camera` but I am still getting the same issue. 
 
Can I view any device access statuses?

MartyG-RealSense, 2023-03-13:
Is the camera detected if you use the **lsusb** Ubuntu command?

If you are using Ubuntu 20.04 with Jetson then you should be using JetPack 5, as earlier JetPacks such as 4.6 are for Ubuntu 18.04.  JetPack 5 is the first one that supports 20.04.

husseinalijaafar, 2023-03-13:
Hi Marty

I have solved the issue by using the ISO provided by QEngineering [here](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image) .

Thank you for your help.

closed 2023-03-13 by husseinalijaafar

MartyG-RealSense, 2023-03-13:
That's great news!  Thanks very much for the update and for sharing the QEngineering link!