realsenseai/realsense-ros #2642: Odemetry in Realsense

shivangi-futuristicbots, 2023-03-04:
@MartyG-RealSense ,Can you please help to defined odemetry parameter and use the odom_frame_id and other that give odemetry topic

i am using the ros2-developement branch **https://github.com/IntelRealSense/realsense-ros/**

MartyG-RealSense, 2023-03-05:
Hi @shivangi-futuristicbots  Could you provide further information about your problem, please.  Is it that you are receiving no odometry?

If you have you odometry, have you enabled the IMU topics as they are disabled by default in the ROS wrapper.  They can be enabled by adding the following instructions to the launch instruction.

**enable_accel:=true enable_gyro:=true unite_imu_method:=2**

For example:

`ros2 launch realsense2_camera rs_launch.py enable_accel:=true enable_gyro:=true unite_imu_method:=2`

shivangi-futuristicbots, 2023-03-13:
when i create the subscriber for this i see the pattern of publish the topic imu output it would be like for the linear acceleartaion and angular velocity, it is give twice value : 

**camera: geometry_msgs.msg.Vector3(x=-0.031415924429893494, y=-0.003490658476948738, z=-0.005235987715423107)
camera: geometry_msgs.msg.Vector3(x=-0.005235987715423107, y=-0.003490658476948738, z=-0.008726646192371845)**  

for the better assessment the code is here:

**self.camera = self.create_subscription(Imu, '/camera/imu', self.camera_imu_data,qos_profile=10)
def camera_imu_data(self, msg: Imu):
        self.camera_imu = msg
        lin_acc = math.sqrt((msg.linear_acceleration.x**2)+(msg.linear_acceleration.y**2)+(msg.linear_acceleration.z**2))
        avg_vel  = math.sqrt((msg.angular_velocity.x ** 2) + (msg.angular_velocity.y ** 2) + (msg.angular_velocity.z ** 2))
        camera_data['linear_acceleration'].append(lin_acc)
        camera_data['angular_velocity'].append(avg_vel)
        
        print("camera:", msg.angular_velocity)**                                                                                                         

why is that?

MartyG-RealSense, 2023-03-13:
Hi @shivangi-futuristicbots  The **msg** terms do not usually have a '2' on the end of them.

```
lin_acc = math.sqrt((msg.linear_acceleration.x)+(msg.linear_acceleration.y)+(msg.linear_acceleration.z))
avg_vel = math.sqrt((msg.angular_velocity.x) + (msg.angular_velocity.y) + (msg.angular_velocity.z))
```

For example in **ros_reader.cpp**:

https://github.com/IntelRealSense/librealsense/blob/master/src/media/ros/ros_reader.cpp#L501-L503

shivangi-futuristicbots, 2023-03-14:
Actually i am comparing the camera/imu angular velocity with other device but i am getting different output for the same place. The code i provided is part of the /camera/imu comparison with other imu data set which running simultaneously.

 **def bopt_data_imu(self, msg: Imu):
        self.imu_data = msg
        lin_acc = math.sqrt((msg.linear_acceleration.x**2)+(msg.linear_acceleration.y**2)+(msg.linear_acceleration.z**2))
        # avg_vel  = math.sqrt((msg.angular_velocity.x ** 2) + (msg.angular_velocity.y ** 2) + (msg.angular_velocity.z ** 2))
        avg_vel = msg.angular_velocity.x + msg.angular_velocity.y + msg.angular_velocity.z
        imu_data['linear_acceleration'].append(lin_acc)
        imu_data['angular_velocity'].append(avg_vel)
        
        # print("bopt:", msg.angular_velocity)
        
    def camera_imu_data(self, msg: Imu):
        self.camera_imu = msg
        lin_acc = math.sqrt((msg.linear_acceleration.x**2)+(msg.linear_acceleration.y**2)+(msg.linear_acceleration.z**2))
        # avg_vel  = math.sqrt((msg.angular_velocity.x ** 2) + (msg.angular_velocity.y ** 2) + (msg.angular_velocity.z ** 2))
        avg_vel =  msg.angular_velocity.x + msg.angular_velocity.y + msg.angular_velocity.z
        camera_data['linear_acceleration'].append(lin_acc)
        camera_data['angular_velocity'].append(avg_vel)
        
        # print("camera:", msg.angular_velocity)**

i am getting the camera msg printing twice at once why

MartyG-RealSense, 2023-03-14:
Aside from the two different def names (bopt_data_imu and camera_imu_data) the two blocks of code are identical.  That makes me wonder if both sets of instructions are printing the data from only one source (the camera or the other IMU data set) instead of printing results from two different sources.

closed 2023-03-27 by MartyG-RealSense