realsenseai/realsense-ros #2643: /camera/imu topic not published by Intel RealSense Depth Camera D435f on Ubuntu 20.04 Noetic

skye221, 2023-03-06:
Hi, 
I am trying to use the Intel RealSense Depth Camera D435f with ROS on Ubuntu 20.04 Noetic. However, I cannot see the /camera/imu topic when running the rostopic list command, even though I have enabled the IMU using the enable_accel and enable_gyro arguments in the roslaunch command.

I have attached an image of the rostopic list command output showing that /camera/imu is not listed. I have also tried running the rs-imu-test command, but it returns an error saying that the IMU is not available.

Is there anything else that I need to do to enable the IMU on the D435f camera on Ubuntu 20.04 Noetic? Thanks
[image]

MartyG-RealSense, 2023-03-06:
Hi @skye221  The D435f is not equipped with an IMU component unfortunately.  The 'f' stands for 'filter'.  It is an IMU-less **D435** camera with sensors that are equipped with a filter that blocks visible light frequencies from entering them and allows only higher near-infrared frequencies through.

The D435 model that has an IMU is the **D435i**.  The D455 model is also equipped with one.

MartyG-RealSense, 2023-03-15:
Hi @skye221  Do you require further assistance with this case, please?  Thanks!

skye221, 2023-03-15:
No, thank you.

closed 2023-03-15 by skye221