realsenseai/realsense-ros #2644: launch error: ros2 launch realsense_camera rs_launch.py

thanujTa, 2023-03-06:
the display of camera is showing when i launch the realsense2 file the log :
[INFO] [launch]: All log files can be found below /home/robotics-lab/.ros/log/2023-03-06-14-01-58-355437-roboticslab-Precision-3660-56779
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [56781]
[realsense2_camera_node-1] [INFO] [1678107718.559793262] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1678107718.560002831] [camera.camera]: Built with LibRealSense v2.53.1
[realsense2_camera_node-1] [INFO] [1678107718.560035370] [camera.camera]: Running with LibRealSense v2.53.1
[realsense2_camera_node-1] [INFO] [1678107718.602946734] [camera.camera]: Device with serial number 148522070121 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1678107718.602985650] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1678107718.602989744] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1678107718.603087821] [camera.camera]: Device with port number 2-8 was found.
[realsense2_camera_node-1] [INFO] [1678107718.603094400] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1678107718.611036168] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1678107718.611658655] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1678107718.611721457] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1678107718.611739211] [camera.camera]: Device Serial No: 148522070121
[realsense2_camera_node-1] [INFO] [1678107718.611752941] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1678107718.611767148] [camera.camera]: Device FW version: 05.14.00.00
[realsense2_camera_node-1] [INFO] [1678107718.611780634] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1678107718.611810555] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1678107718.646697462] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1678107718.666767035] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1678107718.667250108] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1678107718.667268259] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1678107718.678541657] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1678107718.680128222] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1678107718.692483552] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1678107718.692581708] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1678107718.693145991] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1678107718.706168759] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1678107718.706382921] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1678107718.800476467] [camera.camera]: 

and thats it nothing else.

please help me out

thanks

MartyG-RealSense, 2023-03-06:
Hi @thanujTa  Your log looks normal and appears to conclude successfully with **RealSense Node Is Up!**

Once launch has completed then you can view the published streams in the **Rviz** interface.  Have you launched Rviz with the command **ros2 run rviz rviz** in the ROS terminal please?

thanujTa, 2023-03-06:
Hello @MartyG-RealSense I tried to open rviz through the command you sent but it shows package 'rviz ' not found, but if i enter rviz2 in another terminal rviz is launched but no streams are seen. did i miss something here?

MartyG-RealSense, 2023-03-06:
I should have said rviz2 instead of rviz.  I do apologize.

**ros2 run rviz2 rviz2**

closed 2023-03-21 by MartyG-RealSense