realsenseai/realsense-ros #2646: [ROS 1] Stops after publishing a few images while consuming 100% CPU in Jetson AGX Xavier ( JetPack 4.6.3)

ravijo, 2023-03-08:
I am facing a weird problem: the `realsense-viewer` is working fine, but the ROS driver stops after publishing a few images. The ROS driver is consuming 100%. Furthermore, it seems that the issue is happening when `enable_pointcloud` is set as shown below:

```
$ roslaunch realsense2_camera rs_camera.launch align_depth:=true enable_pointcloud:=true depth_fps:=30 color_fps:=30 depth_width:=640 depth_height:=480 color_width:=640 color_height:=480
...
[ INFO] [1678252263.390256920]: Initializing nodelet with 4 worker threads.
[ INFO] [1678252263.630721496]: RealSense ROS v2.3.2
[ INFO] [1678252263.630905496]: Built with LibRealSense v2.50.0
[ INFO] [1678252263.631000152]: Running with LibRealSense v2.50.0
[ INFO] [1678252263.724109368]:  
 08/03 14:11:03,802 WARNING [547004346752] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 08/03 14:11:03,856 WARNING [547004346752] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ INFO] [1678252264.033352888]: Device with serial number 943222073638 was found.
[ INFO] [1678252264.033476024]: Device with physical ID 2-1-3 was found.
[ INFO] [1678252264.033525048]: Device with name Intel RealSense D435I was found.
[ INFO] [1678252264.034807512]: Device with port number 2-1 was found.
[ INFO] [1678252264.034971608]: Device USB type: 3.2
[ INFO] [1678252264.341761912]: Device FW version: 05.14.00.00
[ INFO] [1678252264.341798552]: Device Product ID: 0x0B3A
[ INFO] [1678252264.341833560]: Enable PointCloud: On
[ INFO] [1678252264.473976120]: Setting Dynamic reconfig parameters.
 08/03 14:11:06,738 WARNING [547004346752] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1678252268.498198168]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1678252268.499309592]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
 08/03 14:11:09,427 WARNING [547004346752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1678252269.558560120]: 
 08/03 14:11:09,573 WARNING [547004346752] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ INFO] [1678252269.600913592]: RealSense Node Is Up!
 08/03 14:11:09,951 WARNING [547004346752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/03 14:11:10,659 ERROR [546618446208] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
^C[camera/realsense2_camera-2] killing on exit
[camera/realsense2_camera_manager-1] killing on exit
shutting down processing monitor complete
done
```

The above logs are cropped to make this post cleaner.  Please see [logs.txt](https://github.com/IntelRealSense/realsense-ros/files/10917545/logs.txt) for complete logs.

Below is a screenshot showing CPU usage while executing the above command. Please note that no other application is running during this time, and the display is attached directly to the device with an HDMI cable.

[image]

# Version Information
* JetPack v4.6.3
* Kernel Info
    ```
    $ uname -a
    Linux ubuntu 4.9.299-tegra #1 SMP PREEMPT Tue Nov 22 09:24:30 PST 2022 aarch64 aarch64 aarch64 GNU/Linux
    ```
* GCC Info
    ```
    $ gcc --version
    gcc (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 7.5.0
    Copyright (C) 2017 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
    ```
* Python 2 Info
    ```
    $ python --version
    Python 2.7.17
    ```
* Python 3 Info
    ```
    $ python3 --version
    Python 3.6.9
    ```
* OS Info
    ```
    $ lsb_release -a
    No LSB modules are available.
    Distributor ID:	Ubuntu
    Description:	Ubuntu 18.04.6 LTS
    Release:	18.04
    Codename:	bionic
    ```
* Camera Info
    ```
    $ rs-enumerate-devices -s
    Device Name                   Serial Number       Firmware Version
    Intel RealSense D435I         943222073638        05.14.00.00
    ```
* Package `librealsense2` Info
    ```
    $ apt-cache policy librealsense2
    librealsense2:
      Installed: 2.53.1-0~realsense0.703
      Candidate: 2.53.1-0~realsense0.703
      Version table:
     *** 2.53.1-0~realsense0.703 500
            500 https://librealsense.intel.com/Debian/apt-repo bionic/main arm64 Packages
            100 /var/lib/dpkg/status
         2.51.1-0~realsense0.476 500
            500 https://librealsense.intel.com/Debian/apt-repo bionic/main arm64 Packages
         2.51.1-0~realsense0.474 500
            500 https://librealsense.intel.com/Debian/apt-repo bionic/main arm64 Packages
         2.50.0-0~realsense0.6162 500
            500 https://librealsense.intel.com/Debian/apt-repo bionic/main arm64 Packages
    ```
* Package `librealsense2-utils` Info
    ```
    $ apt-cache policy librealsense2-utils 
    librealsense2-utils:
      Installed: 2.53.1-0~realsense0.703
      Candidate: 2.53.1-0~realsense0.703
      Version table:
     *** 2.53.1-0~realsense0.703 500
            500 https://librealsense.intel.com/Debian/apt-repo bionic/main arm64 Packages
            100 /var/lib/dpkg/status
         2.51.1-0~realsense0.476 500
            500 https://librealsense.intel.com/Debian/apt-repo bionic/main arm64 Packages
         2.51.1-0~realsense0.474 500
            500 https://librealsense.intel.com/Debian/apt-repo bionic/main arm64 Packages
         2.50.0-0~realsense0.6162 500
            500 https://librealsense.intel.com/Debian/apt-repo bionic/main arm64 Packages
    ```
* Package `ros-melodic-realsense2-camera` Info
    ```
    $ apt-cache policy ros-melodic-realsense2-camera
    ros-melodic-realsense2-camera:
      Installed: 2.3.2-1bionic.20221025.211354
      Candidate: 2.3.2-1bionic.20221025.211354
      Version table:
     *** 2.3.2-1bionic.20221025.211354 500
            500 http://packages.ros.org/ros/ubuntu bionic/main arm64 Packages
            100 /var/lib/dpkg/status
    ```

Please note that the ROS driver is installed from the `apt`.

# Additional Info
The `dmesg` does not contain any error message. However, a few entries shown below appeared frequently:
```
$ dmesg | grep error
$ dmesg
...
[ 4421.321332] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 4421.325769] uvcvideo: Unable to create debugfs 2-3 directory.
[ 4421.325939] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 4421.326138] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
[ 4421.326270] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
[ 4421.326649] input: Intel(R) RealSense(TM) Depth Ca as /devices/3610000.xhci/usb2/2-1/2-1:1.0/input/input183
...
```

# Summary
✅ The `realsense-viewer` works
✅ Default `rs_camera.launch` works
❌ The `rs_camera.launch` while `enable_pointcloud` is set does not work
❌ The `rs_camera.launch` while `enable_pointcloud` consumes 100% CPU

Any workaround, please?

MartyG-RealSense, 2023-03-08:
Hi @ravijo  There are known issues with enabling pointcloud on Jetson devices specifically in the ROS1 wrapper, as described at https://github.com/IntelRealSense/realsense-ros/issues/1967

A workaround is to launch with **rs_rgbd.launch** instead of rs_camera.launch.

`roslaunch realsense2_camera rs_rgbd.launch`

Before this launch is performed, support for an RGBD launch in the ROS1 wrapper must first be installed by using the command below in the Ubuntu terminal for your particular ROS version (Melodic in this case).

**Kinetic**
sudo apt-get install ros-kinetic-rgbd-launch

**Melodic**
sudo apt-get install ros-melodic-rgbd-launch

**Noetic**
sudo apt-get install ros-noetic-rgbd-launch

MartyG-RealSense, 2023-03-15:
Hi @ravijo  Do you require further assistance with this case, please?  Thanks!

ravijo, 2023-03-15:
Thank you very much. 

I needed more time to check your suggestions. For now, please leave this issue without closing it.

MartyG-RealSense, 2023-03-15:
That's no problem at all to keep this issue open for a further time period.  Thanks very much for the update!

ravijo, 2023-03-25:
@MartyG-RealSense 

Thank you very much for waiting. Based on your instructions, I installed `ros-melodic-rgbd-launch` and then executed `rs_rgbd.launch`. I monitored CPU usage as well. Please see the screenshot below:

[image]

Below is the output shown on terminal:

```
$ roslaunch realsense2_camera rs_rgbd.launch
... logging to /home/jetson/.ros/log/7986c414-ca36-11ed-a624-00044bcbdddc/roslaunch-jetson-31117.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://127.0.0.1:39021/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
[... 103 lines cut ...]
 24/03 20:24:52,375 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1679657092.411122082]: RealSense Node Is Up!
 24/03 20:24:52,581 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 20:24:52,632 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 20:24:52,785 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 20:25:03,797 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 20:25:50,048 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 20:25:57,127 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
^C[camera/points_xyzrgb_hw_registered-5] killing on exit
[camera/color_rectify_color-4] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[camera/realsense2_camera-3] killing on exit
 24/03 20:26:18,427 ERROR [546836558208] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Below is a screenshot showing Jetson informaiton:

[image]

Thank you for your invaluable help and understanding. I am looking forward to hear fom you.

MartyG-RealSense, 2023-03-25:
I note that you are using librealsense 2.50.0 and firmware 5.14.0.0.  Using this firmware version with 2.50.0 could result in errors as it was designed for use with the newer SDK 2.53.1.  I would therefore recommend downgrading your firmware version to **5.13.0.50** which is the one that is designed for use with 2.50.0.

ravijo, 2023-03-25:
@MartyG-RealSense 

Thanks a lot for the quick response. 

> I would therefore recommend downgrading your firmware version to 5.13.0.50 which is the one that is designed for use with 2.50.0.

How to downgrade RealSense firmware to install 5.13.0.50?

On the side note, the firmware was updated to latest one because `realsense-viewer` alerted to do so.

MartyG-RealSense, 2023-03-25:
You can download the firmware **.bin** file for 5.13.0.50 to your computer from the link below.

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

[image]

Once the .bin is downloaded, go to the **More** option near the top of the Viewer's options side-panel and select **Update Firmware** from its drop-down menu.  A file selector will appear.  Navigate to the folder on your computer that the .bin file was downloaded to and select the .bin file.  The downgrade will then automatically begin.

ravijo, 2023-03-28:
@MartyG-RealSense 

It works well when `realsense-ros` package is compiled from source.

Thanks

closed 2023-03-28 by ravijo

tugbakara, 2023-05-01:
> @MartyG-RealSense
> 
> It works well when `realsense-ros` package is compiled from source.
> 
> Thanks

How much CPU usage did you acheve after doing so?

ravijo, 2023-05-01:
@tugbakara 

Below is the summary:

|Case|RealSense ROS Package Installation|RealSense ROS Package Version|librealsense Version|Firmware Version|Image size|Commanded FPS|Observed Color FPS|Observed Depth FPS|Observed Point Cloud FPS|Avg CPU Usage %|Avg Memory Usage (GB)|Remarks                                  |
|----|----------------------------------|-----------------------------|--------------------|----------------|----------|-------------|------------------|------------------|------------------------|---------------|---------------------|-----------------------------------------|
|1   |catkin_make                       |2.3.2                        |2.53.1              |05.14.00.00     |640x480   |15           |14.987            |14.987            |14.988                  |34.8           |3.1                  |                                         |
|2   |catkin_make                       |2.3.2                        |2.53.1              |05.14.00.00     |640x480   |30           |23.111            |21.955            |20.421                  |47.2           |3.1                  |                                         |
|3   |catkin_make                       |2.3.2                        |2.53.1              |05.14.00.00     |848x480   |15           |14.994            |14.984            |14.991                  |31.2           |3.1                  |                                         |
|4   |catkin_make                       |2.3.2                        |2.53.1              |05.14.00.00     |848x480   |30           |28.262            |22.306            |22.287                  |48.5           |3.1                  |                                         |
|5   |catkin_make                       |2.3.2                        |2.53.1              |05.14.00.00     |1280x720  |15           |7.937             |7.417             |6.965                   |46.1           |3.0                  |No new messages sometimes for 1-2 seconds|
|6   |catkin_make                       |2.3.2                        |2.53.1              |05.14.00.00     |1280x720  |30           |                  |                  |                        |42.5           |3.0                  |No new messages after a few seconds      |
|7   |apt-get                           |2.3.2                        |2.50.0              |05.14.00.00     |          |             |                  |                  |                        |               |                     |Can not run. Firmware unsupported        |