realsenseai/realsense-ros #2648: Depth and infra sequence number alignment with external trigger

dongdongbh, 2023-03-10:
I'm using realsense D435i with my external trigger device, I connect my trigger signal to the D435i sync pin 5 as [Multi-Camera configurations](https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration).

I want to use the timestamp of my trigger signal to get the exact trigger time of D435 depth and infra capture time. I already have my trigger time stamp with trigger sequence number, I set `inter_cam_sync_mode=4`, which is trigger camera capture pre-trigger signal. Then in realsense-ros warper, I cache all video stream with camera sequence number and try to find corresponding trigger sequence number. Then I get the trigger shift and keep tracking the frame and align with my trigger time stamp. I noticed that when I set the trigger signal as 5Hz, the frame rate of d435 is also 5Hz, which suggests the external trigger worked.

But here is the problem, when the first trigger signal send with `trig_seq=0`, the depth frame will came first with `seq[stream] = 1`, and then after `trig_seq=1`, the `seq[stream] = 2` depth frame come, then after `trig_seq=2` the depth frame `seq[stream] = 3` depth frame come, then the infra1 frame with `seq[stream] = 1` come, then the infra2 frame with `seq[stream] = 1` come. 

| trig_seq | c_seq[depth] | c_seq[infra1] | c_seq[infra12] |
|----------|--------------|---------------|----------------|
| 0        | 1            | -             | -              |
| 1        | 2            | -             | -              |
| 2        | 3            | 1             | 1              |
| 3        | 4            | 2             | 2              |

My problem is, does infra frame with `seq[stream] = 1` align with depth frame with `seq[stream] = 1`? If it aligned, why it comes after more than 600ms after trigger signal? If not, how can I determinate which `trig_seq` number corresponds which `seq[stream]`. 

I also tested different trigger frame rate, the seqence of infra shift are also different. I tested if `abs(timeshift) < 1.0 / frame_rate*0.75`, then I think the seqence is aligned, and record the sequence shift for each channel (depth, infra1, infra2), then replace the frame topic timestamp with my trigger time `trig_stamp + exposure_time /2`. I try it with my VIO algorithm, it works really bad, even worse than the orignal time stamp.

## Test environment
System: Ubuntu 20.04
ROS: ROS1 noetic
Realsense-ros: v2.3.2
LibRealSense: v2.50.0
Matadata available with patched kernal

@MartyG-RealSense @doronhi

MartyG-RealSense, 2023-03-10:
Hi @dongdongbh  The External Synchronization (genlock) hardware sync system only timestamp-syncs **depth** to a trigger and not infrared.

MartyG-RealSense, 2023-03-10:
Mode 3 is called Full Slave.  Whilst Mode 2 is 'Slave' and only syncs depth, Mode 3 syncs depth and RGB (but not infrared) but it only works with the D415 camera model.

The depth frame is calculated from **raw** left and right infrared frames in the camera hardware, which are not related to the Infrared and Infrared 2 streams.  This is why depth can still be produced even if Infrared and Infrared 2 are not enabled.

dongdongbh, 2023-03-10:
So there is no way to hardware trigger infrared frames? Is there a way to get the raw infrared frames timestamps, since the infrared frames frequency also changed with external trigger, so I think they are indeed related to the trigger signal. But the timestamp are different, it's so weird.

MartyG-RealSense, 2023-03-10:
With a camera that is not using hardware sync, the left infrared stream is perfectly aligned and synced with the depth stream because the two originate from the same sensor.

The genlock hardware sync system (Mode 4+) synchronizes the depth timestamps of a 'slave' camera with an external trigger.  The genlock system's documentation makes no mention of infrared.

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/6195 obtained raw hardware timestamps through camera **metadata** using `frame.get_frame_metadata(RS2_FRAME_METADATA_SENSOR_TIMESTAMP)`

dongdongbh, 2023-03-10:
I know I can get the raw timestamp of infrared, but is is based on D435's clock, I'm using external trigger, I want to know how the timestamp align with my external trigger, then I can use my external clock. Because the external trigger does has impact on infrared frequency, the infrared frames has same frequency as the trigger signal. 

Can you double check with the developer what is actual relationship between the sequence number between depth frame, external trigger and their timestamps. I'm sure they are clearly related, I just want to know what is going on between them.

MartyG-RealSense, 2023-03-10:
Support for the genlock sync system has ceased unfortunately and its documentation has been withdrawn.  This is because it was an experimental proof of concept rather than an official, validated Intel feature (modes 1 and 2 are the only ones that are officially supported and validated by Intel).  So developer assistance for this external trigger system is now regretfully unavailable.

I can provide an archived PDF copy of the documentation at the link below however if you have not seen it already.

[External Synchronization.pdf](https://github.com/IntelRealSense/realsense-ros/files/10943376/External.Synchronization.pdf)

dongdongbh, 2023-03-10:
But the mode 2 has the same problem and is supported

MartyG-RealSense, 2023-03-10:
Using an external trigger with mode 2 is possible but significantly more difficult than with genlock because the frequency of the trigger has to be matched extremely closely with the 'actual' FPS of the slave camera (e.g 30.02 FPS instead of 30 FPS).  Usually with mode 2, instead of receiving an external trigger the slave is connected by sync cable to a 'master' RealSense camera that outputs a trigger pulse from its sync pin when it is set to mode 1.

As far as mode 1 and 2 hardware sync is concerned, the behaviour of infrared during sync is not considered at all.  It is only about synching the slave's depth timestamps with the master's depth timestamps, or with an external trigger pulse if the pulse is able to be tuned finely enough to match with the actual FPS of the slave camera.

dongdongbh, 2023-03-10:
How about mode 3, seems mode 3 does not need to be triggered with exact frequency from #2323.

MartyG-RealSense, 2023-03-10:
Mode 3 does not require an exactly tuned frequency because it is part of the genlock sync system rather than the original sync system where synchronizng an external trigger to a camera was much more difficult.  As mentioned earlier, the RGB sync will not work with non-D415 camera models though.

dongdongbh, 2023-03-11:
> The behaviour of infrared during sync is not considered at all. 

I think the behaviour of infrared should be state clearly, because the sync indeed has impact to infrared frames.

MartyG-RealSense, 2023-03-11:
Your suggestion would be a documentation related request.  If you wish you can create an issue at the main librealsense forum with a short message containing a request to add information to the hardware sync documentation regarding the influence on infrared.  I can then add a **Documentation** label to the issue and keep it open so that a RealSense team member responsible for documentation can evaluate it at a future date.

https://github.com/IntelRealSense/librealsense/issues

dongdongbh, 2023-03-11:
Great! Thanks for your help.

dongdongbh, 2023-03-14:
After debugging the timestamps, in trigger mode, I found that it seems the RealSense camera first publishes two depth frames, then publishes the third depth frame alongside the first infrared frames, which have the same timestamp as those from librealsense. 

| trig_seq | c_seq[depth] | c_seq[infra1] | c_seq[infra12] | c_time[depth] | c_time[infra12] |
|----------|--------------|---------------|----------------|---------------|---------------|
| 0        | 1            | -             | -              | c_1           | -             |
| 1        | 2            | -             | -              | c_2           | -             |
| 2        | 3            | 1             | 1              | c_3           | c_3           |
| 3        | 4            | 2             | 2              | c_4           | c_4           |

I hope that the RealSense team can help confirm this.

MartyG-RealSense, 2023-03-14:
I have passed on your latest question to my colleagues.  Thank you.

MartyG-RealSense, 2023-03-16:
I have discussed your case with my Intel RealSense colleagues.  Thanks very much for your patience.

They advised that infrared and depth frames are always synchronized even when hardware sync is active.  This is because the depth is derived from infrared frames and so there is no way for them to be out of sync with each other.

dongdongbh, 2023-03-16:
How about the sequence number corespondence? Did they really test the trigger mode? On my test, it first generates two depth frames, then third depth frame seems sync with the first infrared frames, and they start show same timestamp. I think the third depth frame corespond the first infrared frames. I want confirm the this. If the first depth frame coresponds the first infrared frame. Then the first infrared frame comes later after two depth frames. In 5 Hz case, it comes around 600ms later than the first depth frame.

dongdongbh, 2023-03-23:
Any update on this thread?

MartyG-RealSense, 2023-03-23:
I will follow up with my Intel RealSense colleagues about your question.

MartyG-RealSense, 2023-03-23:
I have received further advice from my colleagues about your question.

*******

The frame timestamp on the depth and IR images should be in sync.  Frame counter / number is not an ideal way to sync between frames or cameras.  The RealSense Viewer can reproduce and show the diff / delta in frame number between depth and IR, so the behavior should be expected.  Please see the screenshot below.

[image]

dongdongbh, 2023-03-24:
Thanks for your update, I want to know if `infra_frame_num = depth_frame_num - 2` always hold.

MartyG-RealSense, 2023-03-24:
If there is consistently a difference of 2 between depth and infrared then it is reasonable to use that in your project as a predictable rule for frame-count behaviour.  This rule does not necessarily apply to every RealSense user's project though, since - as my colleagues advised - the frame number will not be an ideal way to check for sync.

dongdongbh, 2023-03-24:
OK, I got it. Because there seeems no other simple way to check for sync, I just use the frame number method for now.

MartyG-RealSense, 2023-03-24:
Thanks very much @dongdongbh

dongdongbh, 2023-04-27:
Here I want to report of the depth and infra sequence number alignment in genlock 259 and 260 for whom also interest in these mode. My previous issue #2592 said that I want to use these mode to do synchronization with external trigger signal with emitter on-off. 

After my test, in genlock 259 and 260  sequence number align as following

## Test configuration
Realsense: D435i
System: Ubuntu 20.04
ROS: ROS1 noetic
Realsense-ros: v2.3.2
LibRealSense: v2.50.0

External trigger frequency: 4Hz
```
<arg name="depth_fps"           default="90"/>
<arg name="infra_fps"           default="90"/>

/camera/stereo_module/emitter_enabled: 1
/camera/stereo_module/global_time_enabled: true
```
The emiter state get from `(int)f.get_frame_metadata(RS2_FRAME_METADATA_FRAME_LASER_POWER_MODE);`

## Genlock 259 emitter ON_OFF mode
| trig_seq | emitter state | c_seq[depth] | c_seq[infra12]      | c_time[depth] | c_time[infra12] | dt = c_n -c_{n-1} (ms) |
|----------|---------------|--------------|---------------------|---------------|-----------------|------------------------|
| 0        | 1             | 1            |                     | c_1           |                 |                        |
| 1        | 1             | 2            |                     | c_2           |                 | 83.7                   |
|          | 0             | 3            | **1(with emmitter on)** | c_3           | c_3             | 11.1                   |
| 2        | 1             | 4            | 2                   | c_4           | c_4             | 238.9                  |
|          | 0             | 5            | 3                   | c_5           | c_5             | 11.1                   |
| 3        | 1             | 6            | 4                   | c_6           | c_6             | 238.9                  |
|          | 0             | 7            | 5                   | c_7           | c_7             | 11.1                   |
| 4        | 1             | 8            | 6                   | c_8           | c_8             | 238.9                  |
|          | 0             | 9            | 7                   | c_9           | C-9             | 11.1                   |

Regarding Genlock 259, it is noteworthy that the first infra frame has an 'emitter on' state, which appears unusual. However, the interval between the trigger frames (emitter on and emitter off) is 11.1ms, conforming to the native frame rate of 90 Hz. It is likely that this interval was set according to the native frame rate. Since the trigger frequency is 4Hz, the trigger interval is 250ms, which is equivalent to the sum of 238.9ms and 11.1ms.

## Genlock 260  emitter OFF_ON mode
| trig_seq | emitter state | c_seq[depth] | c_seq[infra12] | c_time[depth] | c_time[infra12] | dt = c_n -c_{n-1} (ms) |
|----------|---------------|--------------|----------------|---------------|-----------------|------------------------|
| 0        | 0             | 1            |                | c_1           |                 |                        |
| 1        | 0             | 2            |                | c_2           |                 | 56.3                   |
|          | 1             | 3            | 1              | c_3           | c_3             | 11.1                   |
| 2        | 0             | 4            | 2              | c_4           | c_4             | 238.9                  |
|          | 1             | 5            | 3              | c_5           | c_5             | 11.1                   |
| 3        | 0             | 6            | 4              | c_6           | c_6             | 238.9                  |
|          | 1             | 7            | 5              | c_7           | c_7             | 11.1                   |
| 4        | 0             | 8            | 6              | c_8           | c_8             | 238.9                  |
|          | 1             | 9            | 7              | c_9           | C-9             | 11.1                   |

The results indicate that the frames triggered by the first two signals are not stable. However, the system starts working as expected from the third signal onwards.

MartyG-RealSense, 2023-04-27:
Thanks so much @dongdongbh for sharing such detailed feedback!