realsenseai/realsense-ros #2649: Feature request: Add USB information to DeviceInfo.srv

stwirth, 2023-03-10:
On startup, the camera driver reports a lot of interesting information:
```
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.009290]: Device with serial number 147422020966 was found.                                                           
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.009337]: Device with physical ID 5-2.1-10 was found.                                                                 
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.009356]: Device with name Intel RealSense D410 was found.                                                            
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.010046]: Device with port number 5-2.1 was found.                                                                    
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.010070]: Device USB type: 3.2                                                    
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.012230]: getParameters...                                                                                            
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.039811]: setupDevice...                                                                                              
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.039825]: JSON file is not provided                                                                                   
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.039833]: ROS Node Namespace: camera       
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.039841]: Device Name: Intel RealSense D410                                                                           
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.039847]: Device Serial No: 147422020966                
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.039853]: Device physical port: 5-2.1-10                                                                              
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.039859]: Device FW version: 05.12.07.150                                                                             
[INFO] [/camera/nodelet_manager] [2023-03-10 03:17:48.039865]: Device Product ID: 0x0AD2
```
I would like to have this information available at runtime, especially "port number 5-2.1" and "physical ID 5-2.1-10".
I think it would be great if this could be added to the fields in `DeviceInfo.srv`.

MartyG-RealSense, 2023-03-10:
Hi @stwirth  Camera information is published by the ROS wrapper on the **/diagnostics** topic.  Do you get the information that you need if you echo the topic with the following command:

`rostopic echo /diagnostics`

stwirth, 2023-03-10:
Hi @MartyG-RealSense,
no, I don't see that information there.

MartyG-RealSense, 2023-03-10:
The device information can be accessed in the wrapper through a **device_info** 'service call'.

**ROS1**
https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#available-services

**ROS2**
https://github.com/IntelRealSense/realsense-ros/tree/ros2-development#available-services

MartyG-RealSense, 2023-03-11:
You are welcome to submit an edit of rs_node_setup.cpp as a **Pull Request (PR)** for consideration for merging by the RealSense ROS developer team into the ROS wrapper.

https://github.com/IntelRealSense/realsense-ros/pulls

You could also create a custom 'fork' version of the wrapper on your own GitHub account and make the edit there.

[image]

stwirth, 2023-03-13:
PRs for ROS1 (#2654) and ROS2 (#2655) created.

MartyG-RealSense, 2023-03-13:
Thanks so much for the PRs!

commit 7d77cb01 (2023-03-14): Fix #2649: Expose USB port in DeviceInfo service
commit 45386822 (2023-03-20): Fix #2649: Expose USB port in DeviceInfo service
closed 2023-03-14 by Nir-Az