realsenseai/realsense-ros #2650: realsense2_camera complilation error

Shillin123, 2023-03-10:
Hi, after using command: catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
I encounter this problem, can you please help to solve it? 

[ 70%] Built target realsense2_camera_generate_messages
[ 82%] Building CXX object realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o
[ 82%] Building CXX object realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/framos_realsense_node_factory.cpp.o
In file included from /opt/ros/noetic/include/librealsense2/hpp/rs_sensor.hpp:8,
                 from /opt/ros/noetic/include/librealsense2/hpp/rs_device.hpp:8,
                 from /opt/ros/noetic/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /opt/ros/noetic/include/librealsense2/hpp/rs_context.hpp:8,
                 from /opt/ros/noetic/include/librealsense2/rs.hpp:9,
                 from /home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:12,
                 from /home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp:1047:58: warning: comparison of integer expressions of different signedness: ‘int’ and ‘const size_t’ {aka ‘const long unsigned int’} [-Wsign-compare]
 1047 |                         frm.get_profile().stream_index() == index) f = frm;
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp:1069:58: warning: comparison of integer expressions of different signedness: ‘int’ and ‘const size_t’ {aka ‘const long unsigned int’} [-Wsign-compare]
 1069 |                         frm.get_profile().stream_index() == index) f = frm;
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp:1091:58: warning: comparison of integer expressions of different signedness: ‘int’ and ‘const size_t’ {aka ‘const long unsigned int’} [-Wsign-compare]
 1091 |                         frm.get_profile().stream_index() == index) f = frm;
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
In file included from /opt/ros/noetic/include/librealsense2/hpp/rs_sensor.hpp:8,
                 from /opt/ros/noetic/include/librealsense2/hpp/rs_device.hpp:8,
                 from /opt/ros/noetic/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /opt/ros/noetic/include/librealsense2/hpp/rs_context.hpp:8,
                 from /opt/ros/noetic/include/librealsense2/rs.hpp:9,
                 from /home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:12,
                 from /home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/../include/framos_realsense_node_factory.h:6,
                 from /home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/framos_realsense_node_factory.cpp:4:
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp:1047:58: warning: comparison of integer expressions of different signedness: ‘int’ and ‘const size_t’ {aka ‘const long unsigned int’} [-Wsign-compare]
 1047 |                         frm.get_profile().stream_index() == index) f = frm;
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp:1069:58: warning: comparison of integer expressions of different signedness: ‘int’ and ‘const size_t’ {aka ‘const long unsigned int’} [-Wsign-compare]
 1069 |                         frm.get_profile().stream_index() == index) f = frm;
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/noetic/include/librealsense2/hpp/rs_frame.hpp:1091:58: warning: comparison of integer expressions of different signedness: ‘int’ and ‘const size_t’ {aka ‘const long unsigned int’} [-Wsign-compare]
 1091 |                         frm.get_profile().stream_index() == index) f = frm;
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/framos_realsense_node_factory.cpp: In member function ‘void realsense2_camera::FramosRealSenseNodeFactory::setRs2CliArgs()’:
/home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/framos_realsense_node_factory.cpp:106:10: error: ‘rs2::d400e’ has not been declared
  106 |     rs2::d400e::cli_arg carg;
      |          ^~~~~
/home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/framos_realsense_node_factory.cpp:108:5: error: ‘carg’ was not declared in this scope
  108 |     carg.argc = 0;
      |     ^~~~
/home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/framos_realsense_node_factory.cpp:125:14: error: ‘rs2::d400e’ has not been declared
  125 |         rs2::d400e::set_cli_args(carg);
      |              ^~~~~
/home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/framos_realsense_node_factory.cpp: In member function ‘void realsense2_camera::FramosRealSenseNodeFactory::setDeviceFilterList()’:
/home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/framos_realsense_node_factory.cpp:131:17: error: ‘rs2_d400e_filter’ was not declared in this scope
  131 |     std::vector<rs2_d400e_filter> filter_list;
      |                 ^~~~~~~~~~~~~~~~
/home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/framos_realsense_node_factory.cpp:131:33: error: template argument 1 is invalid
  131 |     std::vector<rs2_d400e_filter> filter_list;
      |                                 ^
/home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/framos_realsense_node_factory.cpp:131:33: error: template argument 2 is invalid
/home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/framos_realsense_node_factory.cpp:132:21: error: expected ‘;’ before ‘filter’
  132 |     rs2_d400e_filter filter;
      |                     ^~~~~~~
      |                     ;
/home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/framos_realsense_node_factory.cpp:133:5: error: ‘filter’ was not declared in this scope; did you mean ‘rs2::filter’?
  133 |     filter.serial_range.begin = 0;
[... 38 lines cut ...]
                 from /home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/c++/9/bits/stl_algo.h:3960:5: note:   ‘std::find_if’
 3960 |     find_if(_InputIterator __first, _InputIterator __last,
      |     ^~~~~~~
In file included from /usr/include/boost/mpl/find.hpp:17,
                 from /usr/include/boost/mpl/aux_/contains_impl.hpp:20,
                 from /usr/include/boost/mpl/contains.hpp:20,
                 from /usr/include/boost/math/policies/policy.hpp:10,
                 from /usr/include/boost/math/policies/error_handling.hpp:21,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/noetic/include/ros/time.h:58,
                 from /opt/ros/noetic/include/ros/console.h:39,
                 from /opt/ros/noetic/include/nodelet/nodelet.h:40,
                 from /home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:7,
                 from /home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/franka/cam_fi/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/boost/mpl/find_if.hpp:32:8: note:   ‘boost::mpl::find_if’
   32 | struct find_if
      |        ^~~~~~~
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:76：realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/framos_realsense_node_factory.cpp.o] 错误 1
make[2]: *** 正在等待未完成的任务....
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:89：realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:1485：realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] 错误 2
make: *** [Makefile:130：all] 错误 2
Invoking "make -j8 -l8" failed

MartyG-RealSense, 2023-03-10:
Hi @Shillin123  The command `catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release` is known to cause problems when installing the RealSense ROS wrapper on Windows.  It is caused by **-DCATKIN_ENABLE_TESTING=False** but setting it to True instead of False can cause a different error to occur.  

Examples of past cases where this problem is discussed are https://github.com/IntelRealSense/realsense-ros/issues/1797 and https://github.com/IntelRealSense/realsense-ros/issues/2280

There is not a clear solution for this issue, though a couple of RealSense ROS users affected by it reported that the problem disappeared if they used an older version of the ROS wrapper.  For one of those users a combination of ROS1 wrapper **2.2.22**.  the correct librealsense SDK match for that wrapper version is **2.42.0**.

Shillin123, 2023-03-14:
Thanks, I solved it!

MartyG-RealSense, 2023-03-14:
That's excellent news, @Shillin123 - thanks very much for the update!

closed 2023-03-21 by MartyG-RealSense