realsenseai/realsense-ros #2651: Can't load the yaml file in rs_launch.py

EndlessLoops, 2023-03-12:
I can't load the yaml file in rs_launch.py in ubuntu2204.I only installed the realsense2_camera package with apt and I didn't install the librealsense SDK.Do I have to install the SDK before loading the yaml file.Or it's an error yaml file.

Terminal output:

```
$ ros2 launch realsense2_camera rs_launch.py config_file:=~/d435i.yaml
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-03-12-12-21-46-072222-T4-147050
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): invalid syntax (<string>, line 1)
```
Yaml file:

```
$ cat d435i.yaml 
camera:
  camera:
    ros__parameters:
        serial_no: '' 
        initial_reset: true
        base_frame_id: camera

        align_depth: false
        enable_pointcloud: false
        enable_fisheye: false
        enable_fisheye1: false
        enable_fisheye2: false
        enable_pose: false

        enable_color: true
        color_fps: 30.0
        color_height: 480
        color_width: 640
            
        enable_depth: true
        depth_fps: 30.0
        depth_height: 720
        depth_width: 1280

        enable_infra: false
        enable_infra1: false
        enable_infra2: false

        enable_gyro: false

        enable_accel: false

```

MartyG-RealSense, 2023-03-12:
Hi @EndlessLoops  If you installed the ROS wrapper from packages with apt-get install then the librealsense SDK and ROS wrapper should be both installed together from the same time.  You would therefore not need to install the SDK separately afterwards.

When loading a .yaml file via the launch file, an **absolute path** to the .yaml file should be defined, as advised at https://github.com/IntelRealSense/realsense-ros/issues/2043#issuecomment-904345068

EndlessLoops, 2023-03-12:
I had tried. But it also have the same problem.
```
$ ros2 launch realsense2_camera rs_launch.py config_file:=/home/ubuntu/d435i.yaml 
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-03-12-23-00-55-218656-T4-148156
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): invalid syntax (<string>, line 1)
```

EndlessLoops, 2023-03-12:
`$ cat d435i.yaml ` It just is a command code. And I find the only  the `config_file ` param is abnormal even if the yaml path is not correct.And othe param likes `log_level` can be load normally.

EndlessLoops, 2023-03-12:
I found it [2561](https://github.com/IntelRealSense/realsense-ros/issues/2561).

MartyG-RealSense, 2023-03-12:
Yes, I just found that answer at the same moment.  :)  Thanks fo the update!

EndlessLoops, 2023-03-13:
But It can‘t load normally the yaml file in the ros2 node. 

```
$ ros2 launch realsense2_camera rs_launch.py config_file:="'/home/ubuntu/d435i.yaml'"
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-03-13-11-18-00-445213-T4-35135
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [35136]
[realsense2_camera_node-1] [INFO] [1678677480.656158289] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1678677480.656240098] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1678677480.656252783] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1678677480.766128911] [camera.camera]: Device with serial number 836612073058 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1678677480.766310331] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1678677480.766364821] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1678677480.767212292] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1678677480.767278567] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1678677480.767325826] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1678677486.943423313] [camera.camera]: Device with serial number 836612073058 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1678677486.943601171] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1678677486.943666672] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1678677486.944546445] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1678677486.944664746] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1678677486.952878961] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1678677486.955946759] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1678677486.956059970] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1678677486.956135915] [camera.camera]: Device Serial No: 836612073058
[realsense2_camera_node-1] [INFO] [1678677486.956208399] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1678677486.956281286] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1678677486.956350663] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1678677486.956415600] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1678677487.195090935] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1678677487.205222619] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1]  13/03 11:18:07,205 WARNING [139697569396288] (backend-v4l2.cpp:1444) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[realsense2_camera_node-1] [INFO] [1678677487.210734674] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1678677487.211501863] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1678677487.211591542] [camera.camera]: Set ROS param accel_fps to default: 63
[realsense2_camera_node-1] [WARN] [1678677487.236245856] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {pointcloud.stream_index_filter} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1678677487.238325326] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1678677487.273231567] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1678677487.279634674] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1678677487.280783624] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1678677487.738485121] [camera.camera]: 
```

EndlessLoops, 2023-03-13:
I update my yaml file.Now it works fine.
```
$ cat d435i.yaml 
camera:
  camera:
    ros__parameters:
        camera_name: camera                      # 'default': 'camera', 'description': 'camera unique name'
        serial_no: ''                            # 'default': "''", 'description': 'choose device by serial number'
        usb_port_id: ''                          # 'default': "''", 'description': 'choose device by usb port id'
        config_file: ''                          # 'default': "''", 'description': 'yaml config file'
        unite_imu_method: 0                      # 'default': "0", 'description': '[0-None, 1-copy, 2-linear_interpolation]'
        json_file_path: ''                       # 'default': "''", 'description': 'allows advanced configuration'
        log_level: info                          # 'default': 'info', 'description': 'debug log level [DEBUG|INFO|WARN|ERROR|FATAL]'
        output: screen                           # 'default': 'screen', 'description': 'pipe node output [screen|log]'
        depth_module.profile: 848x480x30         # 'default': '0,0,0', 'description': 'depth module profile'                           
        enable_depth: true                       # 'default': 'true', 'description': 'enable depth stream'
        rgb_camera.profile: 1280x720x30          # 'default': '0,0,0', 'description': 'color image width'
        enable_color: true                       # 'default': 'true', 'description': 'enable color stream'
        enable_infra1: false                     # 'default': 'false', 'description': 'enable infra1 stream'
        enable_infra2: false                     # 'default': 'false', 'description': 'enable infra2 stream'
        infra_rgb: false                         # 'default': 'false', 'description': 'enable infra2 stream'
        tracking_module.profile: 0x0x0           # 'default': '0,0,0', 'description': 'fisheye width'
        enable_fisheye1: true                    # 'default': 'true', 'description': 'enable fisheye1 stream'
        enable_fisheye2: true                    # 'default': 'true', 'description': 'enable fisheye2 stream'
        enable_confidence: true                  # 'default': 'true', 'description': 'enable depth stream'
        gyro_fps: 0                              # 'default': '0', 'description': "''"                           
        accel_fps: 0                             # 'default': '0', 'description': "''"                           
        enable_gyro: false                       # 'default': 'false', 'description': "''"                           
        enable_accel: false                      # 'default': 'false', 'description': "''"                           
        enable_pose: true                        # 'default': 'true', 'description': "''"                           
        pose_fps: 200                            # 'default': '200', 'description': "''"                           
        pointcloud.enable: false                 # 'default': 'false', 'description':    
        pointcloud.stream_filter: 2              # 'default': '2', 'description': 'texture stream for pointcloud'
        pointcloud.stream_index_filter: 0        # 'default': '0', 'description': 'texture stream index for pointcloud'
        enable_sync: false                       # 'default': 'false', 'description': "''"                           
        align_depth.enable: false                # 'default': 'false', 'description': "''"                           
        colorizer.enable: false                  # 'default': 'false', 'description': "''"
        clip_distance: -2.                       # 'default': '-2.', 'description': "''"                           
        linear_accel_cov: 0.01                   # 'default': '0.01', 'description': "''"                           
        initial_reset: false                     # 'default': 'false', 'description': "''"                           
        allow_no_texture_points: false           # 'default': 'false', 'description': "''"                           
        ordered_pc: false                        # 'default': 'false', 'description':   
        calib_odom_file: ''                      # 'default': "''", 'description': "''"
        topic_odom_in: ''                        # 'default': "''", 'description': 'topic for T265 wheel odometry'
        tf_publish_rate: 0.0                     # 'default': '0.0', 'description': 'Rate of publishing static_tf'
        diagnostics_period: 0.0                  # 'default': '0.0', 'description': 'Rate of publishing diagnostics. 0=Disabled'
        decimation_filter.enable: false          # 'default': 'false', 'description': 'Rate of publishing static_tf'
        rosbag_filename: ''                      # 'default': "''", 'description': 'A realsense bagfile to run from as a device'
        depth_module.exposure.1: 7500            # 'default': '7500', 'description': 'Initial value for hdr_merge filter'
        depth_module.gain.1: 16                  # 'default': '16', 'description': 'Initial value for hdr_merge filter'
        depth_module.exposure.2: 1               # 'default': '1', 'description': 'Initial value for hdr_merge filter'
        depth_module.gain.2: 16                  # 'default': '16', 'description': 'Initial value for hdr_merge filter'
        wait_for_device_timeout: -1.             # 'default': '-1.', 'description': 'Timeout for waiting for device to connect (Seconds)'
        reconnect_timeout: 6.                    # 'default': '6.', 'description': 'Timeout(seconds) between consequtive reconnection attempts'
```

closed 2023-03-13 by EndlessLoops

MartyG-RealSense, 2023-03-13:
It's excellent to hear that you were successful, @EndlessLoops - thanks very much for the update!