realsenseai/realsense-ros #2656: Intel Realsense D435i

Trigon123, 2023-03-14:
I am using Intel realsense depth camera d435i. My system is intel NUC8i5BEH with configuration i5,16GB RAM,240GB SSD.
while running roslaunch rs_aligned_depth.launch all system core goes to 100%. How to reduce to system utilization.

Another thing, if I change parameters I am getting following error:

roslaunch rs_aligned_depth.launch 
... logging to /home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/roslaunch-agv-NUC8i5BEH-24541.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.5.101:42659/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 15
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
[... 82 lines cut ...]
[ INFO] [1678774139.282902504]: infra2 stream is enabled - width: 640, height: 480, fps: 15, Format: Y8
[ INFO] [1678774139.283470408]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[ INFO] [1678774139.285755654]: gyro stream is enabled - fps: 400
[ INFO] [1678774139.285803007]: accel stream is enabled - fps: 250
[ INFO] [1678774139.285832614]: setupPublishers...
[ INFO] [1678774139.287940514]: Expected frequency for depth = 15.00000
[ INFO] [1678774139.316879871]: Expected frequency for infra1 = 15.00000
[ INFO] [1678774139.335909399]: Expected frequency for infra2 = 15.00000
[ INFO] [1678774139.354666107]: Expected frequency for color = 15.00000
[ INFO] [1678774139.377992513]: Expected frequency for aligned_depth_to_color = 15.00000
[ INFO] [1678774139.400698475]: setupStreams...
 14/03 11:38:59,457 WARNING [140271338100480] (ds5-motion.cpp:573) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 14/03 11:38:59,466 WARNING [140270880421632] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0001/HID-SENSOR-200076.3.auto/iio:device0/buffer/enable
 14/03 11:38:59,466 WARNING [140271038297856] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0001/HID-SENSOR-200073.2.auto/iio:device1/buffer/enable
[ WARN] [1678774139.467789073]: 
[ INFO] [1678774139.535215816]: SELECTED BASE:Depth, 0
[ INFO] [1678774139.547418742]: RealSense Node Is Up!
nodelet: malloc.c:2379: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
[camera/realsense2_camera_manager-1] process has died [pid 24555, exit code -6, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/camera-realsense2_camera_manager-1.log].
log file: /home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/camera-realsense2_camera_manager-1*.log
^C[camera/realsense2_camera-2] killing on exit
[ WARN] [1678774737.629730742]: Couldn't find service realsense2_camera_manager/unload_nodelet, perhaps the manager is already shut down
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2023-03-14:
Hi @Trigon123  **rs_aligned_depth** is a very old launch file that has not been updated for several years.  The typical approach would be to instead launch with **rs_camera** with aligned depth enabled.

`roslaunch realsense2_camera rs_camera.launch align_depth:=true`

In regard to your error message, please try launching your parameters this way:

`roslaunch realsense2_camera rs_camera.launch align_depth:=true infra1_enabled:=true infra2_enabled:=true enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation accel_fps:=250 gyro_fps:=400 depth_width:=640 depth_height:=480 depth_fps:=15 color_width:=640 color_height:=480 color_fps:=15 infra1_width:=640 infra1_height:=480 infra1_fps:=15 infra2_width:=640 infra2_height:=480 infra2_fps:=15`

Trigon123, 2023-03-14:
I have tried for rs_camera.launch, camera node gets error and exit.

roslaunch realsense2_camera rs_camera.launch align_depth:=true infra1_enabled:=true infra2_enabled:=true enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation accel_fps:=250 gyro_fps:=400 depth_width:=640 depth_height:=480 depth_fps:=15 color_width:=640 color_height:=480 color_fps:=15 infra1_width:=640 infra1_height:=480 infra1_fps:=15 infra2_width:=640 infra2_height:=480 infra2_fps:=15
... logging to /home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/roslaunch-agv-NUC8i5BEH-48327.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.5.101:38075/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 70 lines cut ...]
[ INFO] [1678782175.945657992]: Motion Module was found.
[ INFO] [1678782175.945710291]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1678782175.947533695]: num_filters: 1
[ INFO] [1678782175.947557515]: Setting Dynamic reconfig parameters.
[ WARN] [1678782176.082839209]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1678782176.090998137]: Done Setting Dynamic reconfig parameters.
[ INFO] [1678782176.091723331]: depth stream is enabled - width: 640, height: 480, fps: 15, Format: Z16
[ INFO] [1678782176.092480499]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[ INFO] [1678782176.093749473]: gyro stream is enabled - fps: 400
[ INFO] [1678782176.093799431]: accel stream is enabled - fps: 250
[ INFO] [1678782176.093833807]: setupPublishers...
[ INFO] [1678782176.096289137]: Expected frequency for depth = 15.00000
[ INFO] [1678782176.130343368]: Expected frequency for color = 15.00000
[ INFO] [1678782176.153302464]: Expected frequency for aligned_depth_to_color = 15.00000
[ INFO] [1678782176.173071229]: Start publisher IMU
[ INFO] [1678782176.173875370]: setupStreams...
[... cut ...]
[ INFO] [1678782176.275612250]: SELECTED BASE:Depth, 0
[ INFO] [1678782176.286543589]: RealSense Node Is Up!
malloc(): invalid size (unsorted)
[camera/realsense2_camera_manager-1] process has died [pid 48341, exit code -6, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/camera-realsense2_camera_manager-1.log].
log file: /home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/camera-realsense2_camera_manager-1*.log

MartyG-RealSense, 2023-03-14:
It is not usual for CPU utilization to go to 100% when only enabling depth alignment, especially on a strong computer specification like your NUC.  

Does it still max-out at 100% and produce the error if you reset the camera at launch with **initial_reset:=true**

`roslaunch realsense2_camera rs_camera.launch align_depth:=true initial_reset:=true`

Trigon123, 2023-03-14:
It is not working.
roslaunch realsense2_camera rs_camera.launch align_depth:=true initial_reset:=true
... logging to /home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/roslaunch-agv-NUC8i5BEH-98071.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.5.101:36021/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 76 lines cut ...]
[ INFO] [1678784532.123745639]: RGB Camera was found.
[ INFO] [1678784532.124584469]: Motion Module was found.
[ INFO] [1678784532.124622477]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1678784532.126934008]: num_filters: 1
[ INFO] [1678784532.126967249]: Setting Dynamic reconfig parameters.
[ WARN] [1678784532.161206189]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1678784532.168328155]: Done Setting Dynamic reconfig parameters.
[ INFO] [1678784532.169308456]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1678784532.171158157]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1678784532.173157794]: setupPublishers...
[ INFO] [1678784532.176035163]: Expected frequency for depth = 30.00000
[ INFO] [1678784532.208018882]: Expected frequency for color = 30.00000
[ INFO] [1678784532.231332420]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1678784532.256031015]: setupStreams...
[ INFO] [1678784532.334522173]: SELECTED BASE:Depth, 0
[ INFO] [1678784532.376110347]: RealSense Node Is Up!
[ WARN] [1678784532.404943496]: 
malloc(): invalid size (unsorted)
[camera/realsense2_camera_manager-1] process has died [pid 98085, exit code -6, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/camera-realsense2_camera_manager-1.log].
log file: /home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

MartyG-RealSense, 2023-03-14:
Does the problem still occur if you launch without alignment enabled?

`roslaunch realsense2_camera rs_camera.launch`

If it does still occur, can you tell me what method you used to install the librealsense SDK and the RealSense ROS wrapper, please?

Trigon123, 2023-03-14:
Yes.

roslaunch realsense2_camera rs_camera.launch
... logging to /home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/roslaunch-agv-NUC8i5BEH-117728.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.5.101:33361/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 69 lines cut ...]
[ INFO] [1678786077.129807692]: RGB Camera was found.
[ INFO] [1678786077.130631380]: Motion Module was found.
[ INFO] [1678786077.130673112]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1678786077.133121015]: num_filters: 1
[ INFO] [1678786077.133163879]: Setting Dynamic reconfig parameters.
[ WARN] [1678786077.217617212]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1678786077.225818415]: Done Setting Dynamic reconfig parameters.
[ INFO] [1678786077.226843185]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1678786077.228791734]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1678786077.230707798]: setupPublishers...
[ INFO] [1678786077.233356110]: Expected frequency for depth = 30.00000
[ INFO] [1678786077.268087458]: Expected frequency for color = 30.00000
[ INFO] [1678786077.290846739]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1678786077.316685907]: setupStreams...
[ INFO] [1678786077.357693154]: SELECTED BASE:Depth, 0
[ INFO] [1678786077.367383792]: RealSense Node Is Up!
[ WARN] [1678786077.497214053]: 
malloc(): invalid size (unsorted)
[camera/realsense2_camera_manager-1] process has died [pid 117748, exit code -6, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/camera-realsense2_camera_manager-1.log].
log file: /home/agv/.ros/log/77febb04-c227-11ed-941a-1f07c1083580/camera-realsense2_camera_manager-1*.log
^C^C[camera/realsense2_camera-2] killing on exit
[ WARN] [1678786087.454813482]: Couldn't find service realsense2_camera_manager/unload_nodelet, perhaps the manager is already shut down
^Cshutting down processing monitor...
... shutting down processing monitor complete
done

Trigon123, 2023-03-14:
I have referred following instructions for installing sdk on ubuntu 20.04
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

MartyG-RealSense, 2023-03-14:
Those instructions are fine for installing the librealsense SDK on a NUC.

You would then have to build the RealSense ROS wrapper from source code (the wrapper's **Method 2** instructions).  If you installed it from packages with `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` (Method 1) then this would install both the SDK and wrapper at the same time.

If you had already installed the SDK with the **installation_md** instructions then you would then have installed the SDK twice, causing possible conflicts between the two SDK installs.

Trigon123, 2023-03-14:
How can I check SDK installed twice? if twice installed then how can remove it?

MartyG-RealSense, 2023-03-14:
You cannot check for two-times installation but you can clean up your computer and then try the installation again.  The command below removes all installed packages from the computer.  If you had used `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` then this should remove both the SDK and ROS wrapper but leave behind the SDK installation built from source code with the installation_md instructions.

dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge

You could then build the ROS wrapper from source code using the Method 2 instructions.

************

An alternative approach would be to remove the SDK that was installed from source code with installation_md.  This could be done by inputting the command below whilst in the SDK **build** folder to install the source code build:

`sudo make uninstall && make clean`

Trigon123, 2023-03-15:
Even after clean installation as per your recommendation , i am getting following error,

roslaunch realsense2_camera rs_camera.launch 
... logging to /home/agv/.ros/log/8c351070-c25c-11ed-8b88-f9f06bf7408a/roslaunch-agv-NUC8i5BEH-249366.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.5.101:41579/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 28 lines cut ...]
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [249396]
process[camera/realsense2_camera-2]: started with pid [249397]
[ INFO] [1678852219.840722883]: Initializing nodelet with 8 worker threads.
[ERROR] [1678852219.883524983]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Failed to load library /opt/ros/noetic/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librealsense2.so.2.50: cannot open shared object file: No such file or directory)
[ERROR] [1678852219.883572535]: The error before refreshing the cache was: Failed to load library /opt/ros/noetic/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librealsense2.so.2.50: cannot open shared object file: No such file or directory)
[FATAL] [1678852219.886473109]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-2] process has died [pid 249397, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/agv/.ros/log/8c351070-c25c-11ed-8b88-f9f06bf7408a/camera-realsense2_camera-2.log].
log file: /home/agv/.ros/log/8c351070-c25c-11ed-8b88-f9f06bf7408a/camera-realsense2_camera-2*.log
^C[camera/realsense2_camera_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Trigon123, 2023-03-15:
@MartyG-RealSense Now i used a fresh new NUC and installed realsense SDK and wrapper using 

`sudo apt-get install ros-noetic-realsense2-camera`

But on this NUC system also when i run 

`roslaunch realsense2_camera rs_camera.launch`

I get this error 

`Setting Dynamic reconfig parameters.
/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /opt/ros/noetic/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE`

to overcome this error i deleted 'ddynamic_reconfigure' and reinstalled it using command 'sudo apt-get install ros-noetic-ddynamic_reconigure` but could not solve the probolem.

But when i connect the same camera with my PC it works fine.

MartyG-RealSense, 2023-03-15:
Usually you should not need to install ddynamic_reconfigure when performing a package installation of librealsense and the ROS1 wrapper with **sudo apt-get install**, as the ddynamic_reconfigure installation requirement is part of the **source code** installation instructions for the ROS1 wrapper.

Can you confirm that you have only used **sudo apt-get install ros-noetic-realsense2-camera** and are not then using the source code instructions afterwards, please (as this would install the ROS wrapper a second time).

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#method-2-the-realsense-distribution

closed 2023-03-27 by MartyG-RealSense