realsenseai/realsense-ros #2657: ENU frame

shivangi-futuristicbots, 2023-03-14:
how to view the enu frame i followed the previous issues related to it but it not helping
can you please share this how to view enu frame

shivangi-futuristicbots, 2023-03-15:
i meant robot_localization assumes an ENU frame for all IMU data, so is this set by default or do i have to set it as a parameter?

MartyG-RealSense, 2023-03-15:
In the ROS map convention information at the link below, it states that when defining coordinate frames with respect to a global reference like the earth, the default should be to align the x-axis east, y-axis north, and the z-axis up at the origin of the coordinate frame.  This convention is **ENU**, which stands for East, North, Up.  

East is X, North is Y and Up is Z.

My understanding is that by default ROS uses **FLU** (Forward, Left, Up) where X is forward, Y is left and Z is up), as confirmed by the creator of the RealSense ROS1 wrapper at https://github.com/IntelRealSense/realsense-ros/issues/1099#issuecomment-658101736

closed 2023-03-27 by MartyG-RealSense