realsenseai/realsense-ros #2660: Windows 10 ROS2 Humble Wrapper colcon build issues

Dreoni, 2023-03-14:
Hello,

Im following in the steps of @mjs513 and got stuck with some of the dependencies.

First it seems it cant find "diagnostic_updater"

```
C:\ros2_ws>colcon build --merge-install
Starting >>> cv_bridge
Starting >>> realsense2_camera_msgs
Starting >>> image_geometry
Starting >>> cpp_pubsub
Starting >>> py_pubsub
Starting >>> turtlesim
--- stderr: py_pubsub
c:\python38\lib\site-packages\setuptools\command\easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
c:\python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
c:\python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< py_pubsub [3.09s]
Finished <<< image_geometry [4.34s]
Finished <<< cpp_pubsub [4.36s]
Finished <<< realsense2_camera_msgs [6.78s]
Starting >>> realsense2_description
Finished <<< cv_bridge [8.19s]
Starting >>> opencv_tests
Starting >>> realsense2_camera
Starting >>> vision_opencv
Finished <<< realsense2_description [2.31s]
Finished <<< turtlesim [9.30s]
--- stderr: opencv_tests
c:\python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< opencv_tests [2.20s]
Finished <<< vision_opencv [2.17s]
--- stderr: realsense2_camera
CMake Error at CMakeLists.txt:117 (find_package):
  By not providing "Finddiagnostic_updater.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "diagnostic_updater", but CMake did not find one.

  Could not find a package configuration file provided by
  "diagnostic_updater" with any of the following names:

    diagnostic_updaterConfig.cmake
    diagnostic_updater-config.cmake

  Add the installation prefix of "diagnostic_updater" to CMAKE_PREFIX_PATH or
  set "diagnostic_updater_DIR" to a directory containing one of the above
  files.  If "diagnostic_updater" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< realsense2_camera [3.36s, exited with code 1]

Summary: 9 packages finished [12.0s]
  1 package failed: realsense2_camera
  3 packages had stderr output: opencv_tests py_pubsub realsense2_camera
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)
```

If I add directly the package from [https://github.com/ros/diagnostics/tree/stale/humble](url) I get the following TinyXML2 not found error about 30 times in a row

```
--- stderr: diagnostic_aggregator
CMake Error in CMakeLists.txt:
  Imported target "pluginlib::pluginlib" includes non-existent path

    "TINYXML2_INCLUDE_DIR-NOTFOUND"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Target "diagnostic_aggregator" contains relative path in its
  INTERFACE_INCLUDE_DIRECTORIES:

    "TINYXML2_INCLUDE_DIR-NOTFOUND"

```

I tried reinstalling the tinyxml2.6.0.0 and tinyxml-usestl.2.6.2 as suggested by the ros2 humble troubleshooting [https://docs.ros.org/en/humble/How-To-Guides/Installation-Troubleshooting.html](url) but the issue persists.

Any help will be appreciated!

MartyG-RealSense, 2023-03-14:
Hi @Dreoni  A Humble user who experienced the same error message about diagnostic-updater being missing resolved it by running the installation command below.

`sudo apt-get install ros-humble-diagnostic-updater`

Dreoni, 2023-03-15:
Hi Marty! Unfortunately Im on Windows 10 so no "sudo apt-get" for me. I dont think choco has that package either. 

I think if I can fix the TinyXML2 dependency error I can manually install it.

The same TinyXML2 error also appeared to me before when compiling the realsense-ros wrapper. I commented the source temporarily just in case, but I guess there is no escaping it.

Just as a reference I am trying to complile the packages in a cmd VS-19 admin promt as recommended. This is the first lines I run before compiling 

```
call C:\dev\ros2_humble\local_setup.bat

call C:\ros2_ws\install\setup.bat

cd \ros2_ws

```

MartyG-RealSense, 2023-03-15:
As seen in the @mjs513 installation issue at https://github.com/IntelRealSense/realsense-ros/issues/url it is not easy to get the RealSense ROS2 wrapper running on Humble in Windows due to the number of fixes that need to be made along the way.  The official installation instructions only provide guidance for an Ubuntu colcon installation.

A method to install tinyxml2 on Windows with Chocolatey is:

`choco install -y -s <PATH_TO_DOWNLOADS> tinyxml2`

where <PATH_TO_DOWNLOADS> is the folder into which the package has been downloaded.

A choco package for tinyxml2 can be downloaded from a file list here:

https://github.com/ros2/choco-packages/releases/tag/2022-03-15

Dreoni, 2023-03-15:
Hi Marty! Thanks for the response! 
I already have that package installed and I reinstalled multiple times as suggested by the ros troubleshooting guide:

[CMake packages unable to find asio, tinyxml2, tinyxml, or eigen](https://docs.ros.org/en/humble/How-To-Guides/Installation-Troubleshooting.html#cmake-packages-unable-to-find-asio-tinyxml2-tinyxml-or-eigen)
_We’ve seen that sometimes the chocolatey packages for asio, tinyxml2, etc. do not add important registry entries and CMake will be unable to find them when building ROS 2. We’ve not yet been able to identify the root cause, but uninstalling the chocolatey packages (with -n if the uninstall fails the first time), and then reinstalling them will fix the issue._

I have tried reinstalling with admin cmd and powershell to no avail. I don't know if there is a way to manually the registry's

MartyG-RealSense, 2023-03-15:
https://github.com/ros2/ros2/issues/516#issuecomment-399664146 has some information about tinyxml2 on Windows and the registry entry.

Dreoni, 2023-03-15:
I got it! I had to reinstall cmake and then install the TinyXML2, asio,... to install cmake is as follows:

`choco install -y cmake`

Now I have the same problem @mjs513 had:

```
--- stderr: realsense2_camera
CMake Warning at CMakeLists.txt:119 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2"
  (requested version 2.51.1) with any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.
```
I also downgraded from  2.53.1 to 2.51.1 but I still get the error. Furthermore, in the RealSenseViewer I get the following error:

```
Invalid Value in rs2_get_option(options:000002659B379F10, option:Auto
Exposure Limit):
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
```

Not sure how to fix either of those problems

MartyG-RealSense, 2023-03-15:
It's great to hear that you made significant progress!

The **Auto Exposure Limit** error can occur if you are not using the recommended firmware for each SDK version.  SDK 2.53.1 should be used with firmware **5.14.0.0** whilst 2.51.1 should be used with firmware **5.13.0.50**.

MartyG-RealSense, 2023-03-15:
My reading of https://github.com/IntelRealSense/realsense-ros/issues/2587 gives me the impression that after the "realsense2" package problem was experienced, the solution in https://github.com/IntelRealSense/realsense-ros/issues/2587#issuecomment-1368555528 was to add in **Eigen3** by dowloading from Eigen.org

Dreoni, 2023-03-15:
The issue seems to stay after I downloaded Eigen 3.4 and extracted it into C:\ changing the folder name to "Eigen3". Maybe it has to do with another issue?

For now Im still stick with "realsense2 not found"

I have noticed that my SDK install in "C:\Program Files (x86)\Intel RealSense SDK 2.0" does not have a CMake folder in contrast to the source file for SDK 2.0 1.51.1

MartyG-RealSense, 2023-03-15:
The SDK is installed from a **binary** when using the Intel.RealSense.SDK-WIN10 installer program and so does not need CMake, which is for source code building.

MartyG-RealSense, 2023-03-22:
Hi @Dreoni  Do you require further assistance with this case, please?  Thanks!

Dreoni, 2023-03-22:
Hi Marty! Yes, Im still stuck on the same problem unfortunately

MartyG-RealSense, 2023-03-22:
A question in my mind is whether the installation process is installing ROS2 wrapper **4.51.1** as support for Humble was only added in that wrapper onwards.

Dreoni, 2023-03-22:
Indeed, Im using the 4.51.1 wrapper from this repository [https://github.com/IntelRealSense/realsense-ros/releases](url)

MartyG-RealSense, 2023-03-27:
Hi @Dreoni  Have you been able to make progress with installing the ROS2 wrapper on Humble and Windows, please?

Dreoni, 2023-03-27:
Hi Marty, still working on it. 
I tried doing it the "proper way" with the source files but it still didn't recognize it, so now Im modifying the various CMakeslist in order to set the file paths manually. I'll report if I manage to make it work

Dreoni, 2023-03-29:
I got it compiled!!

As I said I was forcing the code to work by hard codding all the problematic file references that seemed to be broken. Basically just a loong game of Wack a Mole: compile, find the file that it doesnt find using windows search, place it it its corresponding place, repeat. 

I didnt like that solution at all so, after all that, I also tried just putting the library (librealsense-2.51.1) directly into the ros2 workspace. Specifically at "C:\ros2_ws\src\realsense" and to my surprise it also found it and that package was able to compile! I commented out the initial hard code references but Im not 100% sure it it is still referencing the other location of the library. 

The next problem is that it didnt find "Eigen/Geometry" so I did the same and place the source code (version 3.4.0 from https://eigen.tuxfamily.org ) into "C:\ros2_ws\src\eigen3". I had to compile it alone first so that the realsense package could find it:

`colcon build --merge-install --packages-select Eigen3` 

Now It compiles! But when I try to launch the nodes the following happens:

```
C:\ros2_ws>ros2 launch realsense2_camera rs_launch.py
[INFO] [launch]: All log files can be found below C:\Users\David\.ros\log\2023-03-29-17-20-51-830721-DESKTOP-2LI2LOR-20428
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node.EXE-1]: process started with pid [16856]
[realsense2_camera_node.EXE-1] [INFO] [1680103252.235791900] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node.EXE-1] [INFO] [1680103252.235913300] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node.EXE-1] [INFO] [1680103252.235941900] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node.EXE-1] [INFO] [1680103252.295368300] [camera.camera]: Device with serial number 018322071039 was found.
[realsense2_camera_node.EXE-1]
[realsense2_camera_node.EXE-1] [INFO] [1680103252.295441600] [camera.camera]: Device with physical ID \\?\usb#vid_8086&pid_0b07&mi_03#6&33ad48ff&0&0003#{e5323777-f976-4f5b-9b55-b94699c46e44}\global was found.
[realsense2_camera_node.EXE-1] [INFO] [1680103252.295453600] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node.EXE-1] [WARN] [1680103252.295512900] [camera.camera]: Error extracting usb port from device with physical ID: \\?\usb#vid_8086&pid_0b07&mi_03#6&33ad48ff&0&0003#{e5323777-f976-4f5b-9b55-b94699c46e44}\global
[realsense2_camera_node.EXE-1] Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[realsense2_camera_node.EXE-1] [INFO] [1680103252.295535000] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node.EXE-1] [INFO] [1680103252.300647700] [camera.camera]: getParameters...
[realsense2_camera_node.EXE-1] [INFO] [1680103252.300931300] [camera.camera]: JSON file is not provided
[realsense2_camera_node.EXE-1] [INFO] [1680103252.300953000] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node.EXE-1] [INFO] [1680103252.300965000] [camera.camera]: Device Serial No: 018322071039
[realsense2_camera_node.EXE-1] [INFO] [1680103252.300974300] [camera.camera]: Device physical port: \\?\usb#vid_8086&pid_0b07&mi_03#6&33ad48ff&0&0003#{e5323777-f976-4f5b-9b55-b94699c46e44}\global
[realsense2_camera_node.EXE-1] [INFO] [1680103252.300983400] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node.EXE-1] [INFO] [1680103252.300991500] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node.EXE-1] [INFO] [1680103252.301000000] [camera.camera]: Sync Mode: Off
[realsense2_camera_node.EXE-1] [INFO] [1680103253.191316400] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node.EXE-1] [INFO] [1680103254.132993100] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x30
[realsense2_camera_node.EXE-1] [INFO] [1680103254.140453000] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node.EXE-1] [INFO] [1680103254.168398000] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node.EXE-1] [INFO] [1680103254.251667500] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node.EXE-1] [INFO] [1680103254.252239200] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node.EXE-1] [INFO] [1680103254.254107300] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node.EXE-1] [INFO] [1680103254.271480200] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node.EXE-1] [INFO] [1680103254.271886800] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node.EXE-1] [WARN] [1680103254.791839400] [camera.camera]:
[realsense2_camera_node.EXE-1] [WARN] [1680103256.185372600] [camera.camera]: XXX Hardware Notification:Depth stream start failure,1.6801e+12,Error,Hardware Error
[realsense2_camera_node.EXE-1] [WARN] [1680103256.185562600] [camera.camera]: Hardware Notification:Depth stream start failure,1.6801e+12,Error,Hardware Error
```

So good news but I still have work to do!

MartyG-RealSense, 2023-03-29:
That's awesome news that you made significant progress after persisting with it.  Congratulations  :)

Your launch log looks good aside from the **Depth stream start faiure** at the end.  Does it make a difference if you reset the camera at launch by adding **initial_reset:=true** to the end of your roslaunch instruction, please?

Dreoni, 2023-03-29:
That fixed the Depth stream failure! Thank you Marty!

The next step would be to visualize the data. I'll keep working on it

MartyG-RealSense, 2023-03-29:
Excellent!

You can visualize the data by launching the RViz interface in ROS2 by inputting **ros2 run rviz rviz** into the ROS terminal.

Dreoni, 2023-04-05:
Hi Marty im still working on it, it seems that no ros topics are published for some reason

MartyG-RealSense, 2023-04-06:
You can check with **ros2 topic echo** if a topic has data being published on it without using RViz.  For example for the ROS2 depth topic, you could input the command below into the ROS terminal after launch has completed:

`ros2 topic echo /camera/depth/image_rect_raw`

Dreoni, 2023-04-06:
I think I got it! No idea why it wasnt working before, but now the topics are published!

Now topic list shows the following

```
C:\ros2_ws>ros2 topic list
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu
/parameter_events
/rosout
/tf_static

```
To get rviz2 to work I also had to download zlib1.dll and paste it in "C:\Windows\System32" and "C:\Windows\SysWOW64". Then launching rviz2 no longer gave me the error

`ros2 run rviz2 rviz2`

Anyways thank you Marty for all the help in the process!! I still have work to do but I think I will close this.

Thanks again Marty!!

closed 2023-04-06 by Dreoni

MartyG-RealSense, 2023-04-06:
It's my pleasure to have been able to help.  Thanks so much for the detailed updates during this case that will be of future benefit to others attempting the same with Windows  :)

phil123456, 2023-08-29:
> sudo apt-get install ros-humble-diagnostic-updater

: Unable to locate package ros-humble-diagnostic-updater

SamerKhshiboun, 2024-03-22:
Hi @Dreoni,

Can you please share your installation step on windows ?
I'm unable to launch the camera node, getting all the time the same error:
[ros2run]: Process exited with failure 3221226505
as mentioned in issue #2889.
Looks like the realsense2_camera_node is failing to load the realsense2_camera.dll.. 
Tried to build it with different flags (/MT vs /MD, etc..) but nothing works.
Did you make any changes in your CMake files ?