realsenseai/realsense-ros #2661: Error installing Intel RealSense SDK for D405 camera on Jetson Nano with ROS2

kevinDrawn, 2023-03-16:
The Jetson Nano version is 20.04.06 LTS with the code name 'focal'. The ROS version is 'foxy', and when I entered the code to check the Jetpack version, it showed '#R 32, Revision: 3.1'. 
 The SDK installation method for Jetson Nano was only specified for Ubuntu 18.04 or earlier versions. However, the recently released D405 camera doesn't support ROS1, so I upgraded to ROS2 version. I followed the instructions on https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md to download the SDK, but I encountered the following error: 
'sudo apt-get install librealsense2-utils --> E: Unable to locate package librealsense2-utils'. 
How can I install the SDK for the D405 version and move on to the next step?

MartyG-RealSense, 2023-03-16:
Hi @kevinDrawn  The problem in this case may be with Ubuntu rather than librealsense or the ROS wrapper.  Installing librealsense from packages on Arm64 devices such as Jetson is supported on Ubuntu 18.04 but not on 20.04.  There are no plans to add Arm64 packages for 20.04 in future and so librealsense should instead be built from source code.

The correct order of installation is to install librealsense first and then build the ROS wrapper from source code secondly afterwards.

An instruction guide for installing librealsense on Jetson from source code can be found at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

kevinDrawn, 2023-03-18:
Thank you so much for the answer! I just want to confirm the process step by step, can you check if the order I understand is correct?

1. Download Jetpack 4.6 and revert Ubuntu 20.04 to 18.04.
2. Install ROS (either Melodic or Noetic version should be fine, but will there be any issues using the D405 camera?)
3. Install librealsense by following the instructions from the website you provided I want to install the latest SDK version 2.0 (v2.53.1) since I’m using the D405 camera, is that okay?)
4. Install ROS wrapper by following the instructions from this website: https://github.com/leggedrobotics/realsense-ros-rsl. 

If I understand the process correctly, I will let you know after I have completed it. 
Thanks a lot!!!

MartyG-RealSense, 2023-03-18:
1.  That step is correct.

2.  The RealSense ROS1 (Kinetic, Melodic or Noetic) wrapper is not compatible with D405.  Only the ROS2 **ros2-development** wrapper supports it and development of the ROS1 wrapper has ceased so D405 compatibility will not be added to it in future..  

A RealSense user disclosed at https://github.com/IntelRealSense/realsense-ros/issues/2582#issuecomment-1371133133 that the ROS1 wrapper can be edited to add D405 support.  You could likely implement this edit by creating a custom 'forked' D405-supporting version of the ROS1 wrapper on your own GitHub account that can be built from the source code instead of the official ROS1 wrapper, similar to how the leggedrobotics wrapper page that you linked to is a fork of the official wrapper.

3.  That step is correct.

4.  Instead of using the leggedrobotics version of the ROS1 wrapper, if you wanted to add D405 support for ROS1 then you'd likely have to instead create a fork of the official wrapper on your own GitHub account and make the D405 compatibility edit, then build the wrapper from your fork.

*******

Instead of using ROS1, it may be easier to install ROS2 **Dashing** and use that with Ubuntu 18.04, librealsense 2.53.1 and the **ros2-development** wrapper that is compatible with D405.

[image]

kevinDrawn, 2023-03-23:
Thank you! I solved my problem! But i have another question. 
When running a Python code that includes "import pyrealsense2 as rs" using "rosrun", I get an error message: "File "/home/park/catkin_ws/src/depth_camera/src/depth_publisher.py", line 3, in <module> import pyrealsense2 as rs". 
What could be causing this error?

<package.xml>
  <buildtool_depend>catkin</buildtool_depend>
  <build_depend>roscpp</build_depend>
  <build_depend>rospy</build_depend>
  <build_depend>std_msgs</build_depend>
  <build_depend>genmsg</build_depend>
  <build_depend>realsense2_camera</build_depend>
  <build_export_depend>roscpp</build_export_depend>
  <build_export_depend>rospy</build_export_depend>
  <build_export_depend>std_msgs</build_export_depend>
  <build_export_depend>genmsg</build_export_depend>
  <build_export_depend>realsense2_camera</build_export_depend>
  <exec_depend>roscpp</exec_depend>
  <exec_depend>rospy</exec_depend>
  <exec_depend>std_msgs</exec_depend>
  <exec_depend>genmsg</exec_depend>
  <exec_depend>realsense2_camera</exec_depend>

<CMake>
cmake_minimum_required(VERSION 3.0.2)
project(depth_camera)

find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  std_msgs
  genmsg
)

add_message_files(
  FILES
  depth.msg
)

generate_messages(
   DEPENDENCIES
   std_msgs  
)

catkin_package(
)

include_directories(include ${catkin_INCLUDE_DIRS})

catkin_install_python(PROGRAMS
  src/depth_publisher.py
  DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)
catkin_install_python(PROGRAMS
  src/depth_subscriber.py
  DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)

MartyG-RealSense, 2023-03-23:
In a small number of cases, 'import pyrealsense2 as rs' may need to be changed to **import pyrealsense2.pyrealsense2 as rs** in order for a script to work.

Also, CMake 3.8 instead of 3.0.2 is the recommended minimum for librealsense.

kevinDrawn, 2023-03-24:
Even after trying the first method and updating CMake, it still doesn't work. Do I need to make additional modifications to the package.xml or CMake files?

kevinDrawn, 2023-03-24:
Traceback (most recent call last):
  File "/home/park/catkin_ws/src/robot/src/pub.py", line 8, in <module>
    import pyrealsense2.pyrealsense2 as rs 
  File "/home/park/librealsense-2.51.1/wrappers/python/pyrealsense2/__init__.py", line 2, in <module>
    from .pyrealsense2 import *
ImportError: No module named pyrealsense2

I got this error message,

MartyG-RealSense, 2023-03-24:
Have you installed the RealSense Python wrapper **pyrealsense2** please?  The pyrealsense2 wrapper is not included in the ROS wrapper, so a Python script that uses **import pyrealsense2 as rs** would not be able to find pyrealsense2 on the computer in order to enable that instruction to work.

The pyrealsense2 wrapper can be built at the same time as the librealsense SDK if the instructions at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 are used to build librealsense on Jetson from source code.

If you used those instructions and successfully installed the pyrealsense2 wrapper, which Python 3 version are you using?  The instruction guide uses Python 3.6 (**python36**).  If you have a version of Python newer than 36 installed then you should update the version number in the installation commands.  For example, for Python 3.9:

-DPYTHON_EXECUTABLE=/usr/bin/python3.9

export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.9/pyrealsense2

If you are using **Anaconda** instead of Python 3 then the path may be different to when Python 3 is used.  The command **which python3** can tell you what the correct path is.

kevinDrawn, 2023-03-25:
Thank you for your response! I already followed your instructions and successfully installed the necessary components with Python version 3.6.9. 

Upon creating my own package in the catkin_ws/src directory and attempting to write a node to publish and subscribe to depth values at specific coordinates, I encountered an issue. Based on my understanding, I believe I need to add dependencies to the package.xml or CMakeList, but I'm not sure how to do so.

kevinDrawn, 2023-03-25:
I had an issue when dealing with package restructuring between Python 2 and Python 3, specifically when executing the "9. Deal with package restructuring between Python 2 and Python 3 using this method" step. The error message "rs.__path__" was not found, so I had to manually find the location of "pyrealsense" and run the "cp" command. 

I ran the following command: 
"sudo cp ~/librealsense-2.51.1/wrappers/python/pyrealsense2/init.py /usr/local/lib/python3.6/pyrealsense2"

After that, the rest of the process was completed without any issues, and I was able to write a regular Python script (without ROS communication) that imported "pyrealsense2" and ran the code without any problems.

MartyG-RealSense, 2023-03-25:
It's great to hear that you were successful with pyrealsense2, @kevinDrawn - thanks very much for the update!

kevinDrawn, 2023-03-25:
No. I can't solve this problem... 
I think that the the error occurred owing to interpreter. "#!/usr/bin/env python" on top of the publisher script is not match with pyrealsense2. While pyrealsense requires python3, rospy used for melodic refers to python2.  Therefore the python interpreter can't find the pyrealsense package related to python3.

MartyG-RealSense, 2023-03-25:
The documentation for the pyrealsense2 wrapper at the link below states that when building the wrapper from source code with CMake, you can force compilation for a specific Python version on a system with both Python 2 and Python 3 installed by using the CMake build flag **-DPYTHON_EXECUTABLE=[full path to the exact python executable]**

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source

lf it is possible to install for both Python 2 and 3, I would think that an approach for doing so would be to install librealsense without the pyrealsense2 wrapper, and then once librealsense is installed, build the pyrealsense2 wrapper on its own from source code with CMake twice, using different -DPYTHON_EXECUTABLE paths for Python 2 and 3 on each install.

kevinDrawn, 2023-03-26:
Thank you so much for the explanation! I will try it again today. But if I get stuck again, I want to try writing the code in C++. Since Librealsense is based on C++, it should be easier. 
I am planning to import it with #include <librealsense2/rs.hpp>! Then there shouldn't be any version conflict issues like in Python, right? Is my thinking correct?

MartyG-RealSense, 2023-03-26:
If you are writing a librealsense program that does not involve ROS then C++ can be easier, and you do not have to deal with the complications involved in installing the pyrealsense2 wrapper.

If you are using ROS and creating a librealsense **node script** that will be launched from ROS then Python will typically be easier than C++.

kevinDrawn, 2023-03-28:
Oh!! I solve this problem with python!! A lot of thank!! Have a good day!!

MartyG-RealSense, 2023-03-28:
You are very welcome.   It's great to hear that you were successful.  Thanks very much for the update!

closed 2023-04-02 by MartyG-RealSense