realsenseai/realsense-ros #2662: Noetic + CUDA branchlet

arpaterson, 2023-03-16:
Regarding building the wrapper for noetic (v2.3.2) over OpenCV + CUDA and librealsense (v2.50.0) + native USB + CUDA (from source).

It is already known that OpenCV is not linked properly (possibly just later versions of OpenCV).

Is it possible for the noetic branch of realsense-ros to get a fix for this?
Or a single commit branch containing the changes to CMakeLists.txt?

That would make it easier to use as a submodule. Currently I am manually editing CMakeLists.txt after checkout, for lack of knowing a better way to do that.

MartyG-RealSense, 2023-03-16:
Hi @arpaterson  Development of the ROS1 wrapper (Kinetic, Melodic, Noetic) has ceased and the wrapper is now called **ros1-legacy**.  The focus of RealSense ROS development is now on the **ros2-development** ROS2 wrapper.

There is the ability for RealSense ROS users to create a custom **fork** version of the ROS1 wrapper to host on their own GitHub account and make an edit to that forked version themselves.

[image]

MartyG-RealSense, 2023-03-23:
Hi @arpaterson  Do you require further assistance with this case, please?  Thanks!

arpaterson, 2023-03-25:
No, thank you for your help.

MartyG-RealSense, 2023-03-25:
Thanks very much for the update!  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-03-25 by MartyG-RealSense