realsenseai/realsense-ros #2665: RealSense(D455) unfindable on ubuntu 20.04 ROS2 foxy

cms0373, 2023-03-22:
I've installed ROS2 foxy on ubuntu 20.04 and have done all instructions given.
Succeed from STEP 1 to STEP 6 including "colcon build" in STEP 5 but when I start the camera node using "ros2 launch realsense2_camera rs_launch.py" command, something goes wrong with this result.

ms@ms-NUC12WSKi7:~/ros2_ws/build$ ros2 launch realsense2_camera rs_launch.py
[INFO] [launch]: All log files can be found below /home/ms/.ros/log/2023-03-22-16-22-13-015730-ms-NUC12WSKi7-79309
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [79311]
[realsense2_camera_node-1] [INFO] [1679469733.111380351] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1679469733.111439954] [camera.camera]: Built with LibRealSense v2.53.1
[realsense2_camera_node-1] [INFO] [1679469733.111447370] [camera.camera]: Running with LibRealSense v2.53.1
[realsense2_camera_node-1] [WARN] [1679469733.113161779] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1] [WARN] [1679469739.118875532] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1] [WARN] [1679469745.126050206] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1] [WARN] [1679469751.130079530] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1] [WARN] [1679469757.134135917] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1] [WARN] [1679469763.140048847] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1] [WARN] [1679469769.146125728] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1] [WARN] [1679469775.151718069] [camera.camera]: No RealSense devices were found!

But RealSense is actually connected. Here's result of "lsusb" command.
ms@ms-NUC12WSKi7:~/ros2_ws/build$ lsusb
Bus 004 Device 007: ID 8086:0b5c Intel Corp. Intel(R) RealSense(TM) Depth Camera 455 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 003 Device 003: ID 8087:0033 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

What might be the problem? How can I fix this.
Thank you for the response.

MartyG-RealSense, 2023-03-22:
Hi @cms0373  Is the camera detected if you reset it at launch by adding **initial_reset:=true** to your launch instruction, please?

`ros2 launch realsense2_camera rs_launch.py initial_reset:=true`

MartyG-RealSense, 2023-03-22:
What happens if you unplug the camera, wait a couple of seconds and re-insert it and then run the launch command again.

cms0373, 2023-03-22:
Still same. I've plugged RealSense in Windows PC and it is detected.

MartyG-RealSense, 2023-03-22:
I note that when you run the launch instruction you are in the directory **ros2_ws/build**.  If you **cd ~/ros2_ws** one directory back to **ros2_ws/** and launch from there, is the camera still undetected?

cms0373, 2023-03-22:
Here is the result of instruction run on "~/ros2_ws"
ms@ms-NUC12WSKi7:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py initial_reset:=true
[INFO] [launch]: All log files can be found below /home/ms/.ros/log/2023-03-22-18-04-09-297269-ms-NUC12WSKi7-98438
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [98440]
[realsense2_camera_node-1] [INFO] [1679475849.406571127] [RealSenseCameraNode]: RealSense ROS v3.2.3
[realsense2_camera_node-1] [INFO] [1679475849.406609430] [RealSenseCameraNode]: Built with LibRealSense v2.53.1
[realsense2_camera_node-1] [INFO] [1679475849.406614186] [RealSenseCameraNode]: Running with LibRealSense v2.53.1
[realsense2_camera_node-1] [WARN] [1679475849.408617872] [RealSenseCameraNode]: No RealSense devices were found!
[realsense2_camera_node-1] [WARN] [1679475855.414979256] [RealSenseCameraNode]: No RealSense devices were found!
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[realsense2_camera_node-1] [INFO] [1679475856.179488914] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [realsense2_camera_node-1]: process[realsense2_camera_node-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [realsense2_camera_node-1]: sending signal 'SIGTERM' to process[realsense2_camera_node-1]
[ERROR] [realsense2_camera_node-1]: process has died [pid 98440, exit code -15, cmd '/home/ms/camera_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_vurjzong'].

Thank you.

MartyG-RealSense, 2023-03-22:
For a small number of RealSense ROS users, a ros2 launch may not work but a **ros2 run** launch does.

`ros2 run realsense2_camera realsense2_camera_node`

cms0373, 2023-03-22:
Thank you for the response. But when I run "ros2 run realsense2_camera rs_launch.py initial_reset:=true", it says "No executable found". Same result when I run without "initial_reset:=true".

cms0373, 2023-03-22:
Here's the result of running "ros2 run realsense2_camera realsense2_camera_node"

ms@ms-NUC12WSKi7:~/ros2_ws$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1679476070.913186104] [RealSenseCameraNode]: RealSense ROS v3.2.3
[INFO] [1679476070.913226413] [RealSenseCameraNode]: Built with LibRealSense v2.53.1
[INFO] [1679476070.913230385] [RealSenseCameraNode]: Running with LibRealSense v2.53.1
[WARN] [1679476070.914992932] [RealSenseCameraNode]: No RealSense devices were found!
[WARN] [1679476076.921133450] [RealSenseCameraNode]: No RealSense devices were found!
^C[INFO] [1679476078.435760025] [rclcpp]: signal_handler(signal_value=2)

MartyG-RealSense, 2023-03-22:
Can you tell me the method that you used to install the librealsense SDK please?

cms0373, 2023-03-23:
Still undetected even I run below command.
`sudo apt-get install ros-foxy-realsense2-camera`

I went through all processes once more of below link but nothing helped me. :(
https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md

Step 2: Install the latest Intel® RealSense™ SDK 2.0
**ms@ms-NUC12WSKi7:~/ros2_ws$ cat /etc/*release | grep debian
ID_LIKE=debian**
This is my computer's status so I referred to this link.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages
Is this right method for installing SDK?
Succeed other steps including installing ROS2 wrapper from source.

cms0373, 2023-03-23:
I've tried the same method with my manager's PC and it worked. RealSense is properly detected. I don't know what made this difference. What could be the reason?

MartyG-RealSense, 2023-03-23:
As you are using a NUC, using the distribution_linux instructions to install the SDK from packages instead of building it from source code should be fine.

cms0373, 2023-03-23:
I got RealSense detected. But when I turned on "Motion Module" button on the left side, this is what happened.
[image]

MartyG-RealSense, 2023-03-23:
The red auto-exposure limit error is usually caused by not having camera firmware version **5.14.0.0** installed when using SDK 2.53.1.

MartyG-RealSense, 2023-03-23:
As it says **Firmware update done**, the update may have completed successfully despite the error message.  The command **rs-enumerate-devices -s** should show a short summary of the camera information including its currently installed firmware version.

cms0373, 2023-03-23:
[image]

Same error occured :(
What else could be the solution?
Thank you again for helping my task.

cms0373, 2023-03-23:
I don't get the meaning of "topics are being published and displayed". Sorry for my inexperience..
https://github.com/doosan-robotics/doosan-robot2
I'm also referring to this github link and when I run this command`
ros2 launch dsr_launcher2 dsr_joint_state_pub.launch.py model:=a0912 color:=blue`
I can see the example model on Rviz window.

MartyG-RealSense, 2023-03-23:
It's no trouble at all, don't worry.  :)

Once a RealSense camera has been launched then you may want to see the depth and RGB streams displayed visually onscreen.  This can be done in ROS with RViz, which can be run in ROS2 once launch has completed by inputting the command **ros2 run rviz rviz**

cms0373, 2023-03-23:
My worktime is done so i'll try it tomorrow. Thank you so much.

cms0373, 2023-03-27:
Hi, I've succeed running `roslaunch realsense2_camera rs_camera.launch`. 
[image]
This is the result, and it process stopped here.

As you mentioned, I launched Rviz by running `rviz` command at another terminal. (Since the terminal I was using stopped). But I could see nothing on Rviz window. RealSense camera is detected and I can see camera is working by running `realsense-viewer` command.

MartyG-RealSense, 2023-03-27:
If the **control_transfer returned** warning generates continuously in the log then it typically means that there is a communication problem between the camera and computer, such as a faulty USB cable or a glitch on the USB port.  

When there are such problems in the RealSense ROS wrapper, realsense-viewer is usually able to still work correctly, as you found yourself.  realsense-viewer is running directly on librealsense whilst the ROS wrapper is a layer running **on top of** librealsense and can be more sensitive to problems in the camera connection.

https://github.com/IntelRealSense/realsense-ros/issues/2386#issuecomment-1264499208 may be a helpful reference as the RealSense user in that comment was also using D455 with Ubuntu 20.04 and experiencing control_transfer returned warnings and shared advice for what worked for them in resolving it.

MartyG-RealSense, 2023-04-02:
Hi @cms0373  Do you require further assistance with this case, please?  Thanks!

cms0373, 2023-04-03:
I'm fine from now. Got my Realsense detection problem solved by uninstalling SDK and installing again. Thank you so much for your assistance.

closed 2023-04-03 by cms0373

MartyG-RealSense, 2023-04-03:
You are very welcome,, @cms0373 - it's great news that you succeeded.  Thanks very much for the update!