realsenseai/realsense-ros #2666: L515 doesn't publish continuous pointclouds and RGB images

sai2602, 2023-03-24:
Hi Team,

I have the realsense L515 sensor connected to the nvidia jetson nx board and would like to work with the pointcloud published by it. The depth is configured to 640x480 @30Hz and the RGB is at  1280x720 @15Hz with the pointcloud texture RS_STREAM_COLOR. The issue is that the sensor, when pointed towards objects at a distance (about 3m or so from the sensor), doesn't publish any pointcloud messages, but when I stand really close to the sensor, the pointcloud is published. This issue doesn't occur when I select the RS_STREAM_INFRARED texture for the pointcloud. 

Few of the similar issues that I saw mentioned issues with pointcloud being to dense and of high publish rate that the jetson cannot handle, but this issue I am facing is consistent across all resolutions and frequencies of the RGB.

System configuration:

Ubuntu 20.04 with ROS Noetic full
Librealsense version  2.53.1 (Also tried with 2.48.0, 2.51.1)
Realsense ROS v2.3.2 (Also tried with 2.3.1 and 2.3.0)

Hoping you can help me with the issue.

MartyG-RealSense, 2023-03-24:
Hi @sai2602  I recommend trying different **Visual Preset** camera configuration presets to see how altering the camera's values in particular ways affects the ability to generate a pointcloud at distance.  There should be a drop-down menu selector for the Visual Preset in the **dynamic reconfigure** graphical interface that can be opened after the camera has been launched.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#set-camera-controls-using-dynamic-reconfigure-params

You could also try try unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).  L515 cameras can perform better in one of the two available USB insertion directions.

MartyG-RealSense, 2023-03-24:
Certain types of materials may not be rendered on a depth image as well as others.  For example, on L515 see-through materials such as plastic bottles may be ignored and the background depth detail behind the bottle is rendered instead.

The **Short Range** preset can often improve the depth image in a wide range of situations by reducing the values of the Laser Power and the Receiver Gain settings.

If the streams are experiencing lag when both depth and RGB are enabled then having auto-exposure enabled and the RGB option **auto-exposure priority** disabled forces the SDK to try to maintain FPS at a constant rate instead of permitting it to vary.  In the ROS wrapper the **auto_exposure_priority** option should be under the **rgb_camera** section of the dynamic recofigure interface.  Disable this setting.

sai2602, 2023-03-24:
Disabling it still doesn't increase the frequency. The sensor is also now at a place where there are just complete solid object just to be certain that there is no packet loss.

MartyG-RealSense, 2023-03-24:
It is unusual to set depth and RGB to two different FPS (30 and 15 respectively).  With RealSense cameras, synchronization between depth and RGB frames automatically kicks in when both streams are the same FPS.  As you said that "this issue I am facing is consistent across all resolutions and frequencies of the RGB", it sounds though as if you have already tried setting both depth and RGB to 30 FPS and it made no difference.  Is that correct, please?

sai2602, 2023-03-24:
That is correct. I tried synchronizing both the depth and RGB to work at 640x480 @ 30Hz and the issue was the same. I also tried the extreme condition (RGB @60Hz) to check load on Jetson and I don't know if it's a direct correlation, but the numbers of packets lost from the pointcloud even when I am close is higher with higher frequencies. That's the reason I chose 15Hz (6Hz would be too less for me to later work with SLAM).

With respect to switching the direction of the connector, I tried that but it didn't have any impact. I also connected the camera to different USB slots on the Jetson, but that too didn't have an impact.

MartyG-RealSense, 2023-03-25:
Jetsons can also experience slowdown in the ROS wrapper when align_depth is enabled.  How does the RGB FPS react if you use this instruction:

`roslaunch realsense2_camera rs_rgbd.launch align_depth:=false`

sai2602, 2023-03-25:
I did that too. I disabled the align_depth and just for testing, I also disabled the enable_sync. Both didn't change the publish rate.

MartyG-RealSense, 2023-03-25:
I note your comment that the slowdown does not occur when using infrared as the texture instead of color.

In the ROS terminal, does the warning **No stream match for pointcloud chosen texture Process** appear?  If it does, this is an indicator that frame drops may be occurring in the color stream.

Does the frame rate get worse over time like in the case at the link below?

https://answers.ros.org/question/327772/realsense-camera-performance-degrades-quickly/

sai2602, 2023-03-26:
I do get the warnings occasionally, not continuous though. Although, in my case, it doesn't degrade overtime. So for the configuration I chose, I have 640x480 @30Hz for both depth and RGB, and the pointcloud runs at about 7Hz, and this stays the same throughout. I ran it for only about 10mins or so. Maybe if I run it for longer I might see a difference, but for the first 10mins, it stayed consistently at 7Hz.

MartyG-RealSense, 2023-03-26:
Thank you very much for the information.  Are you using the official USB cable supplied with the camera or a longer USB cable of your own choice, please?  I ask this because using longer cables with RealSense cameras can result in performance degradation if the cable is not a high quality one.  This is due to the high volume of data that the camera can transmit through the cable.

sai2602, 2023-03-26:
I am using the standard cable that came with the realsense. As a sidenote, I don't know if this would impact in anyway, but the degraded frequency is the same even when I disable the IMU.

MartyG-RealSense, 2023-03-26:
This likely won't make a difference, but have you tried adding **initial_reset:=true** to your roslaunch instruction to reset the camera at launch, please?

sai2602, 2023-03-26:
Yes, I tried with the initial reset too. But, it didn't make any difference.

MartyG-RealSense, 2023-03-27:
And how does the frequency perform if you enable the **Global Time** function with **global_time_enabled:=true**

Global Time tries to find the best match between sensors and is enabled by default on the RealSense 400 Series camera models but **disabled** by default on L515.

MartyG-RealSense, 2023-03-28:
No problem at all.  I look forward to your next report.  Good luck!

sai2602, 2023-03-29:
Hi @MartyG-RealSense, so I built librealsense now with DFORCE_RSUSB_BACKEND set to true. I have also tried with the global_time_enabled set to true and align_depth set to false. The frequency is still the same. Sub 10Hz for all cases.

MartyG-RealSense, 2023-03-29:
I note that I did not ask which roslaunch command you were originally using. Is it the one at the link below?

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#point-cloud

sai2602, 2023-03-29:
Correct. I adapted it to publish infra and IMU as well. But the base is the rs_camera.launch.

MartyG-RealSense, 2023-03-30:
If you open the **dynamic reconfigure** graphical interface by inputting the command **rosrun rqt_reconfigure rqt_reconfigure*8, are you able to disable **auto_exposure** for RGB in the '/rgb_camera' section of the options?  If you can disable it, does it make a difference to the 10Hz rate?

sai2602, 2023-03-30:
I can disable it, but it doesn't impact the publish rate. The RGB get really dark though because I mostly use the sensor indoors.

MartyG-RealSense, 2023-03-30:
I note that you are using Jetson but have not mentioned CUDA during this discussion.  Pointcloud generation and depth-color alignment can both impose a CPU processing burden on a computer / computing device that may lead to FPS drop.  If the RealSense SDK is built from Jetson packages or from source code with the CMake build flag **-DFORCE_RSUSB_BACKEND=true** included in the CMake build instruction then the SDK's CUDA support is enabled.  This support accelerates YUY to RGB conversion, pointclouds and alignment.

Also, if RGB auto-exposure is enabled then this can affect FPS when the camera is used in the dark.

sai2602, 2023-04-03:
Hi @MartyG-RealSense, I did use the DFORCE_RSUSB_BACKEND=true flag to build the librealsense. Although, I noticed that when I run the realsense, I do not see any load on the GPU, which makes sense since within my docker container, there is no CUDA dev kit installed. I will try to setup CUDA within the container and rebuild the librealsense with CUDA to see if the publish rate improves.

MartyG-RealSense, 2023-04-03:
Thanks very much.  I look forward to your next report.  Good luck!

MartyG-RealSense, 2023-04-23:
Hi @sai2602  Do you require further assistance with this case, please?  Thanks!

sai2602, 2023-04-23:
Hi @MartyG-RealSense, sorry for the delayed response. I still couldn't look into it as I was occupied with some other work. Can I please respond in a few days after I have tested it?

MartyG-RealSense, 2023-04-23:
Yes, it's no problem at all for you to respond after you have tested it.  Thanks very much for the update and good luck!

closed 2023-05-16 by MartyG-RealSense

MartyG-RealSense, 2023-06-07:
Hi @sai2602  The new 2.54.1 version of librealsense with JetPack 5.0.2 support is now available.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

Release notes:
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2541